ZerBea/hcxdumptool #1: Blacklist entire access point

strasharo, 2018-04-22:
Is there a way to blacklist an entire access point without having to specify all of the bssid+client+ssid pairs in the blacklist file? Something like a wildcard for any of the fields in the file?

ZerBea, 2018-04-23:
There is no need to add all bssid+client+ssid pairs to the blacklist.
Only one combination is enough to stop deauthentications/disassociations on this network.

But we can't stop the client trying to connect us because of "MAC randomization".
We use a randomized mac and most of the clients use a randomized mac.

start capturing (stop with ctrl+c)
INTERFACE: wlp39s0f3u4u5
MAC_AP...: 00182534639c (rogue access point)
MAC_STA..: fcc2333c3cf1 (rogue client)

Take a look at this example:
We start hcxdumptool against an Android 6 test client

$ sudo hcxdumptool -i wlp39s0f3u4u5 -o test.pcap -s -t 60 -c 11
start capturing (stop with ctrl+c)
INTERFACE: wlp39s0f3u4u5
MAC_AP...: 00269f8920cf (rogue access point)
MAC_STA..: fcc23386b99d (rogue client)
[08:38:55] 00269f8920d0:f072cea7edfd:Testnetwork [HANDSHAKE]
terminated...

we use a randomized  mac and the client used a randomized mac.
the client connected to us and we retrieved a handshake.
we stopped hcxtumptool.

Now we started hcxdumptool again.

$ sudo hcxdumptool -i wlp39s0f3u4u5 -o test.pcap -s -t 60 -c 11
start capturing (stop with ctrl+c)
INTERFACE: wlp39s0f3u4u5
MAC_AP...: 24bf747e299a (rogue access point)
MAC_STA..: fcc233144966 (rogue client)
[08:39:55] 00269f8920d0:a68e3357e491:Testnetwork [HANDSHAKE]
terminated...

as you can see, the client used a different mac
and tries to connect us with his the stored informations from the last attempt to connect us
 
Now we restart the client (turn off / turn on cell phone)
$ sudo hcxdumptool -i wlp39s0f3u4u5 -o test.pcap -s -t 60 -c 11
start capturing (stop with ctrl+c)
INTERFACE: wlp39s0f3u4u5
MAC_AP...: 0418b6d009f8 (rogue access point)
MAC_STA..: f0a2250bbd7b (rogue client)
[08:48:01] 0418b6d009f9:c35f72a6f9d3:Testnetwork [HANDSHAKE]

now we have complete new mac addresses.

as you can see, a BPF or a balcklist is useless (and waste only cpu time).

ZerBea, 2018-04-23:
Generally MAC randomization is a good idea. It prevents tracking.
But it also prevents that we can use a blacklist.

The same applies to us. To prevent counter measures against us, we mus randomize our mac.

strasharo, 2018-04-23:
Thanks for the detailed explanation. My point was mostly around completely excluding a specific bssid from being captured like I was doing previously with the BPF filters. For example I have a bunch of access points in the area that I'm not interested in and I don't want to capture handshakes from them at all, so in wlandump-ng I am doing !(wlan host DE:AD:BE:EF:00:00) and it is never capturing handshakes from that network. Is that currently possible with hcxdumptool?

ZerBea, 2018-04-23:
Ok. You can do this using the Option -D
-D             : do not transmit deauthentications or disassociations
In that case no accesspoint is going to be attacked.

But to answer your question:
Right now it isn't possible to implement your requested feature without a complete rewrite of hcxdumptool. That's a reason why I didn't remove wlandump-ng from hcxtools and created a new
thread for hcxdumptool.
If you compare both tools, you will see, that hcxdumptool will capture much more handshakes from clients, than wlandump-ng.
The only chance to filter unwanted handshakes is using wlanhcx2ssid.

strasharo, 2018-04-23:
Okay, got it. Just wanted to confirm if I'm not missing some functionality there. I'll be stripping them using wlanhcx2ssid. Thanks a lot for your help. :)

closed 2018-04-23 by strasharo