ZerBea/hcxdumptool #12: failed to read packet: Network is down

mweinelt, 2018-08-05:
I'm trying to capture using an Intel Wireless-AC 9200 (iwlwifi kmod)

```
06:00.0 Network controller: Intel Corporation Wireless-AC 9260 (rev 29)
```

and run into the following issue, without the tool finding anything usable.

```
$ sudo ./hcxdumptool/hcxdumptool -o test.pcapng -i wlp6s0 --enable_status 

start capturing (stop with ctrl+c)
INTERFACE:...............: wlp6s0
FILTERLIST...............: 0 entries
MAC CLIENT...............: fcc23399c311 (client)
MAC ACCESS POINT.........: 11111141b95a (start NIC)
EAPOL TIMEOUT............: 1000000
DEAUTHENTICATIONINTERVALL: 20 beacons
GIVE UP DEAUTHENTICATIONS: 10 tries
REPLAYCOUNTER............: 61654
ANONCE...................: 43d3a6696159fbcca67531e333d7946b0e8c9a914ce90137613211cea69d43e9

INFO: cha=13, rx=0, rx(dropped)=0, tx=7, powned=0, err=0
failed to read packet: Network is down
INFO: cha=9, rx=0, rx(dropped)=0, tx=70, powned=0, err=1
failed to read packet: Network is down
INFO: cha=5, rx=0, rx(dropped)=0, tx=133, powned=0, err=2
failed to read packet: Network is down
INFO: cha=1, rx=0, rx(dropped)=0, tx=196, powned=0, err=3 
failed to read packet: Network is down
INFO: cha=10, rx=0, rx(dropped)=0, tx=259, powned=0, err=4
failed to read packet: Network is down
INFO: cha=6, rx=0, rx(dropped)=0, tx=322, powned=0, err=5
failed to read packet: Network is down
INFO: cha=2, rx=0, rx(dropped)=0, tx=385, powned=0, err=6 
failed to read packet: Network is down
INFO: cha=11, rx=0, rx(dropped)=0, tx=448, powned=0, err=7
failed to read packet: Network is down
INFO: cha=7, rx=0, rx(dropped)=0, tx=511, powned=0, err=8
failed to read packet: Network is down
INFO: cha=3, rx=0, rx(dropped)=0, tx=574, powned=0, err=9
failed to read packet: Network is down
INFO: cha=12, rx=0, rx(dropped)=0, tx=637, powned=0, err=10
failed to read packet: Network is down
INFO: cha=8, rx=0, rx(dropped)=0, tx=700, powned=0, err=11
failed to read packet: Network is down
INFO: cha=4, rx=0, rx(dropped)=0, tx=763, powned=0, err=12
failed to read packet: Network is down
INFO: cha=13, rx=0, rx(dropped)=0, tx=826, powned=0, err=13
failed to read packet: Network is down
INFO: cha=9, rx=0, rx(dropped)=0, tx=889, powned=0, err=14
failed to read packet: Network is down
INFO: cha=5, rx=0, rx(dropped)=0, tx=913, powned=0, err=15
[...]
```

I put the device into monitor mode beforehand:

```
$ iw dev
phy#0
	Unnamed/non-netdev interface
		wdev 0x4
		addr 30:24:32:**:**:**
		type P2P-device
		txpower 0.00 dBm
	Interface wlp6s0
		ifindex 2
		wdev 0x1
		addr 7e:0b:f7:**:**:**
		type monitor
		txpower 22.00 dBm
```

And made sure that the admin state is up.

```
2: wlp6s0: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state DORMANT group default qlen 1000
    link/ieee802.11/radiotap 7e:0b:f7:**:**:** brd ff:ff:ff:ff:ff:ff
```

What could be the issue?

ZerBea, 2018-08-05:
Does hcxdumptool have full and only access to the interface?
All services with access to the device must be disabled.
For example:
NetworkManager
wpa_spupplicant

2) hcxdumptool running a raw socket. Is this supported by the driver? Does packet injection work?

ZerBea, 2018-08-06:
Does tcpdump work on Intel Wireless-AC 9260 (iwlwifi kmod)?
sudo tcpdump -i wlp6s0 -w test.cap

mweinelt, 2018-08-06:
I disconnected WiFi in NetworkManager so it probably wouldn't interfere. 

When I previously used tcpdump on the WiFi interface I only so the probe requests sent out by hcxdumptool. That probably means that monitor mode isn't working properly, right?

```
#  sudo tcpdump -i wlp6s0 
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wlp6s0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 262144 bytes
13:32:37.839587 fragmented [bit 15] Probe Request () [1.0* 2.0* 5.5* 11.0* 6.0* 9.0* 12.0* 18.0* Mbit]
13:32:42.839576 fragmented [bit 15] Probe Request () [1.0* 2.0* 5.5* 11.0* 6.0* 9.0* 12.0* 18.0* Mbit]
13:32:47.839794 fragmented [bit 15] Probe Request () [1.0* 2.0* 5.5* 11.0* 6.0* 9.0* 12.0* 18.0* Mbit]
13:32:52.839962 fragmented [bit 15] Probe Request () [1.0* 2.0* 5.5* 11.0* 6.0* 9.0* 12.0* 18.0* Mbit]
13:32:57.840173 fragmented [bit 15] Probe Request () [1.0* 2.0* 5.5* 11.0* 6.0* 9.0* 12.0* 18.0* Mbit]
13:33:02.840392 fragmented [bit 15] Probe Request () [1.0* 2.0* 5.5* 11.0* 6.0* 9.0* 12.0* 18.0* Mbit]
13:33:07.840558 fragmented [bit 15] Probe Request () [1.0* 2.0* 5.5* 11.0* 6.0* 9.0* 12.0* 18.0* Mbit]
13:33:12.840744 fragmented [bit 15] Probe Request () [1.0* 2.0* 5.5* 11.0* 6.0* 9.0* 12.0* 18.0* Mbit]
```

ZerBea, 2018-08-06:
Yes, that looks like a driver issue.
tcpdump doesn't receive packets from outside, so hcxdumptool will fail, too.
The shown proberequests are loopback's.

mweinelt, 2018-08-06:
Generally iwlwifi seems to support both monitor mode and packet injection, I'll retry later tonight on a AC-7260 and if that doesn't work I'll close this issue for lack of driver support.

mweinelt, 2018-08-06:
For your forum post: ath9k and ath10k cards usually work pretty well on linux. I tried this on a Compex WLE600VX (QCA9882, https://www.compex.com.sg/product/wle600vx/) and capturing it just worked.

ZerBea, 2018-08-06:
Thanks,
That is good to know, because we have some really ugly issues with ath9k_htc:
https://hashcat.net/forum/thread-6661-post-41311.html#pid41311
are the transmitted ack frames damaged?

mweinelt, 2018-08-06:
ath9k_htc are 11n usb sticks, can't recommend any, nor their driver. For WiFi MiniPCIe is still king.

That being said, after ~20 minutes I don't seem to have found any PMKID on the ath10k card.
```
INFO: cha=116, rx=24274, rx(dropped)=147, tx=1901, powned=0, err=0
```

```
file name....................: test.pcapng-1
file type....................: pcapng 1.0
file hardware information....: x86_64
file os information..........: Linux 4.9.0-6-amd64
file application information.: hcxdumptool 4.2.0
network type.................: DLT_IEEE802_11_RADIO (127)
endianess....................: little endian
read errors..................: flawless
packets inside...............: 306
skipped packets..............: 0
packets with FCS.............: 0
WDS packets..................: 60
beacons (with ESSID inside)..: 70
probe requests...............: 4
probe responses..............: 9
association requests.........: 1
association responses........: 76
reassociation responses......: 63
authentications..............: 8
authentications (OPEN SYSTEM): 3
authentications (SHARED KEY).: 8
```

Looking into the pcapng with Wireshark I can see alot of malformed (re)association responses, is that the issue you're seeing with ath9k_htc as well?

ZerBea, 2018-08-06:
Yes, I think so. Maybe I'm able to find a solution for that Frame CheckSum (FCS) issue.
Not all packets are affected. I noticed it on many frames, but not on all.
It seems that they are doing their own thing:
ATH_TX_RADIOTAP_PRESENT(
	(1 << IEEE80211_RADIOTAP_TSFT) | \
	(1 << IEEE80211_RADIOTAP_FLAGS) | \
	(1 << IEEE80211_RADIOTAP_RATE)  | \
	(1 << IEEE80211_RADIOTAP_DBM_TX_POWER)  | \
	(1 << IEEE80211_RADIOTAP_ANTENNA)  | \
	(1 << IEEE80211_RADIOTAP_XCHANNEL)  | \
	0)

struct ath_tx_radiotap_header
{
struct ieee80211_radiotap_header wt_ihdr;
uint64_t	wt_tsf;
uint8_t	wt_flags;
uint8_t	wt_rate;
uint8_t	wt_txpower;
uint8_t	wt_antenna;
uint32_t	wt_chan_flags;
uint16_t	wt_chan_freq;
uint8_t	wt_chan_ieee;
int8_t	wt_chan_maxpow;
} __packed;

ZerBea, 2018-08-06:
Added hardware handshake support (instead of software handshake). Now ATHEROS driver should work, too.
https://github.com/ZerBea/hcxdumptool/commit/750f870492b8e79f75fb8e4fbdb8f92374a5e86e
Please test.

mweinelt, 2018-08-07:
This shrunk the amount of malformed packets from 50% to around 25%.

```
start reading from ../hcxdumptool/test.pcapng-3
failed to read packet

summary:
--------
file name....................: test.pcapng-3
file type....................: pcapng 1.0
file hardware information....: x86_64
file os information..........: Linux 4.9.0-6-amd64
file application information.: hcxdumptool 4.2.0
network type.................: DLT_IEEE802_11_RADIO (127)
endianess....................: little endian
read errors..................: yes
packets inside...............: 5038
skipped packets..............: 0
packets with FCS.............: 0
WDS packets..................: 121
beacons (with ESSID inside)..: 2083
probe requests...............: 209
probe responses..............: 122
association requests.........: 23
association responses........: 390
reassociation responses......: 134
authentications..............: 13
authentications (OPEN SYSTEM): 182
authentications (SHARED KEY).: 20
authentications (BROADCOM)...: 152
EAPOL packets................: 275
best handshakes..............: 2 (ap-less: 0)
```

No PMKID found yet :(

closed 2018-08-12 by ZerBea

MKHDNP, 2018-09-16:
root@kali:~/Desktop/hcxdumptool# ./hcxdumptool -o hash -i wlan0mon --filterlist=filter.txt --filtermode=2 --enable_status=1

start capturing (stop with ctrl+c)
INTERFACE:...............: wlan0mon
FILTERLIST...............: 1 entries
MAC CLIENT...............: fcc233dbee6a
MAC ACCESS POINT.........: 000e2a01d37a (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 65532
ANONCE...................: 89388910225444cf835c4c9215bb75c3cc679b819230856875b12ec384750c0e

INFO: cha=7, rx=3695, rx(dropped)=930, tx=72, powned=0, err=0
failed to read packet: Network is down
INFO: cha=5, rx=3695, rx(dropped)=930, tx=72, powned=0, err=753
terminated...

anyone can solve this problem? i wll use Alfa wireless device and airmon-ng check kill all process!

ZerBea, 2018-09-16:
Are you running hcxdumptool an the physical interface or on a virtual interface created by airmon-ng?

What type of Alfa do you use awus....?
$ lsusb | grep Wire
output should look like this:
Bus 001 Device 004: ID 0bda:8187 Realtek Semiconductor Corp. RTL8187 Wireless Adapter

What is the output of 
$ iw dev wlan0mon info

Please read this:
https://hashcat.net/forum/thread-6661-post-41821.html#pid41821

MKHDNP, 2018-09-30:
ALFA
802.11g High Power
Wirelwss USB Adapter
Model: AWUS036H	
Bus 001 Device 003: ID 0bda:8187 Realtek Semiconductor Corp. RTL8187 Wireless Adapter
this out info
Interface wlan0mon
	ifindex 5
	wdev 0x100000002
	addr 00:c0:ca:30:0e:fb
	type monitor
	wiphy 1
	channel 10 (2457 MHz), width: 20 MHz (no HT), center1: 2457 MHz
	txpower 20.00 dBm
But use another two wireless adapter using same error message failed to read packet: Network is down?

ZerBea, 2018-09-30:
I assume, you use the latest version of hcxdumptool. If not, please do git pull or git clone.
Do you use the latest version of KALI?
Do not use airmon-ng to set monitor mode. hcxdumptool creates its own monitor mode.
Frome README.md:
- do not run hcxdumptool on logical interfaces (monx, wlanxmon)
- do not use hcxdumptool in combination with other 3rd party tools, which take access to the interface
airmon-ng is designed to run with aircrack-ng tools and not with hcxdumptool
aircrack-ng is know to work on many platforms, but it fails sometimes on KALI.
https://null-byte.wonderhowto.com/how-to/fix-network-is-down-airodump-ng-0166996/
Identify all(!) services that take access to the device and stop them (NetworkManager, wpa-supplicant)
https://docs.kali.org/installation/troubleshooting-wireless-driver-issues
KALI is designed for penetration testers and neither easy to configure nor easy to use!!!
Check your KALI configuration (is wireless-regdom configured).
https://kali.training/lessons/5-configuring-kali-linux/

Please read this:
https://unix.stackexchange.com/questions/399626/why-is-kali-linux-so-hard-to-set-up-why-wont-people-help-me