ZerBea/hcxdumptool #14: Not get result even after more than half hour

Beegrekokto, 2018-08-08:
I have "Ralink corp. RT3290 Wireless 802.11n 1T/1R PCIe" adapter which comes with my laptop.
I tried:
hcxdumptool -o test.pcapng -i wlo1mon --filterlist=aashik.txt --filtermode=2 --enable_status=4
And waited about an half hour but still don't get PMKId.

ZerBea, 2018-08-08:
Pushed an update. Please do a git pull an test it again.
But it's also possible that there are no vulnerable routers in your range.

Beegrekokto, 2018-08-08:
what are the specs of router those are vulnerable routers. I have been testing my own router.

Beegrekokto, 2018-08-08:
> 802.11i or 802.11r activated on router side.

is 802.11n vulnerable

ZerBea, 2018-08-08:
Well, 802.11n (a,b,g,ac) is a draft for wireless networking transmission methods (for example data rates),  and 802.11i for security mechanisms (protocols).
If 802.11 is activated on  802.11n, 802.11n is vulnerable.

ZerBea, 2018-08-08:
Closed, since it isn't an issue.
If you need more information about techniques tactics and procedures, please read this threads (completely) and ask your questions there:
https://hashcat.net/forum/thread-6661-page-33.html
https://hashcat.net/forum/thread-7717.html

closed 2018-08-08 by ZerBea

ashleybroughton, 2018-08-10:
@Beegrekokto Make sure you enable monitor mode on the wireless adapter