ZerBea/hcxdumptool #15: big endian problem

bol-van, 2018-08-08:
It does not recognize frames on big endian systems. Also accesses memory outside of the buffer (because length is not 0x30 but 0x3000)
Fixed it by changing
rth->it_len    to    le16toh(rth->it_len);

bol-van, 2018-08-08:
Bit endian is actual because its good to run this tool on home routers with MIPS cpu

ZerBea, 2018-08-08:
Pushed an update.
I thought, we write all stuff into a pcapng file and set BE or LE flag in SHB block.
Well, so I forgot RTH.
I hope this fixed it:
	#ifdef BIG_ENDIAN_HOST
	ieee82011_ptr = packet_ptr +byte_swap_16(rth->it_len);
	ieee82011_len = packet_len -byte_swap_16(rth->it_len);
	#else
	ieee82011_ptr = packet_ptr +rth->it_len;
	ieee82011_len = packet_len -rth->it_len;
	#endif

bol-van, 2018-08-08:
Yes it also works but why not use le16toh, its shorter

ZerBea, 2018-08-08:
For shure. I was so blind (copy and paste).
Done and pushed.

bol-van, 2018-08-08:
Also check second occurence in dotargetscan()

ZerBea, 2018-08-08:
Yes, thanks.
This will be a different status out to do a quick scan and see targets in tx range.
Will finish this next week.

closed 2018-08-08 by ZerBea