ZerBea/hcxdumptool #21: enable_status option requires argument

susmithHCK, 2018-08-09:
Im using this tool to test latest WPA2 flaw (PMKID capturing). In many tutorials, the command is issued with "--enable_status" option. however it takes values 1,2,4,8. Whenever i use any of these options, if PMKID is captured its not logged in terminal.

**enable_status=1**
```
sudo hcxdumptool -o test2.pcapng -i wlp2s0mon  --filterlist=f2-matrix.txt --filtermode=2 --enable_status=1      
[sudo] password for thor: 

start capturing (stop with ctrl+c)
INTERFACE:...............: wlp2s0mon
FILTERLIST...............: 1 entries
MAC CLIENT...............: fcc233c3026a (client)
MAC ACCESS POINT.........: 1100aa7f0d1a (start NIC)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 64933
ANONCE...................: 27eeea4b47739b815b2c536e46bac31215c891269ebb4633e41376aec13bd269

INFO: cha=3, rx=119, rx(dropped)=0, tx=2, powned=0, err=0^C
```

**enable_status=2**
```
sudo hcxdumptool -o test2.pcapng -i wlp2s0mon  --filterlist=f2-matrix.txt --filtermode=2 --enable_status=2

start capturing (stop with ctrl+c)
INTERFACE:...............: wlp2s0mon
FILTERLIST...............: 1 entries
MAC CLIENT...............: fcc233703b20 (client)
MAC ACCESS POINT.........: 0418b60ad190 (start NIC)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 61990
ANONCE...................: 5fbd0055af0dcc03143a14bf3cddaa1acc31f9b755dc4900fd5bf0727f909bde

[15:03:34 - 003] b4e62a183a06 -> ffffffffffff matrix [PROBEREQUEST, SEQUENCE 2920]
[15:03:40 - 007] f4f26d25b268 -> fcc233703b20 matrix [PROBERESPONSE, SEQUENCE 1215, AP CHANNEL 8]
INFO: cha=7, rx=192, rx(dropped)=16, tx=14, powned=1, err=0^C

```
The same happens for option 4 and 8 as well.

BUT for some strange reason if "**--enable_status=1,2,4,8**"  its showing the status properly.

```
sudo hcxdumptool -o test2.pcapng -i wlp2s0mon  --filterlist=f2-matrix.txt --filtermode=2 --enable_status=1,2,4,8

start capturing (stop with ctrl+c)
INTERFACE:...............: wlp2s0mon
FILTERLIST...............: 1 entries
MAC CLIENT...............: f0a225ee1382 (client)
MAC ACCESS POINT.........: acde482d1654 (start NIC)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 63478
ANONCE...................: 75ec843719a0d17613de85ca32cb4cd8c02b56e41d68b39f0de285c384e794f9

[14:42:05 - 001] f4f26d25b268 -> f0a225ee1382 [FOUND PMKID CLIENT-LESS]
[14:42:06 - 001] f4f26d25b268 -> e8de27125d90 [FOUND PMKID]
```

Please fix this issue, Thanks

ZerBea, 2018-08-09:
It's not an issue. It's just wrong use of --enable-status:
We are using a bitmask!
--enable_status=<digit>            : enable status messages
                                     bitmask:
                                     1: EAPOL
                                     2: PROBEREQUEST/PROBERESPONSE
                                     4: AUTHENTICATON
                                     8: ASSOCIATION
That means:
If you would like to see EAPOL and PROBEREQUESTS, you have to use:
--enable_status=1 --enable_status=2
or
--enable_status=3
00000000
00000001 EAPOL
00000010 PROBEREQUEST/PROBERESPONSE
00000100 AUTHENTICATON
00001000 ASSOCIATION

So, EAPOL and PROBEREQUEST/PROBERESPONSE =
00000001 or 00000010 == 00000011
in decimal
1 or 2 = 3
--enable_status=3 will show you  EAPOL and PROBEREQUEST/PROBERESPONSE.

Read more here:
https://en.wikipedia.org/wiki/Bitwise_operation

closed 2018-08-09 by ZerBea

renonet, 2020-09-13:
Hi ZerBea,

I got this from [PMKIDROGUE:55bcf627349fd188abda13fefd997461 KDV:2] hcxdumptool.

What does PMKIDROUGE mean?

ZerBea, 2020-09-13:
Answere here:
https://github.com/ZerBea/hcxdumptool/issues/138

asentroy, 2024-02-05:
when I run this command it's show me this result.. why it's error?
-o isn't also -h section!

hcxdumptool -o testresult.pcapng -i wlan0mon --filterlist=f2-text.txt --filtermode=2 --enable_status=1 
hcxdumptool: invalid option -- 'o'
hcxdumptool 6.3.1 (C) 2023 by ZeroBeat
usage: hcxdumptool -h for help

mounavri, 2026-01-12:
why all the versions i install doesn't have the windows with what's going on live, now everything runs in the back and you can't see if you have something or not live

ZerBea, 2026-01-12:
@mounavri , due to performance reasons the status display is disabled by default. The impact (especially on slow systems) is huge! 

To enable the status display, use use one of the options mentioned below:
```
$ hcxdumptool -h
--rds=<digit>    : enable real time display
                    attack mode:
                     0 = off(default)
                     1 = show APs on current channel, show CLIENTs (M1M2ROGUE)
                     2 = show all APs (M1M2, M1M2M3 or PMKID), show CLIENTs (M1M2ROGUE)
                     3 = show all APs, show CLIENTs (M1M2ROGUE)
                     4 = show waterfall (PMKID & EAPOL)
                          TIME TYPE MAC_AP MAC_CLIENT ESSID
                     columns:
                      E = encryption (e)ncrypted / (o)pen
                      A = AKM (p)re-shared key
                      1 = received M1 (+)
                      2 = received M1M2 (+)
                      3 = received M1M2M3 (+)
                      P = received PMKID (+)
                    rcascan mode
                     0 = show APs on current channel sorted by BEACON timestamp
                     1 = show APs on current channel sorted by PROBERESPONSE timestamp
                     2 = show APs on current channel sorted by RSSI
                     3 = show APs of all channels sorted by RSSI
--rdt            : disable TIOCGWINSZ for real time displays

```