ZerBea/hcxdumptool #22: Error compiling: too many arguments for format

bswinnerton, 2018-08-10:
I tried running `make` today and ran up against:

```
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c -lpthread 
hcxdumptool.c: In function ‘process80211probe_resp’:
hcxdumptool.c:1842:18: warning: too many arguments for format [-Wformat-extra-args]
  fprintf(stdout, " [PROBERESPONSE, SEQUENCE %d, AP CHANNEL %d]\n", c, macfrx->sequence >> 4, apchannel);
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

It looks like that line was touched today in 3c486c8480ad11941ca6ae6e70a1e088eae2b456. Is it possible that commit added a regression?

AltRuskie, 2018-08-10:
+1. Tried installing about an hour ago and appears to still be an issue.

bswinnerton, 2018-08-10:
ba185b7 did indeed fix it, thank you @ZerBea!

closed 2018-08-10 by bswinnerton