ZerBea/hcxdumptool #27: unable to use --filterlist

yahway, 2018-08-11:
hello,

I try to use hcxdumptool through the following cmd 👍 

hcxdumptool - o hash -i wlan0mon --filterlist=list.txt --filtermode=2 --enable_status

I get the following error 👍 

hcxdumptool: option '--enable_status' require an argument
invalid argument specified.

Any idea about the argument waited by hcxdumptool?

thanks for help.

kali-user1, 2018-08-12:
i have the same issue ....

yahway, 2018-08-12:
Thanks it works like a charm.

ZerBea, 2018-08-12:
Nice to hear that.
Most of the reported issues are related to a misconfigured system, lack of driver support or wrong usage of the options.

kali-user1, 2018-08-12:
bitmask:
1: EAPOL
2: PROBEREQUEST/PROBERESPONSE
4: AUTHENTICATON
8: ASSOCIATION

if i want to get all these bitmask in one command then?what should i use:
--enable_status=???

i have run it for atleast 10 mints but i am not getting the "pmkid found" message..
plz help me!!!!  :(

ZerBea, 2018-08-12:
First, answer yourself this questions:
does my driver support full monitor mode?
is the interface set to monitor mode?
are services like NetworkManager and wpa_supplicant stopped?
do I use the latest commit of hcxdumptool?
did I read help menu and README.md?
If you answered all with yes run:
sudo hcxdumptool -o test.hccapx -i <interface> -t 15 --enable-status=3

hashemahmad, 2018-11-18:
--filterlist is not working! v5.0.0

ZerBea, 2018-11-18:
you need at least 2 options: filterlist and filtermode
git clone latest commit
add target to filterlist

then use
--filterlist=<your filterlist> --filtermode=x

--filterlist=<file>                : mac filter list
                                     format: 112233445566 + comment
                                     maximum line lenght 255, maximum entries 64
--filtermode=<digit>               : mode for filter list
                                     1: use filter list as protection list (default) in transmission branch
                                        receive everything, interact with all APs and CLIENTs in range,
                                        except(!) the ones from the filter list
                                     2: use filter list as target list in transmission branch
                                        receive everything, only interact with APs and CLIENTs in range,
                                        from the filter list
                                     3: use filter list as target list in receiving branch
                                        only receive APs and CLIENTs in range,
                                        from the filter list

hashemahmad, 2018-11-18:
This is my command:
hcxdumptool -o hash -i wlan1mon --filterlist=list.txt --filtermode=2 --enable_status=3
list.txt has only 1 MAC address in 112233445566 format, still hcxdumptool scans all APs. Thank you for replaying.

ZerBea, 2018-11-18:
--filterlist=list.txt --filtermode=2
use filter list as target list in transmission branch
receive everything, only interact with APs and CLIENTs in range,
from the filter list
That means:
interact only with the entries in the filter list
and(!)
receive complete traffic!
So if hcxdumptool receive a handshake and/or PMKID it will save it to the capture file,
but it will not interact with the access point or the client.

If you do not want this, you must use filter mode 3
That means:
interact only with the targets in the target list
and(!)
receive only traffix from entries of the target list!

hashemahmad, 2018-11-18:
Its working sir!

ZerBea, 2018-11-18:
That's great. Many users have asked for this feature, so I added filter mode 3 to hcxdumptool.

axeldog, 2023-10-23:
Need help with hcxdumptool on nethunter. When running command "hcxdumptool -i wlan0 -w pmkid.pcapng" it runs and pumps in pmkid.pcapng but if i add any more to the command like --filterlist_ap=my filter.txt --filternode=3 or --enable_status=1 it throws "2 errors during run time" "failed to arm interface" .

ZerBea, 2023-10-23:
This option (--filterlist_ap=my filter.txt --filternode=3 or --enable_status=1) have been removed since v6.3.0.
From now on, it is mandatory to add a Berkeley Packet Filter (BPF) that do the entire filtering.
More information is here:
https://github.com/ZerBea/hcxdumptool/discussions/343
https://github.com/ZerBea/hcxdumptool/discussions/330
as well as here:
`$ hcxdumptool --help`
and in changelog:
https://github.com/ZerBea/hcxdumptool/blob/master/changelog#L83