ZerBea/hcxdumptool #28: No results

anonymouse65, 2018-08-12:
Hi there, 

./hcxdumptool -i wlo1 --enable_status -c 11 -o test.pcapng

start capturing (stop with ctrl+c)
INTERFACE:...............: wlo1
FILTERLIST...............: 0 entries
MAC CLIENT...............: fcc23336aea3 (client)
MAC ACCESS POINT.........: 5c6b4f05fcf3 (start NIC)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 64221
ANONCE...................: 143c3b909c823a0bb81c45274616f0135532e35013d178698562a0d0fc75d712

Beyond this I receive no further results. Even if I don't specify what channel or try bitmasking, still no results come through. 

If I try a specific enable status I get this:

./hcxdumptool -i wlo1 --enable_status=1 -c 11 -o test.pcapng

start capturing (stop with ctrl+c)
INTERFACE:...............: wlo1
FILTERLIST...............: 0 entries
MAC CLIENT...............: f0a2255ed74c (client)
MAC ACCESS POINT.........: 000101485130 (start NIC)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 63472
ANONCE...................: 9d1fdf0518bd421217bce5866614ac2b334e5f581682c08a31b10ababf8fdfdd

INFO: cha=11, rx=10, rx(dropped)=10, tx=1, powned=0, err=0^C

(nothing else).

ZerBea, 2018-08-12:
1. Does your driver support full monitor mode
2. is the interface set to monitormode?
3. are services like NetworkManager and wpa_supplicant stopped?

closed 2018-08-12 by ZerBea

ZerBea, 2018-08-29:
Correct, but iwconfig is deprecated:
https://dougvitale.wordpress.com/2011/12/21/deprecated-linux-networking-commands-and-their-replacements/

ip link set $WLANDEV down
iw dev $WLANDEV set type monitor
ip link set $WLANDEV up
iw dev $WLANDEV info

ZerBea, 2018-08-30:
I get rid of 3rd party tools (ifconfig, iw, ip, ...). Added this functionality to hcxdumptool.
read more here:
https://hashcat.net/forum/thread-6661-post-41821.html#pid41821