ZerBea/hcxdumptool #30: Cant install tool

siempay, 2018-08-21:
I downloaded the tool and tried `make` but I get this  `make: Nothing to be done for `all'.`

ZerBea, 2018-08-21:
Do you use latest version?
Do you use LINUX?

run
sudo make clean
sudo make install

output should look like this:
$ sudo make clean
rm -f hcxdumptool
rm -f *.o *~

$ sudo make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c -lpthread 
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxdumptool
rm -f *.o *~

pojda, 2018-08-22:
`$ make` Worked on a Debian box, but throws the same message on my Mac.

ZerBea, 2018-08-22:
Correct. ,aOS is not supported.
From README.md.
Requirements
* Operatingsystem: Arch Linux (strict), Kernel >= 4.14 (strict). It should work on other Linux systems (notebooks, desktops) and distributions, too (no support for other distributions). Don't use Kernel 4.4 (rt2x00 driver regression)
But it's not designed to run on maOS.
Closed this issue beacaue it's not hcxdumptool related.

closed 2018-08-22 by ZerBea

ZerBea, 2018-08-22:
hcxdumptool do not support macOS (hcxtools support macOS), maybe this will work for you (wifibroot):
https://www.shellvoide.com/wifi/how-to-crack-wifi-password-using-pmkid-new-method-wifibroot/

pojda, 2018-08-22:
Awesome! Will check it out. Thanks, @ZerBea!

ZerBea, 2018-09-03:
Do you use latest version of hcxdumptool?
What is the output of:
$ sudo iw dev wlan0 set type monitor
$ iw dev wlan0 info

geraldchecka, 2018-09-03:
The output is as follows:
`$ sudo iw dev wlan0 set type monitor`
command failed: Device or resource busy (-16)
`$ sudo iw dev wlan0 set type monitor`
Interface wlan0
	ifindex 3
	type managed
	wiphy 0

ZerBea, 2018-09-04:
Have you stopped all services that take access to the interface (like NetworkManager and wpa-supplicant)? hcxdumptool need full access to the interface!

help789, 2019-09-27:
[image]

help789, 2019-09-27:
sir after cloning the hcxdumptool i get this error after "make" command

ZerBea, 2019-09-28:
After cloning the repository you must(!) satisfy the dependencies, which are not(!) installed by KALI default settings.
From README.md:

Requirements
libopenssl and openssl-dev installed

From README.md hcxtools:
To install requirements on Kali use the following 'apt-get install libcurl4-openssl-dev libssl-dev zlib1g-dev'

So a:
apt-get install libcurl4-openssl-dev libssl-dev zlib1g-dev
should satisfy them.