ZerBea/hcxdumptool #32: Installation problem

Mohsinali14, 2018-09-11:
Cant install on Ubuntu Please help
<pre><font color="#8AE234"><b>mohsin@mohsin-HP-Pavilion-15-Notebook-PC</b></font>:<font color="#729FCF"><b>~/hcxdumptool-master</b></font>$ make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c -lpthread 
make: cc: Command not found
Makefile:29: recipe for target &apos;build&apos; failed
make: *** [build] Error 127
</pre>

ZerBea, 2018-09-11:
Your error message is: "make: cc: Command not found", which tells you that you are missing the C compiler. That is a configuration error of UBUNTU. Check your configuration/environment and/or install build-essential.
sudo apt-get install build-essential
Then try again.
read more here:
https://stackoverflow.com/questions/22557029/c-compiler-not-found-ubuntu
https://github.com/jwilm/alacritty/issues/1440

closed 2018-09-11 by ZerBea

kolet, 2018-09-15:
> Cant install on Ubuntu Please help
> 
> mohsin@mohsin-HP-Pavilion-15-Notebook-PC:~/hcxdumptool-master$ make install
> cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c -lpthread 
> make: cc: Command not found
> Makefile:29: recipe for target 'build' failed
> make: *** [build] Error 127

or just , sudo apt install gcc

kamesh413, 2019-08-08:
root@kali:~/hcxdumptool# make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c 
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c 
install -m 0755 -D hcxpioff /usr/local/bin/hcxpioff
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxpioff
rm -f hcxdumptool
rm -f *.o *~