ZerBea/hcxdumptool #33: Trouble Installing

Camer-n, 2018-09-30:
I'm quite new to Linux so it may be a simple fix but I just can't figure this out.

`camer@user:~/hcxdumptool$ make
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c -lpthread 
/usr/bin/ld: cannot open output file hcxdumptool: Is a directory
collect2: error: ld returned 1 exit status
Makefile:29: recipe for target 'build' failed
make: *** [build] Error 1`

strasharo, 2018-09-30:
There's a directory named 'hcxdumptool' within your current directory - remove it.

ZerBea, 2018-09-30:
As strasharo mentioned: clean up your installation directory
Or remove it complete and do a new
$ git clone https://github.com/ZerBea/hcxdumptool
Then cd into the source diretory
$ cd hcxdumptool
run
$ make
or to install
$ sudo make install

Read more here:
https://askubuntu.com/questions/18618/cannot-open-output-file-test-is-a-directory-collect2-ld-returned-1-exit-status

Closed this issue, because it isn't a hcxdumptool issue.

closed 2018-09-30 by ZerBea