ZerBea/hcxdumptool #36: Stuck!

frequencyg, 2018-10-19:
using a RT2080 usb card my attack  gets stuck to this state
https://imgur.com/a/V6qyQER Also,can someone tell me how to properly install my network drivers for this card?
Thank you!

ALBARAA78, 2018-10-29:
Same but with intel 8260

ZerBea, 2018-10-29:
The driver must support monitor mode and packet injection!
Most of the AC adapters doesn't support this features.

Read more here:
https://forums.kali.org/archive/index.php/t-37076.html
Hello everyone. I'm in the market for a new 'ac' chip. Currently I have this one (https://wikidevi.com/wiki/Intel_Dual_Band_Wireless-AC_8260_(8260NGW)) which seems to support monitor mode under "iwlwifi" but does not support packet injection properly.

Closed this because it is a driver issue.

closed 2018-10-29 by ZerBea

ALBARAA78, 2018-10-29:
about month ago on rtl8723ae i tested this and it was working and i was able to capture pmkid
i tested it today and it i got suck 
is it an update issue

ZerBea, 2018-10-29:
Is the driver updated, too?
From wikidevi:
https://wikidevi.com/wiki/Rtl8723ae
Supported modes
    STA (Station) mode: supported
    IBSS (Ad-Hoc) mode: supported
    AP (Master) mode: supported
    Mesh (802.11s) mode: supported
    P2P mode: supported
    Monitor mode: supported
    Packet injection: unknown

ZerBea, 2018-10-29:
What is the output of:
iw dev <your interface> info