ZerBea/hcxdumptool #37: hcxdumptool  not filter right mac address

PcBlackbelt, 2018-11-06:
heres  what my my capture looks like and the first two captures are from my  router the second two are not is this normal? anyway when i turn the  pmkid into hash and try to crack it.. hashcat gives me password from  wrong access point. 
hcxdumptool -o hashingit -i wlan0 --filterlist=tvfp2filer.txt --filtermode=2 --enable_status=1

the mac address that is suppose to filter is:  009fa9073914
the filter works seems like for a short period then starts grabbing from all access points in area
still havent been able to capture pmkid from 009fa9073914

the pmkid captureed belongs to 2c56dc54e238 using awus036nha. i have a awus036nh on the way to my house for testing.
capturing (stop with ctrl+c) 
INTERFACE:...............: wlan0 
FILTERLIST...............: 1 entries 
MAC CLIENT...............: fcc233dadfba 
MAC ACCESS POINT.........: b025aa99a8f8 (incremented on every new client) EAPOL TIMEOUT............: 150000 
REPLAYCOUNT..............: 63156 
ANONCE...................: 1f963f900448da8c1182e40ba288928877f40543759cd5e6425deb3548407868 [16:26:17 - 001] 009fa9073914 -> 5c93a20b3897 [FOUND AUTHORIZED HANDSHAKE, EAPOL TIMEOUT 2516] 
[16:40:15 - 001] 009fa9073914 -> ccfb65942f7e [FOUND AUTHORIZED HANDSHAKE, EAPOL TIMEOUT 2651] [17:19:39 - 001] b44bd20fd6ba -> 5c8fe0bf2984 [EAPOL 4/4 - M4 RETRY ATTACK] [17:39:53 - 011] 2c56dc54e238 -> 5ccf7f48a404 [FOUND PMKID] INFO: cha=11, rx=145350, rx(dropped)=3145, tx=2453, powned=3, err=0^C terminated... 

the filter is working somewhat atleast cuz i tried without filter and got slammed with tons more
--

strasharo, 2018-11-06:
https://github.com/ZerBea/hcxdumptool/commit/f57338faca30bec31bee509cf99104a471286816#diff-56f0554ece9939c694f65b93a8468553R4435

PcBlackbelt, 2018-11-06:
so am i doing it wrong by running: echo "009FA9073914">filter.txt
then running hcxdumptool -o hash -i wlan0mon --filterlist=filter.txt --filtermode=2 --enable_status 3

PcBlackbelt, 2018-11-07:
not sure what that strasharo but im trying to capture traffic from the ap in the filter list

ZerBea, 2018-11-07:
hcxpcaptool will convert handshakes and PMKIDs to hccapx and/or 16800 hashfiles.

Now you must delete the unwanted PMKIDs from your 16800 hashfiles.
Or you can retrieve the wanted handshakes from your hccapx file, by wlanhcx2ssid.

ZerBea, 2018-11-07:
009fa9073914 is your filter list entry.
You got 2 handshakes from this ssid:
[16:26:17 - 001] 009fa9073914 -> 5c93a20b3897 [FOUND AUTHORIZED HANDSHAKE, EAPOL TIMEOUT 2516]
[16:40:15 - 001] 009fa9073914 -> ccfb65942f7e [FOUND AUTHORIZED HANDSHAKE, EAPOL TIMEOUT 2651]
But the AP doesn't transmit a PMKID!!!!

closed 2018-11-07 by ZerBea

donbariii, 2019-04-13:
How much time does it takes to fetch the PMKID because I tried a lot but could not get the PMKID even after waiting for 30 minutes or more.

ZerBea, 2019-04-13:
If AP has enabled PMKID less than a minute running this options:
hcxdumptool -i INTERACE -o output.pcapng -t 5 --enable_status=1

To check if AP is in range run:
hcxdumptool -i INTERACE --do_rcascan

Driver of your interface must support monitor mode and packet injection.

donbariii, 2019-04-14:
If AP is not in range what should I do?

ZerBea, 2019-04-14:
To answer this question we need an information about your equipment (WLAN device), because:
1. case: device doesn't support full packet injection (get a supported device)
2. case: AP is far away (you need a better antenna or get closer to the AP)
3. case: hardware is broken (antenna connector)

If you receive PMKIDs from other APs, but not from your target, target hasn't active this feature

I assume there are now warnings if you start hcxdumptool. In that case please add output of
$ hcxdumptool -I

$ lsusb if you have an USB device)
or
$ lspci if you have an internal device)

BTW:
Not all drivers support monitor mode.
Not all drivers that support monitor mode, support packet injection, too. Intel drivers are well known for this.

donbariii, 2019-04-14:
That mean using only laptop without wireless adapter cannot do PMKID
cracking? What wireless adapter would you recommend me for PMKID cracking
to buy which would support both monitor mode and packet injection?

ZerBea, 2019-04-14:
As I wrote, it depends on several factors.  If the AP doesn't run PMKID features, this attack vector will not work.
But you can improve your equipment:
USB WLAN device with external antenna connector and a high gain antenna
Driver must support monitor mode and full packet injection. Read more here:
https://wikidevi.com
This drivers are known as working out ouf the box:
kernel driver mt7601ut:
ID 7392:7710 Edimax Technology Co., Ltd
ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter

kernel driver rt2800usb:
ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter

They are part of the kernel. Do not buy a device that require to install a third party driver!

I do not recommend a special device by name, because VENDORs very often change the chipset, So a device V1 will work, while V2 dosn't work.

donbariii, 2019-04-15:
Can you recommend me a wireless adapter V1 with vendor name from amazon
please. With that how can we check out whether our target mac's AP has
PMKID enabled or not from our machine?

ZerBea, 2019-04-15:
If you run
$ hcxdumptool -i INTERFACE --do_rcascan
and you get a message AP IN RANGE, you know, that the AP receive our request and respond to it.
if you run
$ hcxdumptool -i INTERFACE --enable_status=1
and you receive a PMKID, you know that the AP is vulnerable against this attack.

My favorite adapters are:
EDIMAX EW-7711UAN (ID 7392:7710 Edimax Technology Co., Ltd)
ALLNET ALLNET ALL-WA0150N (ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter)
TENDA W311U+ (ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter)

donbariii, 2019-04-16:
Thank you for your wonderful responses clearing my doubts.

ZerBea, 2019-04-16:
Updated hcxdumptool device wiki and README.md.
The listed devices are (still) working on latest kernel. They are not affected by xhci bug.

donbariii, 2019-04-23:
This is the only adapter available for shipment to India. Would it serve
the purpose?

donbariii, 2019-04-23:
I found this adaptor, TP-Link TL-WN722N 150Mbps Wireless USB Adapter whose
Version is V1.X. Would it work for Monitor mode and packet injection?

strasharo, 2019-04-23:
Yes, 1.x has Atheros chipset and works fine with hcxdumptool.

donbariii, 2019-04-24:
Unfortunately what I received wireless adapter is of version 3.0. Sir,
would version 3.0 support the minister mode and packet injections.

ZerBea, 2019-04-27:
No. Only v1 of this device is working.
Manufacturers do change chipsets without changing model numbers. Sometimes they add (v)ersion or (rev)vision.

donbariii, 2019-05-18:
Panda Wireless PAU09 N600 Dual Band (2.4GHz and 5GHz) Wireless N USB
Adapter W/Dual 5dBi Antennas - Windows XP/Vista/7/8/8.1/10, Mint, Ubuntu,
openSUSE, Fedora, Centos, Kali Linux and Raspbian. Will this wireless
adapter support monitor mode and packets injection?

ZerBea, 2019-05-19:
That depends on the chipset. It looks like there are different versions.
https://www.wirelesshack.org/review-panda-wireless-pau09-n600-dual-band-wifi-dongle.html
https://www.phoronix.com/scan.php?page=article&item=panda-pau09-linux&num=1

donbariii, 2019-05-24:
Will Alfa AWUS036NHA support monitor mode and packet injection. I have
already bought one and tried but could not get the PMKID.

ZerBea, 2019-05-24:
$ hcxdumptool -i interface --enable_status=1 -o capture.pcapng

ZerBea, 2019-05-24:
To find out if AP's are in range:
$ hcxdumptool -i interface --do_rcascan

donbariii, 2019-05-25:
I have tried to crack my android phone's hotspot password by using PMKID
capture method but am not able to capture the PMKID. Will this method not
work in android hotspot password cracking?

ZerBea, 2019-05-25:
Not all access points use this method. On some access points this feature is activated by default - on others not, because it absolutely make no sense for a single hotspot (like your Android phone) to run roaming features.

https://mrncciew.com/2014/09/11/cwsp-pmk-caching-preauthentication/

https://documentation.meraki.com/MR/WiFi_Basics_and_Best_Practices/Pairwise_Master_Key_and_Opportunistic_Key_Caching_-_PMK_and_OKC

donbariii, 2019-05-27:
After getting the PMKID found, I have tried to crack it using hashcat but
comes with an error saying failed to get magic number from .pcapng . What
is it? Please help me

ZerBea, 2019-05-27:
Which command lin did you use?

Correct is for older hashcat:
$ haxpcaptool -z pmkids.16800 capture.pcapng
or for latest hashcat git:
$ haxpcaptool -k pmkids.16800 capture.pcapng

If you get an error like "failed to get magic number", your pcapng file is destroyed.

Please attach that pcapng file, to see what went wrong

donbariii, 2019-07-20:
Would be grateful if sir would help me out with commands to bypass public
wifi login portal in Kali Linux using Alfa adapter.

ZerBea, 2019-07-21:
That is neither part of hcxdumptool, hcxtools, hashcat nor part of john. Also I do not use KALI Linux.
So you can't expect an answer here. I suggest you, to ask this question in KALI forum.