ZerBea/hcxdumptool #38: why hcxdumptool scan all network

rusamer, 2018-11-07:
why hcxdumptool scan all network even if i select filter.txt ?

is it possible to scan only filter ?

thx

ZerBea, 2018-11-07:
Because it is hcxdumptool policy:
- receive everything that is sent by APs or CLIENTs
- use filter list an filter mode to interact with entries or to prevent interaction with entries
- use hcxtools to retrieve and select handshakes from captured traffic

It will take too much time to filter all incomming traffic.
It will take not so much time to filter outgoing traffic.

ZerBea, 2018-11-07:
Also keep in mind:
The filter list works on APs and(!) CLIENTs.

ZerBea, 2018-11-07:
Since many user asked for this feature, I added the new filter mode option 3:
--filtermode=<digit>               : mode for filter list
                                     1: use filter list as protection list (default) in transmission branch
                                        receive everything, interact with all APs and CLIENTs in range,
                                        except(!) the ones from the filter list
                                     2: use filter list as target list in transmission branch
                                        receive everything, only interact with APs and CLIENTs in range,
                                        from the filter list
                                     3: use filter list as target list in receiving branch
                                        only receive APs and CLIENTs in range,
                                        from the filter list

commit:
https://github.com/ZerBea/hcxdumptool/commit/7d71b77b688b7cd59c2388109a99111d101e79d3

ZerBea, 2018-11-07:
Please test and close issue if it works for you.

closed 2018-11-12 by ZerBea