ZerBea/hcxdumptool #41: Will not work with USB wifi device based on RTL8192 chipset

EoinBullock, 2018-12-29:
I have tried many different guides to try and get this tool to work with the RTL8192 cu chipset, but no luck. I do have this dongle working with airmon and airgeddon no problem.

My setup is RPi 3 B7+
Linux Raspberry Pi 4.14.90-v

When I run sudo hcxdumptool -wlan0 -o ....... --enable_status=2 , I get:

initialization....
interface is not up
failed to init socket

ZerBea, 2019-01-02:
Driver doesn't support full monitor mode/packet injection:
From here: https://wikidevi.com/wiki/Rtl8192cu
Supported modes
    STA (Station) mode: supported
    IBSS (Ad-Hoc) mode: unknown
    AP (Master) mode: unknown
    Mesh (802.11s) mode: unknown
    P2P mode: unknown
    Monitor mode: supported, no data packets?
    Packet injection: unknown

and here:
https://forums.kali.org/showthread.php?20867-Why-not-work-well-REALTEK-rtl8192cu-USB-wireless

and here:
https://www.quora.com/The-TP-Link-support-responded-to-my-mail-and-says-that-the-TL-WN822N-wireless-adapter-does-not-support-monitor-mode-and-package-injection-Is-it-true

and here:
https://forum.aircrack-ng.org/index.php?topic=1017.0

ZerBea, 2019-01-04:
Closed this issue, because it is distribution and/or driver related:

https://github.com/raspberrypi/linux/issues/369
https://bugs.launchpad.net/ubuntu/+source/linux-raspi2/+bug/1577421

closed 2019-01-04 by ZerBea