ZerBea/hcxdumptool #44: nexmon support

JuniorJPDJ, 2019-02-18:
Why can't I use nexmon patched firmware to dump?

```
$ LD_PRELOAD="./libfakeioctl.so" ./hcxdumptool -i wlan0mon -o hcx_fun
initialization...
warning: wlan0mon is probably a monitor interface
failed to set monitor mode: Not supported
failed to init socket
```

nexmon: https://github.com/seemoo-lab/nexmon

ZerBea, 2019-02-18:
Did you activate nexmon monitor?
https://github.com/seemoo-lab/nexmon
Using the Monitor Mode patch
    Thanks to the prior work of Mame82, you can setup a new monitor mode interface by executing: iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
    To activate monitor mode in the firmware, simply set the interface up: ifconfig mon0 up.
    At this point, monitor mode is active. There is no need to call airmon-ng.
    The interface already set the Radiotap header, therefore, tools like tcpdump or airodump-ng can be used out of the box: tcpdump -i mon0

JuniorJPDJ, 2019-02-18:
This modified driver for Raspberry Pi is not working on phones as far as I know, I'm using libfakeioctl in LD_PRELOAD to simulate monitor mode on non-monitor-mode capable interface.
This way I can make it work with bettercap.

ZerBea, 2019-02-19:
You got 2 error messages from hcxdumptool:
failed to set monitor mode: No such device -> hcxdumptool can't find the interface
failed to init socket -> hcxdumptool can't create a raw socket to receive and transmit raw packets
Error 2 is related to error 1. If we have no interface, we can't send or receive packets .

So let's see if monitor mode is working and iw is able to detect the device. Please set the interface to monitor mode: 
iw dev <interface> set type monitor
and comment the result:
sudo iw dev <interface> info

If everything is ok, we expect no error message running this command:
iw dev wlp3s0f0u1 set type monitor
and we expect an output like this for the second command:
Interface wlp3s0f0u1
	ifindex 3
	wdev 0x1
	addr c8:3a:35:e0:f4:52
	type monitor
	wiphy 0
	txpower 20.00 dBm

JuniorJPDJ, 2019-02-23:
nexmon doesn't allow me to set interface to monitor mode, it's emulating it by preloaded library
It's not modifying driver to allow monitor mode, it's modifying just wifi firmware and emulating monitor for apps using preloaded library
iw doesn't show the interface as monitor mode even with preloading library

ZerBea, 2019-02-23:
In that case hcxdumptool will not work for you, because the emulator doesn't allow this:
"nexmon doesn't allow me to set interface to monitor mode"

$ hcxdumptool --help
         do not run hcxdumptool on logical interfaces (monx, wlanxmon)
         do not use hcxdumptool in combination with other 3rd party tools, which take access to the interface

Closed this issue, because it is related to the emulator.

closed 2019-02-23 by ZerBea

ZerBea, 2019-02-23:
Please read also all comments here:
https://github.com/ZerBea/hcxdumptool/issues/42
to understand  that the driver (in your case the emulator) must support full and hardware near monitor mode, included raw packet injection.
Only if the driver supports this, hcxdumptool will work.

JuniorJPDJ, 2019-02-23:
It supports monitor mode and packet injection, but doesn't report it.
Ioctls are overwritten on preloaded library.
To set firmware to monitor mode you need to use custom software, not linux API.
I commented section that sets interface to monitor mode and it ALMOST works.
It captures packets with NEXMON header, which should be handled by preloaded library and passed to hcxdumptool as normal RADIOTAP.

ZerBea, 2019-02-24:
The issue is within the preloaded library:
__nex_driver_io: error

and hcxdudmptool reports correctly:
failed to set monitor mode: No such device
failed to init socket

ioctl(3, SIOCSIWMODE, 0xb6fa0e04)       = -1 EOPNOTSUPP (Not supported)
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 4
ioctl(4, SIOCDEVPRIVATE, 0xb6f76110)    = -1 ENODEV (No such device)
writev(1, [{iov_base="__nex_driver_io", iov_len=15}, {iov_base=": error\n", iov_len=8}], 2) = 23
...
ioctl(3, SIOCSIWMODE, 0xb6fa01ec)       = -1 EOPNOTSUPP (Not supported)
socket(AF_INET, SOCK_DGRAM, IPPROTO_IP) = 4
ioctl(4, SIOCDEVPRIVATE, 0xb6f76110)    = -1 ENODEV (No such device)
writev(1, [{iov_base="__nex_driver_io", iov_len=15}, {iov_base=": error\n", iov_len=8}], 2) = 23

hcxdumptool needs the ioctls, otherwise it will not work:

start capturing (stop with ctrl+c)
INTERFACE:...............: wlp3s0f0u1
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries
MAC CLIENT...............: b0ece1f25217
MAC ACCESS POINT.........: 0084edce9fef (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 64046
ANONCE...................: 60f6f16c3028e0d9ed6450d82b176c7be213d7a4162e64f486638b4b8bb474ec

INFO: cha=6, rx=0, rx(dropped)=0, tx=30, powned=0, err=0