ZerBea/hcxdumptool #45: Wrong timestamp

ovakisan, 2019-02-26:
Hi,
I've tried on two different rpie's with raspbian images and they have same timestamp with weird values. Debian VM gives normal epoch time.
Date/time is set up correctly.

(This is opened with wireshark)
```
Frame 1: 226 bytes on wire (1808 bits), 226 bytes captured (1808 bits) on interface 0
    Interface id: 0 (wlan1)
    Encapsulation type: IEEE 802.11 plus radiotap radio header (23)
    Arrival Time: Jan  1, 1970 01:25:10.864502000 Romance Standard Time
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1510.864502000 seconds
    [Time delta from previous captured frame: 0.000000000 seconds]
    [Time delta from previous displayed frame: 0.000000000 seconds]
    [Time since reference or first frame: 0.000000000 seconds]
    Frame Number: 1
    Frame Length: 226 bytes (1808 bits)
    Capture Length: 226 bytes (1808 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: radiotap:wlan_radio:wlan]
```
In bigger capture file I noticed that starting packets have epoch values 1000/2000 and then on frame 255 it jumps to (-) value, like seen below. Later packets switch back to 1000/2000 and to (-) value again. That was on approx. 1hr long capture file.
```
Frame 255: 106 bytes on wire (848 bits), 106 bytes captured (848 bits) on interface 0
    Interface id: 0 (wlan0)
    Encapsulation type: IEEE 802.11 plus radiotap radio header (23)
    Arrival Time: Not representable
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: -140464757.873059000 seconds
    [Time delta from previous captured frame: -140466901.308239000 seconds]
    [Time delta from previous displayed frame: -140466901.308239000 seconds]
    [Time since reference or first frame: -140466833.115197000 seconds]
    Frame Number: 255
    Frame Length: 106 bytes (848 bits)
    Capture Length: 106 bytes (848 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: radiotap:wlan_radio:wlan]
```

I don't quite understand the code and I'm not really sure what to check next.
Any ideas?
Thx

ZerBea, 2019-02-26:
That was a really ugly issue. Thank you for reporting it.
It should be fixed by this commit:
https://github.com/ZerBea/hcxdumptool/commit/a911c15698efd52292120c467a6281f2d2ec2800
Frame 1: 293 bytes on wire (2344 bits), 293 bytes captured (2344 bits) on interface 0
    Interface id: 0 (wlan0)
    Encapsulation type: IEEE 802.11 plus radiotap radio header (23)
    Arrival Time: Feb 26, 2019 17:17:57.698497000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1551197877.698497000 seconds
    [Time delta from previous captured frame: 0.000000000 seconds]
    [Time delta from previous displayed frame: 0.000000000 seconds]
    [Time since reference or first frame: 0.000000000 seconds]
    Frame Number: 1
    Frame Length: 293 bytes (2344 bits)
    Capture Length: 293 bytes (2344 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: radiotap:wlan_radio:wlan]
Radiotap Header v0, Length 18
802.11 radio information
IEEE 802.11 Beacon frame, Flags: ........
IEEE 802.11 wireless LAN

Please test and let me know if it works for you. In that case I'll move hcxdumptool and hcxtools to 5.1.3

ovakisan, 2019-02-26:
Ok, from a quick test capture it looks like it is fixed!
Thanks for quick response/fix!

closed 2019-02-26 by ovakisan

ZerBea, 2019-02-26:
Great. Fixed also hcxhash2cap and hcxpcaptool and moved hcxdumptool and hcxtools up to v5.1.3

anjumaltaf, 2022-12-20:
hi,
I have a question like one asked above.

Q. The time column in Wireshark displays 0.000000 for some packets (Seconds since previous captured packet). My students asked me why it happens and unfortunately I couldn't answer them.

What's the reason? Guide me in very simple way (as I am a novice in Wireshark).

119	0.000393	142.250.181.35	192.168.2.157	UDP	77	63194 → 443 Len=35
120	0.000902	192.168.2.157	142.250.181.35	UDP	1292	443 → 63194 Len=1250
**121	0.000000**	192.168.2.157	142.250.181.35	UDP	1292	443 → 63194 Len=1250
122	0.000272	192.168.2.157	142.250.181.35	UDP	1292	443 → 63194 Len=1250
**123	0.000000**	192.168.2.157	142.250.181.35	UDP	1292	443 → 63194 Len=1250
**124	0.000000**	192.168.2.157	142.250.181.35	UDP	1292	443 → 63194 Len=1250

Thanks in Advance
Anjum Altaf

ZerBea, 2022-12-20:
How the timestamp works is OS dependent:
https://wiki.wireshark.org/Timestamps
https://www.wireshark.org/docs/wsug_html_chunked/ChAdvTimestamps.html
https://www.wireshark.org/docs/wsug_html_chunked/ChWorkTimeFormatsSection.html

The timestamp presentation format and the precision in the packet list can be chosen using the View menu of Wireshark:
https://www.wireshark.org/docs/wsug_html_chunked/ChWorkTimeFormatsSection.html

BTW:
hcxdumptool/hcxlabtool use:
```
gettimeofday(&tv, NULL);
timestamp = ((uint64_t)tv.tv_sec *1000000) +tv.tv_usec -512;
```