ZerBea/hcxdumptool #46: EAPOL timeout is low cannot convert after capture

mysterdee, 2019-02-27:
Hi, i get this error with hcxdumptool  when i capture the PMKID.

i capture the PMKID no problem, FOUND PMKID

but when i try to convet the capture file that i -o utput in hcxdumptool

i get this error

EAPOL timout is to low

So i cannot convert the file for hashcat

Any ideas why im getting this error any1

is it a common error with hcxdumptool

thanks for any advice...

i capture with this code.

hcxdumptool -i wlan0mon -o pmkid.pcapng --enable_status=1

get resuts
[08:50:37 - 006] 002417bdb675 -> d8cf9c805f44 [FOUND PMKID]

the convert code i use is as follows after i capture the pmkid

hcxdumptool -E essidlist -I identitulist -U usernamelist -z capturedthis.16800 pmkid.pcapng

then i get the error

EAPOL timeout is low

ZerBea, 2019-02-27:
There are several mistakes your commandlines:
1) hcxdumptool -i wlan0mon -o pmkid.pcapng --enable_status=1
do not run hcxdumptool in combination with other tools like airmon-ng
hcxdumptool use its own monitor mode!

2) hcxdumptool -E essidlist -I identitulist -U usernamelist -z capturedthis.16800 pmkid.pcapng
you run hcxdumptool again to convert the pcapng file und use hcxpcatools options. That can't work
To convert the pcapng use this command:
$ hcxpcaptool -E essidlist -I identitylist -U usernamelist -z capturedthis.16800 pmkid.pcapng

-E is only needed if you assume password is in WLAN traffic
-I, -U is only needed if you're attacking EAP and/or RADIUS

Now you have several options to recover the password like this ones for example:
$ hcxwltool -i essidlist --straight > testlist.tmp
$ hcxwltool -i essidlist --digit >> testlist.tmp
$ hcxwltool -i essidlist --xdigit >> testlist.tmp
$  cat testlist.tmp | sort | uniq | hashcat -m 16800  capturedthis.16800

or:
$ hcxpsktool --weakpass --eudate --usdate -z capturedthis.16800 | sort | uniq | hashcat -m 16800 capturedthis.16800

correct chain:
hcxdumptool -> hcxpcaptool -> hashcat 
or
hcxdumptool -> hcxpcaptool -> john

Closed, because it isn't a hcxdumptool/hcxtools issue.

closed 2019-02-27 by ZerBea

mysterdee, 2019-02-27:
bloomin eck, really sorry, noobs mistake i guess..

ZerBea, 2019-02-27:
No problem, that can happen.