ZerBea/hcxdumptool #47: gpsd data invalid

ovakisan, 2019-03-01:
How do you process gpsd data?
I have a problem where if i run hcxdumptool on location where no fix is available then even if i get to a location where GPS acquires fix, coordinates comment will stay on value 0 for all packets captured after the gps acquired fix. 

Same is valid if you do other way around, you start with gps fixed and then loose fix, all packets will continue to use last fixed coordinate even if gps acquired fix again in mean time.

ZerBea, 2019-03-01:
Right now, hcxdumptool ignore gps fix. We check the received json string from the device and grep for lat, long and alt:
sscanf(gpsdptr +6, "%Lf", &lat);
sscanf(gpsdptr +6, "%Lf", &lon);
sscanf(gpsdptr +6, "%Lf", &alt);

Please attach an example pcapng and output of gpspipe -r

ZerBea, 2019-03-02:
Also make sure, GPSD daemon is running and working like expected:
http://www.catb.org/gpsd/troubleshooting.html#gpstroubleshooting
We are running this protocol:
http://www.catb.org/gpsd/gpsd_json.html
and looking for this protocol values:
char *gpsd_enable_json = "?WATCH={\"json\":true}";
char *gpsd_disable = "?WATCH={\"enable\":false}";
char *gpsd_version = "\"proto_major\":3";
char *gpsd_json = "\"json\":true";
char *gpsd_tpv = "\"class\":\"TPV\"";

as described here:
http://www.catb.org/gpsd/gpsd_json.html

Here's an example:
{"class":"TPV","device":"/dev/pts/1",
    "time":"2005-06-08T10:34:48.283Z","ept":0.005,
    "lat":46.498293369,"lon":7.567411672,"alt":1343.127,
    "eph":36.000,"epv":32.321,
    "track":10.3788,"speed":0.091,"climb":-0.085,"mode":3}

Then we grep for lat, lon and alt.

ZerBea, 2019-03-02:
How do you start gpsd?

Here is an example for AktivePilot JENTRO BT-GPS-8 bluetooth connection:
1) start bluetooth service: $ systemctl start bluetooth
2) start bluethooth control, pair and trust device: $ bluetoothctl
3) run rfcomm and bind device:  $ rfcomm bind /dev/rfcomm0 <device mac>
4) run gpsd: $ gpsd -N -n -b -D4 /dev/rfcomm0
5) retrieve gps data: $ gpspipe -r

ovakisan, 2019-03-04:
> Also make sure, GPSD daemon is running and working like expected:
> http://www.catb.org/gpsd/troubleshooting.html#gpstroubleshooting
> We are running this protocol:
> http://www.catb.org/gpsd/gpsd_json.html
> and looking for this protocol values:
> char *gpsd_enable_json = "?WATCH={"json":true}";
> char *gpsd_disable = "?WATCH={"enable":false}";
> char *gpsd_version = ""proto_major":3";
> char *gpsd_json = ""json":true";
> char *gpsd_tpv = ""class":"TPV"";
> 
> as described here:
> http://www.catb.org/gpsd/gpsd_json.html
> 
> Here's an example:
> {"class":"TPV","device":"/dev/pts/1",
> "time":"2005-06-08T10:34:48.283Z","ept":0.005,
> "lat":46.498293369,"lon":7.567411672,"alt":1343.127,
> "eph":36.000,"epv":32.321,
> "track":10.3788,"speed":0.091,"climb":-0.085,"mode":3}
> 
> Within the main packet loop, we ask for data from the interface (line 3494 - 3507). If this data are valid, we use them (grep for lat, lon and alt and write them to pacpng comment field for example line 661: epblen += addoption(epb +epblen, SHB_COMMENT, gpsdlen, gpsdatabuffer);).
> If we didn't receive data from the gps device, we use the last correct received ones.
> 
> You can verify this, if you run gpspipe in parallel to hcxdumptool. Then compare the pcapng comment fields with the output of gpspipe. The values should be the same for "class":"TPV" fields.

I did a quick capture with hcxdumptool, gpspipe and cgps. It is doing as you say, when it loses fix it continues to use last valid coordinate, gaining fix again starts to pull new valid coordinate. 
This was all on one location, I will repeat it again while moving.

I am starting gpsd with: $ gpsd -N -n  /dev/ttyUSB0
All the gps clients do get the coordinates including hcxdumptool. 

Looking at all this, things should work. I have to revisit the whole setup it might be i have some antenna issues.

Thanks for whole explanation, code started to make sense after some time spent looking at it (i'm trying to get more into C)

One suggestion regarding using last valid coordinate when you lose fix. It would be good to maybe append "NOFIX" to coordinates so you know which packets are on wrong position.

ZerBea, 2019-03-05:
Good idea. I'll check if I can add NOFIX. Please do not close this issue, until I find a solution.

ZerBea, 2019-03-06:
Added GPS date and GPS time to status and pcapng comment field by latest commit:
https://github.com/ZerBea/hcxdumptool/commit/dafe0a6a2d209514775f4504228598ba59d21638
Now, you can use this values to determine when the last valid GPS coordinate was received:
 sudo hcxdumptool -i wlp3s0f0u1 --use_gpsd -o test.pcapng
initialization...
connecting to GPSD...
waiting up to 5 seconds to retrieve first position
GPSD activated

start capturing (stop with ctrl+c)
GPS LATITUDE.............: 49.126403
GPS LONGITUDE............: 4.626175
GPS ALTITUDE.............: 129.500000
GPS DATE.................: 06.03.2019
GPS TIME.................: 21:52:41
INTERFACE................: wlp3s0f0u1
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries
MAC CLIENT...............: f04f7c89dabb
MAC ACCESS POINT.........: 980ee432604d (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 61455
ANONCE...................: 182972399cd2e65deb7941601cca14b644681c092dcf6f704935c7f3d2eaceea

INFO: cha=11, rx=7080, rx(dropped)=1676, tx=408, powned=10, err=0, lat=49.126342, lon=4.626268, alt=129.500000, gpsdate=06.03.2019, gpstime=22:01:55
terminated...

Wireshark and tshark are able to retrieve the values:
$ tshark -r test.pcapng-0 -Y frame.comment -T fields -E header=y -e frame.number -e frame.time -e wlan.sa -e frame.comment
172	Mar  6, 2019 23:01:48.793212000 CET	1a:f8:7c:91:24:a3	lat:49.126337,lon:4.626268,alt:129.500000,date:06.03.2019,time:22:01:48

ZerBea, 2019-03-07:
added debug code in main packet loop [processpackets()]
to print raw GPS data, received from GPSD:
/*
			printf("\ndebug: %s\n", gpsddata);
*/
uncomment this before you run make to retrieve this values.

ZerBea, 2019-03-10:
I did several tests in combination with GPSD.

This is the GPS sentence, which we receive from GPSD:
{"class":"TPV","device":"/dev/rfcomm0","mode":3,"time":"2019-03-10T08:16:24.000Z","ept":0.005,"lat":49.156245000,"lon":5.626160000,"alt":449.500,"track":51.1500,"speed":0.041,"climb":0.000}

We receive this sentence every second. Here is the next one
{"class":"TPV","device":"/dev/rfcomm0","mode":3,"time":"2019-03-10T08:16:25.000Z","ept":0.005,"lat":49.146245000,"lon":5.626160000,"alt":449.500,"track":34.8100,"speed":0.041,"climb":0.000}

If the device loose the GPS Signal, GPSD will transmit no(!) sentence until it get the next  valid signal.
I added the time stamp from the sentence received from GPSD to the pcapng comment field, but unfortunately I can't verify that, if the hw clock of the system on which is hcxdumptool running (for example Raspberry Pi) is not in sync with the GPS signal.
I will say, I cant do this by automatic within hcxdumptool code, but you can compare pcapng timecode with GPS timecode in pcapng comment field to determine that the GPS position is correct, or not.

So we can close this issue, because it is related to GPSD. There is nothing inside the GPS sentence to work on, except the timecode (which we store into comment field).

closed 2019-03-10 by ZerBea

katzenheinz, 2022-08-19:
Hi ZerBea,

first of all: thank you for your work and effort spent on this great project (aswell as sharing it) - I like using this tool very much. :)

Unfortunately I still got problems using gpsd in combination with an android phone running the gpsd client app.

Here is an example for the response I get from the phone using gpsd:

`{"class":"TPV","device":"udp://0.0.0.0:9001","mode":3,"time":"<TIME>","ept":<FLOAT>,"lat":<FLOAT>,"lon":<FLOAT>,"altHAE":<FLOAT>,"altMSL":<FLOAT>,"alt":<FLOAT>,"epx":<FLOAT>,"epy":<FLOAT>,"epv":<FLOAT>,"magvar":<FLOAT>,"speed":0.000,"climb":<FLOAT>,"eps":<FLOAT>,"epc":<FLOAT>,"geoidSep":<FLOAT>,"eph":<FLOAT>,"sep":<FLOAT>}`

airodump-ng using switch "--gpsd" works fine, also cgps or gpsmon shows me valid data.

I start hcxdumptool with the "--use_gpsd" switch. I see hcxdumptool trying to connect, but somehow it fails doing so.

```
connecting GPSD...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
```

I start gpsd with "gpsd -N -n udp://0.0.0.0:9001", I can see, that the gpsd server interface runs on 127.0.0.1 on port 2947 using netstat -tuna.

Is this a bug in hcxdumptool or am I doing something wrong?

ZerBea, 2022-08-21:
Neither a bug nor you are doing something wrong. 
hcxdumptool use NMEA 183 protocol, whiich must be provided by your device,

From --help:
```
--use_gpsd                         : use GPSD device
                                     NMEA 0183 $GPGGA, $GPRMC
```

Example of NMEA 183:
```
$GPGGA,092750.000,5321.6802,N,00630.3372,W,1,8,1.03,61.7,M,55.2,M,,*76
$GPGSA,A,3,10,07,05,02,29,04,08,13,,,,,1.72,1.03,1.38*0A
$GPGSV,3,1,11,10,63,137,17,07,61,098,15,05,59,290,20,08,54,157,30*70
$GPGSV,3,2,11,02,39,223,19,13,28,070,17,26,23,252,,04,14,186,14*79
$GPGSV,3,3,11,29,09,301,24,16,09,020,,36,,,*76
$GPRMC,092750.000,A,5321.6802,N,00630.3372,W,0.02,31.66,280511,,,A*43
$GPGGA,092751.000,5321.6802,N,00630.3371,W,1,8,1.03,61.7,M,55.3,M,,*75
$GPGSA,A,3,10,07,05,02,29,04,08,13,,,,,1.72,1.03,1.38*0A
$GPGSV,3,1,11,10,63,137,17,07,61,098,15,05,59,290,20,08,54,157,30*70
$GPGSV,3,2,11,02,39,223,16,13,28,070,17,26,23,252,,04,14,186,15*77
$GPGSV,3,3,11,29,09,301,24,16,09,020,,36,,,*76
$GPRMC,092751.000,A,5321.6802,N,00630.3371,W,0.06,31.66,280511,,,A*45
```
https://en.wikipedia.org/wiki/NMEA_0183
https://gpsd.io/replacing-nmea.html

json is not used by hcxdumptool:
```
{"class":"TPV","device":"/dev/pts/1",
    "time":"2005-06-08T10:34:48.283Z","ept":0.005,
    "lat":46.498293369,"lon":7.567411672,"alt":1343.127,
    "eph":36.000,"epv":32.321,
    "track":10.3788,"speed":0.091,"climb":-0.085,"mode":3}
```
https://gpsd.io/gpsd_json.html

Set your GPS hardware to speak NMEA, as gpsd is guaranteed to be able to process that.
https://gpsd.gitlab.io/gpsd/gpsd.html

ZerBea, 2022-08-22:
Major reason to use NMEA 183 instead of JSON = less overhead!
`$GPGGA,092750.000,5321.6802,N,00630.3372,W,1,8,1.03,61.7,M,55.2,M,,*76`
versus
`{"class":"TPV","device":"/dev/pts/1","time":"2005-06-08T10:34:48.283Z","ept":0.005,"lat":46.498293369,"lon":7.567411672,"alt":1343.127,"eph":36.000,"epv":32.321,"track":10.3788,"speed":0.091,"climb":-0.085,"mode":3}
`