ZerBea/hcxdumptool #48: unable to set channel with rtl8188eu

respz, 2019-03-08:
I am using kali and was able to set the adapter to monitor mode using kimocoder driver (https://github.com/kimocoder/rtl8188eus). I can use airodump to receive packets but it is not working using hcxdumptool.
I only get this as a result:
```
# hcxdumptool -i wlan0 -o output.pcapng -c 1,6,11 --enable_status=1
initialization...
warning: unable to set channel 1 (removed this channel from scan list)
warning: unable to set channel 6 (removed this channel from scan list)
warning: unable to set channel 11 (removed this channel from scan list)
no available channel found in scan list

terminated...
```

Packet injection with aireplay is also working
```
# aireplay-ng -9 wlan0
15:37:11  Trying broadcast probe requests...
15:37:13  No Answer...
15:37:13  Found 2 APs

15:37:13  Trying directed probe requests...
15:37:13  XX:XX:XX:XX:XX:XX - channel: 11 - 'HOST1'
15:37:13  Ping (min/avg/max): 2.717ms/10.364ms/24.075ms Power: -41.67
15:37:13  30/30: 100%

15:37:13  Injection is working!

15:37:13  XX:XX:XX:XX:XX:XX - channel: 11 - 'HOST2'
15:37:13  Ping (min/avg/max): 2.921ms/9.882ms/22.017ms Power: -40.87
15:37:13  30/30: 100%
```

Here are the commands I've run:
```
# lsusb
Bus 001 Device 002: ID 2001:3310 D-Link Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 80ee:0021 VirtualBox USB Tablet
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

```
# ip link set wlan0 down
# iw dev wlan0 set type monitor
# ip link set wlan0 up
# iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr c2:67:84:6f:42:9c
		type monitor
		txpower 13.00 dBm
```

```
# hcxdumptool -I
wlan interfaces:
1062eb30659b wlan0 (rtl8188eu)
warning: NetworkManager is running with pid 441
warning: wpa_supplicant is running with pid 799
```

```
# kill 441 799
# hcxdumptool -I
wlan interfaces:
1062eb30659b wlan0 (rtl8188eu)
```

ZerBea, 2019-03-08:
It looks like the driver doesn't like the way hcxdumptool set the channel by SIOCSIWFREQ.
Please run:
$ iw dev wlan0 set channel 6
followed by
$ iw dev
and post the output.

ZerBea, 2019-03-08:
It seems to be a known issue on that dongles:
https://github.com/lwfinger/rtl8188eu/issues/203

ZerBea, 2019-03-08:
Also please read this notice here:
https://github.com/aircrack-ng/rtl8188eus

MONITOR MODE ATTENTION
There is a bug or interferrence with the driver and/or Network-Manager, so if you wan't to use the monitor mode, DON'T kill the network-manager or use airmon-ng (infact)

ZerBea, 2019-03-08:
Pushed an update:
https://github.com/ZerBea/hcxdumptool/commit/dd8925f0640c832ae80cf28dbc853aa50f91029e
For example:
Now hcxdumptool print the error message received from the driver inside the brackets
$ hcxdumptool -i wlp3s0f0u1 -c 14
initialization...
warning: unable to set channel 14 (Invalid argument) - removed this channel from scan list

In this case the user tried to set channel 14. The driver doesn't support this
and responds with: Invalid argument

Please update and post the error message - I'm sure it is a driver issue.

respz, 2019-03-09:
I did not know those git entries you posted and the networkmanager.conf change. After changing the conf file and rebooting I tried again. Here it goes:
```
# iw dev wlan0 set channel 6
# iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 10:62:eb:30:65:9b
		type managed
		txpower 13.00 dBm
```

Then, I updated sources, compiled and I run:
```
# git pull
remote: Enumerating objects: 14, done.
remote: Counting objects: 100% (14/14), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 14 (delta 7), reused 11 (delta 4), pack-reused 0
Unpacking objects: 100% (14/14), done.
From https://github.com/ZerBea/hcxdumptool
   0637f4c..dd8925f  master     -> origin/master
Updating 0637f4c..dd8925f
Fast-forward
 changelog     | 22 ++++++++++++++++++++++
 hcxdumptool.c |  9 ++++++---
 2 files changed, 28 insertions(+), 3 deletions(-)
# make
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c 
# hcxdumptool -i wlan0 -c 14
initialization...
warning: NetworkManager is running with pid 433
warning: wpa_supplicant is running with pid 541
failed to set monitor mode: Operation not permitted
failed to init socket
```
Even though there is the "monitor mode attention" section, I had to kill network manager to set the device to monitor mode.
```
# kill 541
# hcxdumptool -i wlan0 -c 14
initialization...
warning: NetworkManager is running with pid 433
failed to set monitor mode: Operation not permitted
failed to init socket
# kill 433
# hcxdumptool -i wlan0 -c 14
initialization...
failed to set monitor mode: Operation not permitted
failed to init socket
```
And I set the device to monitor mode manually
```
# ip link set wlan0 down
# iw dev wlan0 set type monitor
# ip link set wlan0 up
# hcxdumptool -i wlan0 -c 14
initialization...
warning: unable to set channel 14 (removed this channel from scan list)
no available channel found in scan list

terminated...
```

I am puzzled because I can capture packages with airodump. It is just too manual using airodump, trying to connect to wifi networks from another device to generate the pkmid data and checking with aircrack latter if the handshakes are there.

ZerBea, 2019-03-09:
Latest commits are not uploaded correctly, I did another git push. Please try it again
$ hcxdumptool -i wlan0 -c 1,6,11

Warning now should look like this:
warning: failed to set channel 1 (xxxxxx) - removed this channel from scan list
We need the error message in the brackets from the driver.

Also please set monitor mode by iw and then try to set channel:
$ ip link set wlan0 down
$ iw dev wlan0 set type monitor
$ ip link set wlan0 up
$ iw dev wlan0 set channel 6
$ iw dev

ZerBea, 2019-03-09:
Please post also the output of:
$ hwinfo --wlan

ZerBea, 2019-03-09:
Added another option: --ignore_warning
From now on, hcxdumptool will not terminate during the initalization
if another service take access on the interface. Expect problems if the service tries to change channel or hcxdumptool tries to change channel.

$ hcxdumptool -i wlp0s20f0u3 --ignore_warning
initialization...
warning: NetworkManager is running with pid 449
warning: wpa_supplicant is running with pid 498
interface is possible in use by another service

start capturing (stop with ctrl+c)
INTERFACE................: wlp0s20f0u3
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries
MAC CLIENT...............: b4e1ebbd1fb3
MAC ACCESS POINT.........: 28ef01e8a2e7 (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 62524
ANONCE...................: edbde4881adf034f909108fce3032451bf3326887c1e62065b0885e1480441b5

ZerBea, 2019-03-09:
We have a possible similar issue in combination with NetworkManager and mt76x0 driver.
read more here:
https://github.com/openwrt/mt76/issues/216#issuecomment-470430002

That is the reason, why a run code that let hcxdumptool terminate, if NetworkManager is active.
--ignore_warning will disable this

ZerBea, 2019-03-09:
BTW:
Please keep in mind that you are running a third party driver, which is patched from the staging driver by the aircrack-ng maintainer to work with aircrack-ng suite.

ZerBea, 2019-03-09:
Pushed another update:
if option --ignore_warning is set, hcxdumptool doesn't check whether the interface is operational or not
(IFF_RUNNING). We only check that the interface is up (IFF_UP).

Now it should working with the modified NetworkManager.conf

[device]
wifi.scan-rand-mac-address=no

[ifupdown]
managed=false

[connection]
wifi.powersave=0

[main]
plugins=keyfile

[keyfile]
unmanaged-devices=mac:xx:xx:xx:xx:xx

respz, 2019-03-09:
Did a git pull and compiled:
```
# git pull
remote: Enumerating objects: 21, done.
remote: Counting objects: 100% (21/21), done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 21 (delta 9), reused 12 (delta 4), pack-reused 0
Unpacking objects: 100% (21/21), done.
From https://github.com/ZerBea/hcxdumptool
   dd8925f..f126b17  master     -> origin/master
Updating dd8925f..f126b17
Fast-forward
 README.md             |  4 ++++
 changelog             | 11 +++++++++++
 hcxdumptool.c         | 45 ++++++++++++++++++++++++++++++++++-----------
 include/hcxdumptool.h |  3 ++-
 4 files changed, 51 insertions(+), 12 deletions(-)

# make
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c 
# make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c 
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxdumptool
rm -f *.o *~
```

> Latest commits are not uploaded correctly, I did another commit. Please try it again
> $ hcxdumptool -i wlan0 -c 1,6,11
> 
> Warning now should look like this:
> warning: failed to set channel 1 (xxxxxx) - removed this channel from scan list
> We need the error message in the brackets from the driver.
```
# hcxdumptool -i wlan0 -c 1,6,11
initialization...
warning: NetworkManager is running with pid 463
warning: wpa_supplicant is running with pid 785
failed to set monitor mode: Operation not permitted
failed to init socket
```

> Also please set monitor mode by iw and then try to set channel:
> $ ip link set wlan0 down
> $ iw dev wlan0 set type monitor
> $ ip link set wlan0 up
> $ iw dev wlan0 set channel 11
> $ iw dev

```
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 10:62:eb:30:65:9b
		type monitor
		txpower 13.00 dBm
```

> Please post also the output of:
> $ hwinfo --wlan

```
# hwinfo --wlan
14: USB 00.0: 0282 WLAN controller                              
  [Created at usb.122]
  Unique ID: ADDn.V1buGFi11QA
  Parent ID: k4bc.NE9nepdzT05
  SysFS ID: /devices/pci0000:00/0000:00:0b.0/usb1/1-1/1-1:1.0
  SysFS BusID: 1-1:1.0
[... 95 lines cut ...]

# iw dev wlan0 set channel 6
# iwconfig wlan0
wlan0     IEEE 802.11b  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:2.437 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=-100 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

# iw dev wlan0 set channel 1
# iwconfig wlan0
wlan0     IEEE 802.11b  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Monitor  Frequency:2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=-100 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
```

ZerBea, 2019-03-09:
Ok, pushed another update, now we will retrieve the frequency, too
instead of:
warning: unable to set channel 11  (removed this channel from scan list)
we retrieve this message:
warning: unable to set channel 11 (24xx) - removed this channel from scan list
please do a git pull and test again

respz, 2019-03-09:
Ok, now I got this:
```
# hcxdumptool -i wlan0 -c 1,6,11
initialization...
warning: NetworkManager is running with pid 461
warning: wpa_supplicant is running with pid 951
warning: unable to set channel 1 (241200000) - removed this channel from scan list
warning: unable to set channel 6 (243700000) - removed this channel from scan list
warning: unable to set channel 11 (246200000) - removed this channel from scan list
no available channel found in scan list

terminated...
```

respz, 2019-03-09:
Could this be just a problem of number formatting before setting the channel?

ZerBea, 2019-03-09:
Great, we got it.
The driver is reporting a wrong frequency! Normally we expect the frequency in MHz
 1 / 2412MHz
 6 / 2437MHz
11 / 2462MHz

I'll check how I can fix this.

ZerBea, 2019-03-09:
Pushed another update: fixed issue if driver reports the frequency in Hz instead of MHz
Please try it.

respz, 2019-03-09:
Sorry for the delay. I updated kali and had to reinstall wifi driver.
Ok, now it is working.
```
# hcxdumptool -i wlan0 -c 1,6,11
initialization...
warning: NetworkManager is running with pid 421
warning: wpa_supplicant is running with pid 743

start capturing (stop with ctrl+c)
INTERFACE................: wlan0
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries
MAC CLIENT...............: b0ece1a33b3b
MAC ACCESS POINT.........: 586ed6b8141f (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 62517
ANONCE...................: 2c3c16c327b298a3c21280a1afd3ebcce3b52ca0e633c7f1f271dd169ce0aa48

INFO: cha=6, rx=4889, rx(dropped)=43, tx=389, powned=0, err=0
INFO: cha=11, rx=12363, rx(dropped)=195, tx=1494, powned=1, err=0
```
I am going to let it run for a few minutes to see if it captures a some pkmids.
Thank you.

ZerBea, 2019-03-09:
run hcxdumptool with option -enable_status=1
and close issue if everything is working like expected.

closed 2019-03-09 by respz

ZerBea, 2019-03-09:
Great. Thanks for reported that issue. Now we can mark this driver as working in combination with hcxdumptool.

respz, 2019-03-09:
I am glad that I could help.