ZerBea/hcxdumptool #49: improvement to Debian package

kretcheu, 2019-03-15:
Hi,
I'm packaging hcxdumptool to Debian.
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=924592

First of all, I like hcx* tools! And I use it with wifite to pentests.

Some thinks could be made to improve compliance of Debian policies and turn hcxdumptool even better.

Following what Debian Lintian says about it.

**P: hcxdumptool source: source-contains-empty-directory include/android-ifaddrs/**

Having an empty directory in itself does not cause problems but ...
Keeping the empty directory in the source package can prevent others from contributing to the package when using tools like git-buildpackage. In this workflow the empty directory would be lost. Potentially causing errors if the installed binary package or its tests subsequently rely upon them.

**I: hcxdumptool source: testsuite-autopkgtest-missing**
Having a test suite aids with automated quality assurance of the archive outside of your package.

**X: hcxdumptool source: debian-watch-does-not-check-gpg-signature**

Of course, not all upstreams provide such signatures but you could
request them as a way of verifying that no third party has modified the
code after its release (projects such as phpmyadmin, unrealircd, and
proftpd have suffered from this kind of attack).

I found some typos:
[02.fix.spellerros.txt](https://github.com/ZerBea/hcxdumptool/files/2972692/02.fix.spellerros.txt)

Thanks a lot.
Regars.
kretcheu
:x:

ZerBea, 2019-03-15:
fixed:
P: hcxdumptool source: source-contains-empty-directory include/android-ifaddrs/
fixed:
02.fix.spellerros.txt

not fixed, because debian dependant:
I: hcxdumptool source: testsuite-autopkgtest-missing

not fixed, because debian dependant:
X: hcxdumptool source: debian-watch-does-not-check-gpg-signature

kretcheu, 2019-03-15:
Hi,
great! So quickly!

not fixed, because debian dependant:
X: hcxdumptool source: debian-watch-does-not-check-gpg-signature

About this, it's not a impediment but I think its a good practice, what do you think about?

In Debian packaging we used a copyright file and a upstream email but I can't found one.

[]'s
kretcheu
:x

kretcheu, 2019-03-16:
Thanks, email retrived.

Copyright file is very simple:
https://www.debian.org/doc/manuals/maint-guide/dreq.de.html#copyright

ZerBea, 2019-03-16:
Looked for an example, tried gzip and found 4 files related to gzip:
Except of the upstream tarball (gzip_1.6.orig.tar.gz), all of them,  looking debian specific - and the upstream tarball doesn't contain a "copyright" file.
First file, which contains the copyright file is the debian specific tarball (debian.tar.xz), with any changes made to upstream source, plus all the files created for the Debian package. This should be done by the debian maintainer:

gzip_1.6-5.dsc
gzip_1.6.orig.tar.gz
gzip_1.6-5.debian.tar.xz
gzip_1.6-5+b1_amd64.deb

Because Arch Linux doesn't need this files to build a package, I decided not to add it.

Example of Arch Linux PKGBUILD:
# Maintainer: Levente Polyak <anthraxx[at]archlinux[dot]org>

pkgname=hcxdumptool
_gitcommit=a703613e98da075dbc1b193f7d5631647c477b89
pkgver=5.1.3
pkgrel=1
pkgdesc='Small tool to capture packets from wlan devices'
url='https://github.com/ZerBea/hcxdumptool'
arch=('x86_64')
license=('MIT')
depends=('glibc')
makedepends=('git')
source=("${pkgname}::git+https://github.com/ZerBea/hcxdumptool#commit=${_gitcommit}")
sha512sums=('SKIP')

pkgver() {
  cd ${pkgname}
  git describe --long --tags | sed -E 's|v?(.+)|\1|' | sed 's/\([^-]*-g\)/r\1/;s/-/./g' | sed -E 's|\.r0\.\w+$||'
}

build() {
  cd ${pkgname}
  make
}

package() {
  cd ${pkgname}
  make DESTDIR="${pkgdir}" PREFIX=/usr install
  install -Dm 644 README.md changelog -t "${pkgdir}/usr/share/doc/${pkgname}"
  install -Dm 644 license.txt -t "${pkgdir}/usr/share/licenses/${pkgname}"
}

# vim: ts=2 sw=2 et:

kretcheu, 2019-03-17:
Hi, 
I think that I generated misunderstood.
Two subjects.
1. Copyright file: Is about put your email on there.
 I don't have any problem.
2. signed tarball.
 If you can generate signed tarball with github when release is lauched is good for all distributions

ZerBea, 2019-03-17:
Ok, done with sha256:
946f1d8ce6ff8cc41b295f61236522143d6ec9378afbdc94b79169b24b53d36c  hcxdumptool-5.1.4.tar.gz
3752a2722065782dd94f5a0c541d537665a38c55c078013f2f1845e95aa7c4c0  hcxdumptool-5.1.4.zip

https://github.com/ZerBea/hcxdumptool/releases/tag/5.1.4

Now let's see if git changes the content of the tarball.

kretcheu, 2019-03-18:
Hi,

Scuse me if I'm not clear.

I'm talking about this:
https://wiki.debian.org/Creating%20signed%20GitHub%20releases

ZerBea, 2019-03-18:
You are clear, but I prefer sha256sum (not pgp) to verify integrity. That is more than enough, because everybody can download the sha256sum from here:
https://github.com/ZerBea/hcxdumptool/releases/tag/5.1.4
to verify the tarball and/or the zip, without using a key server!

BTW: 
Arch Linux does the same:
https://wiki.archlinux.org/index.php/PKGBUILD#Integrity

ZerBea, 2019-03-18:
Ok, additionally I signed both (tarball and zip) by gpg:
https://github.com/ZerBea/hcxdumptool/releases/tag/5.1.4

ZerBea, 2019-03-18:
hcxdumptool is 100% compatible to wireshark (tshark) suite - but to get full benefit (hashcat and john support), you need hcxtools, too

kretcheu, 2019-03-18:
> hcxdumptool is 100% compatible to wireshark (tshark) suite - but to get full benefit (hashcat and john support), you need hcxtools, too.

I know that, I packaged hcxtools too, for my use, but another Debian collaborator open a bug to intent to package it, som I'm waiting...

I'm working on hcxkeys too.

ZerBea, 2019-03-19:
Please check also this issue report here:
https://github.com/ZerBea/hcxtools/issues/89

ZerBea, 2019-09-01:
hcxdumptool found it's way to Debian:
https://packages.debian.org/sid/hcxdumptool
So we can close this issue, now.
But I really wonder, why hcxtools still missing there.

closed 2019-09-01 by ZerBea