ZerBea/hcxdumptool #51: hcxdumptool  has no mac os compatibility

qingyangcx, 2019-03-19:
When will it support the mac os?

ZerBea, 2019-03-19:
No way.
As well as WINDOWS, MAC OS is a proprietary operating system. I will not support a proprietary system, because I don't like the idea to reverse engineer their code to make hcxdumptool work.

Try to run Linux in a VM.

closed 2019-03-19 by ZerBea

ZerBea, 2019-03-19:
Of course, you can capture packets (very limited). Read more here:
https://null-byte.wonderhowto.com/how-to/mac-for-hackers-set-up-macos-system-for-wi-fi-packet-capturing-0186354/
Packet injection is not supported natively, but you can try to make a "feature request"  here:
https://support.apple.com/en-ca

qingyangcx, 2019-03-19:
I encountered segmentfault when i execute "airport en0 sniffer 6", this is the reason that i look for other tool to caputure handshake

ZerBea, 2019-03-19:
That is exactly the problem, because the driver doesn't support it and the code is proprietary. So there is nothing we can do to make it work.

qingyangcx, 2019-03-19:
I think it is a serious bug, don't know why apple still not fix it?

ZerBea, 2019-03-19:
Maybe no priority here for Apple to fix it.
One of the reasons, I moved to Linux. On an issue, I can talk with a kernel developer and we fix it:
https://bugzilla.kernel.org/show_bug.cgi?id=202243
Everything is open source!

qingyangcx, 2019-03-19:
Excellent job!Wish i could do some contribute to open source!

Goldenstarc, 2019-07-29:
> I encountered segmentfault when i execute "airport en0 sniffer 6", this is the reason that i look for other tool to caputure handshake

use tcpdump,
you can wait for handshake and crack it!

ZerBea, 2019-08-24:
Some additional information about macOS
https://github.com/aircrack-ng/rtl8812au/issues/376#issuecomment-524198833

FreshNewMac, 2023-06-18:
> I encountered segmentfault when i execute "airport en0 sniffer 6", this is the reason that i look for other tool to caputure handshake

airport en0 sniff 6... you don't even need to specify en0, because it's the default interface, so airport sniff 6 will work