ZerBea/hcxdumptool #52: Isues installing HCXDUMPtools and hcxtools.

djfunny007, 2019-03-19:
hi all. i am keep getting errors while installing hcxdumtools, hcxtools and hashcat. 

the error i am gettitng for the hcxdumptool is as below:

usr/bin/ld: cannot open output file hcxdumptool: Is a directory
collect2: error: ld returned 1 exit status
make: *** [Makefile:29: build] Error 1

can someone please specify the steps by steps how to install the above tools.

thank you

ZerBea, 2019-03-19:
delete all directories with hcxdumptool, hcxtools and hashcat.
do a fresh git clone for every tool:
git clone https://github.com/ZerBea/hcxdumptool
git clone https://github.com/ZerBea/hcxtools
git clone https://github.com/hashcat/hashcat

cd into every folder and run make install:
cd hcxtools
make install
cd ..
cd hcxdumptool
make install
cd ..
cd hashcat
make install
cd ..
forget the youtube videos where all tools are copied together into one directory.

This is an issue tracker. If you have questions, please ask here for help:
https://hashcat.net/forum/
https://hashcat.net/forum/thread-6661.html

Closed, because it ins't an issue.

closed 2019-03-19 by ZerBea

huitc, 2019-03-19:
same problem?...

ZerBea, 2019-03-19:
This issue was caused by a broken installation.
possible reasons:
- user copied source code from all tools into one folder
- user used source code folder as working directory

I assume OS is KALI!

@huitc you're running Android, but a required include file wasn't pulled. So you have to pull it and copy it to include folder.

Elia387, 2019-07-05:
Hi i am trying to clone, but it's asking for username and password. Where can I find those?

strasharo, 2019-07-06:
Use the https url instead of the SSH one.

ZerBea, 2019-07-06:
No password required to clone hcxtools/hcxdumptool:
$ git clone https://github.com/ZerBea/hcxdumptool
Klone nach 'hcxdumptool' ...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (31/31), done.
remote: Total 1336 (delta 26), reused 34 (delta 15), pack-reused 1290
Empfange Objekte: 100% (1336/1336), 460.31 KiB | 1.09 MiB/s, Fertig.
Löse Unterschiede auf: 100% (879/879), Fertig.

Elia387, 2019-07-06:
Hi, thanks. I was able to clone hcxdumptool amd hcxtools and hashcat but when i complie;

1.hcxdumptool using

 cd /usr/local/bin/hcxdumptool
make
make install

then it gives error " install: "hcxdumptool" and  /usr/local/bin/hcxdumptool are the same name file.
make***:[makefile:29: install error 1]

2. After installation of hcxtools
When i go to
 /usr/local/bin/hcxtools
make
then it says: fatal error curl/curl.h no file or directory found. 
#include <curl/curl.h>

3. After clonning hashcat;
when i go to cd /usr/local/bin/hashcat
then it gives error

 " install: cannot overwrite directory /usr/local/bin/hashcat with non directory.
make***:[src/makefile:455: install hashcat] Error 1

ZerBea, 2019-07-06:
Your workflow is an absolute no go!
1. first install dependencies:
To install requirements on Kali use the following 'apt-get install libcurl4-openssl-dev libssl-dev zlib1g-dev'
2. never clone to your binary directory /usr/local/bin/ !!!!!
Instead cd to your home directory and do a git clone, followed by a cd to the hcxtools folder. Than run make and make install (as super user)
cd $HOME
git clone https://github.com/ZerBea/hcxtools
cd hcxtools
make
make install

cd $HOME
git clone https://github.com/ZerBea/hcxdumptool
cd hcxdumptool
make
make install

For hashcat you must install GPU driver and opencl runtime!!!
https://hashcat.net/wiki/doku.php?id=frequently_asked_questions#installation

cd $HOME
git clone https://github.com/hashcat/hashcat
cd hashcat
make
make install

Read more here:
https://null-byte.wonderhowto.com/how-to/hack-wi-fi-cracking-wpa2-passwords-using-new-pmkid-hashcat-attack-0189379/

Elia387, 2019-07-06:
@ZerBea, thanks a lot. I did it. :)

ZerBea, 2019-07-06:
Fine.

matt8835, 2019-11-12:
gracias amigo !!!!

ZerBea, 2019-11-12:
No hay de qué.

ChaarSaal, 2024-06-17:
I am getting an error while running the 'apt-get install libcurl4-openssl-dev libssl-dev zlib1g-dev' command:-
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libssl-dev is already the newest version (3.0.13-1~deb12u1).
libssl-dev set to manually installed.
zlib1g-dev is already the newest version (1:1.2.13.dfsg-1).
zlib1g-dev set to manually installed.
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libcurl4-openssl-dev : Depends: libcurl4 (= 7.88.1-10+deb12u6) but 8.8.0-1~bpo12+1 is to be installed
E: Unable to correct problems, you have held broken packages.

Actually, I want to run make and make install for hcxdumptools and hcxtools but get the following error:-
hcxdumptool.c:41:10: fatal error: pcap/pcap.h: No such file or directory
   41 | #include <pcap/pcap.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:75: hcxdumptool] Error 1

Any fix for the situation?

Jake-Grafton, 2024-06-17:
@ChaarSaal 
I'm not certain what documentation you are reading as current installation instructions should be as follows:
1. Update your system to it's latest: `sudo apt-get update && sudo apt-get upgrade -y`
2. Install required system packages: `sudo apt-get install git gcc libpcap libpcap-dev`

**Note:** These packages may come under different names! Make sure you're installing the correct packages!

3. Clone the repository: `git clone https://github.com/ZerBea/hcxdumptool.git`
4. cd into that directory: `cd hcxdumptool`
5. Compile and install: `make -j $(nproc) && sudo make install`

From here, the tool should be installed and ready for use.

ZerBea, 2024-06-18:
@ChaarSaal 
By default, Debian based distributions are not designed to compile and install additional tools (e.g. hcxtools, hcxdumptool).
If that's what you want, it is mandatory to install the development tools and the development files (as mentioned by @Jake-Grafton ).

Unfortunately your package manager is broken on your distribution and you got this error;
`E: Unable to correct problems, you have held broken packages.`
To fix that please take a look at the documentation of your distribution or e.g. here:
https://superuser.com/questions/1793779/parrot-os-unmet-dependencies-held-broken-packages.