ZerBea/hcxdumptool #53: installaction problem

huitc, 2019-03-19:
what's the problem?...
```
$ make
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c
hcxdumptool.c:23:10: fatal error: 'include/android-ifaddrs/ifaddrs.h' file not found
#include "include/android-ifaddrs/ifaddrs.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [Makefile:29: build] Error 1
```
```
$ sudo make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c
hcxdumptool.c:23:10: fatal error: 'include/android-ifaddrs/ifaddrs.h' file not found
#include "include/android-ifaddrs/ifaddrs.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [Makefile:29: build] Error 1
```

huitc, 2019-03-19:
Linux localhost 4.4.78-perf-ge57b3e5 #1 SMP PREEMPT Thu Mar 14 04:05:42 CST 2019 aarch64 Android
android 8.0.0
https://play.google.com/store/apps/details?id=com.termux

ZerBea, 2019-03-19:
It looks like your environment doesn't pull Android ifaddrs.h automatically.
Download ifaddrs.h from here:
https://github.com/morristech/android-ifaddrs
copy ifaddrs.h to include/android-ifaddrs/
and try it again.

ZerBea, 2019-03-19:
I assume that you:
did a fresh git clone of hcxdumptool
and prepared the clone for Android:
git submodule init
git submodule update

ZerBea, 2019-03-19:
From the README.md:
Compile for Android

You need:
* Android NDK installed in your system and in path variable

* This repository cloned with all submodules (`--recursive` flag in `git clone` or `git submodules update` command run)

Just run `ndk-build` - built executables for some architectures should be created inside `libs` directory.
Copy it to your phone and enjoy.

huitc, 2019-03-20:
```
$ make
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c
In file included from hcxdumptool.c:24:
./include/android-ifaddrs/ifaddrs.c:89:9: warning: unused variable 'l_result'
      [-Wunused-variable]
    int l_result;
        ^
./include/android-ifaddrs/ifaddrs.c:291:15: warning: unused variable 'l_rtaData'
      [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
              ^
./include/android-ifaddrs/ifaddrs.c:376:15: warning: unused variable 'l_rtaData'
      [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
              ^
hcxdumptool.c:221:20: warning: unused function 'debugprint' [-Wunused-function]
static inline void debugprint(int len, uint8_t *ptr)
                   ^
hcxdumptool.c:233:20: warning: unused function 'debugprint2' [-Wunused-function]
static inline void debugprint2(int len, uint8_t *ptr, int len2, uint8_t *ptr2)
                   ^
5 warnings generated.
$ sudo make install
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c
In file included from hcxdumptool.c:24:
./include/android-ifaddrs/ifaddrs.c:89:9: warning: unused variable 'l_result' [-Wunused-variable]
    int l_result;
        ^
./include/android-ifaddrs/ifaddrs.c:291:15: warning: unused variable 'l_rtaData' [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
              ^
./include/android-ifaddrs/ifaddrs.c:376:15: warning: unused variable 'l_rtaData' [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
              ^
hcxdumptool.c:221:20: warning: unused function 'debugprint' [-Wunused-function]
static inline void debugprint(int len, uint8_t *ptr)
                   ^
hcxdumptool.c:233:20: warning: unused function 'debugprint2' [-Wunused-function]
static inline void debugprint2(int len, uint8_t *ptr, int len2, uint8_t *ptr2)
                   ^
5 warnings generated.
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
install: cannot create directory '/usr': Read-only file system
make: *** [Makefile:38: install] Error 1
```

huitc, 2019-03-20:
i can't!!... t.t

huitc, 2019-03-20:
`/data/data/com.termux/files/usr/`???...

ZerBea, 2019-03-20:
Ok, now we have several different issues:

1) I fixed this 2 warnings by latest commit:
https://github.com/ZerBea/hcxdumptool/commit/41c4547eac5be94e76bd60c9277b2b01338ee2fa

hcxdumptool.c:221:20: warning: unused function 'debugprint' [-Wunused-function]
static inline void debugprint(int len, uint8_t *ptr)
                   ^
hcxdumptool.c:233:20: warning: unused function 'debugprint2' [-Wunused-function]
static inline void debugprint2(int len, uint8_t *ptr, int len2, uint8_t *ptr2)

2) I can't fix that warnings, because the source code is not part of hcxdumptool. This Android source code is pulled from here: https://github.com/morristech/android-ifaddrs
./include/android-ifaddrs/ifaddrs.c:89:9: warning: unused variable 'l_result'
      [-Wunused-variable]
    int l_result;
        ^
./include/android-ifaddrs/ifaddrs.c:291:15: warning: unused variable 'l_rtaData'
      [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);
              ^
./include/android-ifaddrs/ifaddrs.c:376:15: warning: unused variable 'l_rtaData'
      [-Wunused-variable]
        void *l_rtaData = RTA_DATA(l_rta);

That are just warnings, but you can report that issue here:
https://github.com/morristech/android-ifaddrs

3) You are running into a broken Android environment and you have to fix that.
Is Android NDK properly installed in your system and in path variable?
Did you run ndk-build to build the envirnonment?
Running simple "make" without building Android environment before isn't enough!

huitc, 2019-03-20:
> From the README.md:
> Compile for Android
>
> You need:
>
> * Android NDK installed in your system and in path variable
> * This repository cloned with all submodules (`--recursive` flag in `git clone` or [...]

`git submodule update`?

ZerBea, 2019-03-20:
By git submodules update you make sure, that external parts are loaded before you compile the source code. That can be done with git clone recursive flag or if the clone allready exists on your local system, with  git submodule init and git submodule update.
At this point you finished the hcxdumtool part.
Next step is to build your Android envirnonment for your phone, so that you are able to compile it:
Make sure, you understand this: https://developer.android.com/ndk/guides/ndk-build
than run ndk-build

ZerBea, 2019-03-20:
Ok, I think we are talking about different things. The make file is for Linux only. It doesn't work on an Android device.

Here are the basic steps you have to do:

1. you need a Linux system

2. do a git clone of hcxdumptool on that Linux system (either with recursive flag)
or cd to hcxtools and run git submodule init and git submodule update

3. Install Android NDK using the packaging system of your distribution or by hand
Read more here:
https://stackoverflow.com/questions/26967722/how-to-install-android-ndk-in-linux
https://wiki.appcelerator.org/display/guides2/Installing+the+Android+NDK
https://www.xda-developers.com/understanding-cross-compiling-and-how-it-relates-to-android/

do not try to do this on a live system, because that are several GB!
do not try to this on a penetration testing system like KALI - you will run into dependency trouble

4. cd to hcxdumptool directory and run ndk-build

5. if the make process finished without errors, copy the binary to your phone

ZerBea, 2019-03-21:
Closed this issue, because it is related to missing Android NDK.

closed 2019-03-21 by ZerBea

kamesh413, 2019-08-08:
showing
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c 
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c 
install -m 0755 -D hcxpioff /usr/local/bin/hcxpioff
install -m 0755 -D hcxdumptool /usr/local/bin/hcxdumptool
rm -f hcxpioff
rm -f hcxdumptool
rm -f *.o *~