ZerBea/hcxdumptool #56: MT76x8 support

cirolaferrara, 2019-04-16:
I am on [GL.iNet GL-MT300N v2](https://openwrt.org/toh/hwdata/gl.inet/gl.inet_gl-mt300n_v2) with Openwrt. I can go into monitor mode:
```
# iw phy phy0 interface add mon0 type monitor
# ifconfig mon0 up
# hcxdumptool -i mon0 -o /tmp/wifi.pcapng --enable_status=1
```

But I have always timeouts:

> initialization...
> warning: mon0 is probably a monitor interface
> 
> start capturing (stop with ctrl+c)
> ...
> ...
> [18:10:34 - 001] 00888888002a -> e8abfa960007 [FOUND AUTHORIZED HANDSHAKE, > [18:11:52 - 001] 54600930adc4 -> c4ea1d3d2a2f [FOUND HANDSHAKE AP-LESS, EAPOL TIMEOUT 2894]
> INFO: cha=1, rx=6194, rx(dropped)=1460, tx=599, powned=3, err=0

I compiled hcxdumptool with adde88 [Makefile](https://github.com/adde88/hcxtools-hcxdumptool-openwrt/blob/master/net/hcxdumptool/Makefile) without his [patch](https://github.com/adde88/hcxtools-hcxdumptool-openwrt/blob/master/net/hcxdumptool/patches/100-musl-compitability.patch).

What's wrong? How can I check if is my device (unofficially) supported?

ZerBea, 2019-04-16:
hcxdumptool measures the EAPOL time between EAPOL frames within a 4-way handshake.
We call this EAPOL TIMEOUT. We use this EAPOL TIMEOUT value to detect valid (good) handshakes.
Also we use this value to detect if we can do nonce-error-corrections in case of a packet loss.
So, that is a normal behavior.
To check if everything is fine and to check that injection is working just run
$ hcxdumptool -i INTERFACE --do_rcascan
If you got messages like this one (AP IN RANGE), every thing's fine
If you don't get this messages, either
injection isn't working or
the AP is too far away.
Also you can get information about targets for your target list from --do_rcascan.

You get this warning:
warning: mon0 is probably a monitor interface
because you do not follow the recommendation from help menu:
-i <interface> : interface (monitor mode will be enabled by hcxdumptool)
                 can also be done manually:
                 ip link set <interface> down
                 iw dev <interface> set type monitor
                 ip link set <interface> up

you added an interface, instead of setting it directly to monitor mode!
hcxdumptool doesn't like additional virtual interfaces - that will us make slow.

You still can ask further questions here, but I close this report, because it isn't an issue.

ZerBea, 2019-04-16:
BTW:
Mediatek devices are very  good and cheap. I really like them because they are sensitive and well supported by kernel driver out of the box.

closed 2019-04-16 by ZerBea

cirolaferrara, 2019-04-16:
What do You think abouth the [patch](https://github.com/adde88/hcxtools-hcxdumptool-openwrt/blob/master/net/hcxdumptool/patches/100-musl-compitability.patch)?

ZerBea, 2019-04-16:
I think the patch is ok, because it doesn't touch the "engine" of hcxdumptools.  ‎The patch makes sure that hcxdumptool can compile without problems, running musl.

cirolaferrara, 2019-04-16:
I can compile hcxdumptool also without the patch.

ZerBea, 2019-04-16:
First part of the patch cleans up the includes. I included them twice and that was fixed by latest commit:
https://github.com/ZerBea/hcxdumptool/commit/267b03c774c5ab273c110caeec2751e19a75c03b
Removed this dupes:
#include <net/if.h>
#include <arpa/inet.h>

__GLIBC__ macro is used to shield some features unless you explicated want a glibc only feature.
So, from my opinion, the patch  is ok. If __GLIBC__ is not defined, we don't use it.

cirolaferrara, 2019-04-16:
See also [hcxtools patch](https://github.com/adde88/hcxtools-hcxdumptool-openwrt/blob/master/net/hcxtools/patches/001-build-wlanhcx2psk.patch) please.

ZerBea, 2019-04-16:
This patch is ok, too. I dropped wlanhcx2psk. It is replaced by hcxpsktool.
If you would like to use old version and new version together, you need to patch the makefile.

ZerBea, 2019-04-16:
BTW:
That reminds me to finally remove wlanhcx2psk from the suite. Done by latest commit:
https://github.com/ZerBea/hcxtools/commit/5a335344be50e712e2e9bc43840c47367667fba6
Thanks!