ZerBea/hcxdumptool #57: Not working with mt7612u on kernel 4.19

neheb, 2019-04-16:
# ./hcxdumptool -i wlx00c0ca95876c
initialization...
failed to save current interface mode: Operation not supported
failed to init socket

ZerBea, 2019-04-17:
Thanks for reporting that. Now, the driver bug arrived in 4.19.

ZerBea, 2019-04-17:
Yesterday, I made an update on device wiki and README.me.
Due to several driver issues, many devices are not working any longer. The issues are still present in 5.1
And this one
https://bugzilla.kernel.org/show_bug.cgi?id=202541
affects all of them.

neheb, 2019-04-17:
That's unfortunate given that 4.19 introduces mt7612u...

ZerBea, 2019-04-17:
Please test the driver running this commands:
$ hcxdumptool -I wlx00c0ca95876c
$ hcxdumptool -i wlx00c0ca95876c -c --ignore_warning
$ hcxdumptool -i wlx00c0ca95876c --enable_status=1 --ignore_warning

Maybe we can find out, which driver issue exactly caused it.
The xhci bug is a really ugly one and still unfixed - all ALFAs are affected!

ZerBea, 2019-04-18:
BTW:
AWUS036ACM seen in linux-5.0.8/drivers/net/wireless/mediatek/mt76/mt76x2/usb.c:

static const struct usb_device_id mt76x2u_device_table[] = {
	{ USB_DEVICE(0x0e8d, 0x7612) }, /* Alfa AWUS036ACM */
	{ USB_DEVICE(0x0b05, 0x1833) },/* Asus USB-AC54 */
	{ USB_DEVICE(0x0b05, 0x17eb) },/* Asus USB-AC55 */
	{ USB_DEVICE(0x0b05, 0x180b) },/* Asus USB-N53 B1 */
	{ USB_DEVICE(0x0e8d, 0x7612) },/* Aukey USB-AC1200 */
	{ USB_DEVICE(0x057c, 0x8503) },	/* Avm FRITZ!WLAN AC860 */
	{ USB_DEVICE(0x7392, 0xb711) },/* Edimax EW 7722 UAC */
	{ USB_DEVICE(0x0846, 0x9053) },/* Netgear A6210 */
	{ USB_DEVICE(0x045e, 0x02e6) },/* XBox One Wireless Adapter */
	{ },
};

but unfortunately not in linux-4.19.35/drivers/net/wireless/mediatek/mt76/mt76x2_usb.c:
static const struct usb_device_id mt76x2u_device_table[] = {
	{ USB_DEVICE(0x0b05, 0x1833) },/* Asus USB-AC54 */
	{ USB_DEVICE(0x0b05, 0x17eb) },/* Asus USB-AC55 */
	{ USB_DEVICE(0x0b05, 0x180b) },/* Asus USB-N53 B1 */
	{ USB_DEVICE(0x0e8d, 0x7612) },/* Aukey USB-AC1200 */
	{ USB_DEVICE(0x057c, 0x8503) },	/* Avm FRITZ!WLAN AC860 */
	{ USB_DEVICE(0x7392, 0xb711) },/* Edimax EW 7722 UAC */
	{ USB_DEVICE(0x0846, 0x9053) },/* Netgear A6210 */
	{ USB_DEVICE(0x045e, 0x02e6) },/* XBox One Wireless Adapter */
	{ },
};

neheb, 2019-04-18:
Eh no that’s incorrect. The ID is the same as the Aukey. My ALFA works fine in 4.19

ZerBea, 2019-04-18:
Ok, see it. WIll send Stanislaw a notice about that double entry.
Thanks
Cheers
Mike

ZerBea, 2019-05-20:
unfortunately the issue is still present on kernel 5.1.2
https://bugzilla.kernel.org/show_bug.cgi?id=202541#c23
Additionally non of my rt2800usb devices is working any longer.

neheb, 2019-05-20:
That's seriously unfortunate given that mt7612u support was added in 4.19.

strasharo, 2019-05-27:
Is rt2800usb on 4.19 also affected?

ZerBea, 2019-05-27:
The issue is related to the xhci system and affects all connected devices (irrespective of the driver)

$ hcxdumptool -I
wlan interfaces:
00c0ca367a0d wlp3s0f0u10u2 (rt2800usb)

ALFA AWUSH036NH
ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter

$ sudo hcxdumptool -i wlp3s0f0u10u2 --enable_status=1
initialization...
start capturing (stop with ctrl+c)
INTERFACE................: wlp3s0f0u10u2
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries
MAC CLIENT...............: dc7014bd61f7
MAC ACCESS POINT.........: 00238c314a88 (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 62111
ANONCE...................: 641329e38d27618344e7dadbbd54fea1b9963876543e70a9d85f65c2067b2f6a

INFO: cha=6, rx=0, rx(dropped)=0, tx=20, powned=0, err=0^C
terminated...

$ dmesg
[  855.923953] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.

and the issue still exists in
$ uname -r
5.1.4-arch1-1-ARCH

ZerBea, 2019-05-30:
$ uname -r
5.1.5-arch1-2-ARCH

and no fix:
[19881.339443] ieee80211 phy0: rt2x00_set_rt: Info - RT chipset 5390, rev 0502 detected
[19882.017891] ieee80211 phy0: rt2x00_set_rf: Info - RF chipset 5370 detected
[19882.039403] ieee80211 phy0: Selected rate control algorithm 'minstrel_ht'
[19882.039932] usbcore: registered new interface driver rt2800usb
[20001.915937] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[20001.921788] ieee80211 phy0: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
[20069.638944] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.

Most of the WiFi dongles are still useless!

ZerBea, 2019-07-11:
tested mainline kernel 5.2, still no fix:
https://bugzilla.kernel.org/show_bug.cgi?id=202541#c38

but we get a funny new error message:
[77.562305] ieee80211 phy1: rt2800_wait_csr_ready: Error - Unstable hardware

Tested device is an ALFA AWUS 036NH

ZerBea, 2019-08-13:
Pushed another update.
Now hcxdumptool has a new option --check_driver.
If everything is ok, result should be something like this:
$ sudo hcxdumptool -i wlp3s0f0u10u4 --check_driver
driver tests passed - all required ioctl() system calls are supported by driver

If an error ocured, hcxdumptool will inform you about possible unsupported/not working driver system calls.

ZerBea, 2019-09-01:
The mt7612u support (mt76 driver) should be arrived in kernel >= 5.1:
https://wikidevi.com/wiki/Mt76
https://github.com/raspberrypi/linux/issues/3014

ZerBea, 2020-01-23:
Monitor mode should work:
https://github.com/openwrt/mt76/issues/139

closed 2020-01-23 by ZerBea

neheb, 2020-01-23:
Yes it does.

ZerBea, 2020-01-23:
Great. The mt76 family is amazing and Lorenzo is doing a great job to let it work.
BTW:
I still have no idea how to fix this:
https://github.com/ZerBea/hcxtools/issues/106

neheb, 2020-01-24:
Small attempt on one of the tools:

```
diff --git a/Makefile b/Makefile
index 6f20784..2953549 100644
--- a/Makefile
+++ b/Makefile
@@ -36,7 +36,7 @@ TOOLS+=wlanhcx2john
 TOOLS+=hcxpcaptool
 hcxpcaptool_libs=-lz -lcrypto
 TOOLS+=hcxhashcattool
-hcxhashcattool_libs=-lcrypto -lpthread
+hcxhashcattool_libs=-lcrypto -lpthread -lmbedcrypto
 hcxpmkidtool_libs=-lcrypto -lpthread
 TOOLS+=hcxmactool
 TOOLS+=hcxessidtool
diff --git a/hcxhashcattool.c b/hcxhashcattool.c
index e69587f..9e2d1d8 100644
--- a/hcxhashcattool.c
+++ b/hcxhashcattool.c
@@ -16,7 +16,7 @@
 #include <stdio_ext.h>
 #endif
 #include <pthread.h>
-#include <openssl/evp.h>
+#include <mbedtls/pkcs5.h>
 
 #include "include/version.h"
 #include "include/hashcatops.h"
@@ -123,6 +123,11 @@ pmklist_t *zeiger;
 
 uint8_t emptypmk[32];
 
+mbedtls_md_context_t sha1_ctx;
+mbedtls_md_init(&sha1_ctx);
+const mbedtls_md_info_t *info_sha1 = mbedtls_md_info_from_type(MBEDTLS_MD_SHA1);
+mbedtls_md_setup(&sha1_ctx, info_sha1, 1);
+
 zeiger = realarg->pmkpos;
 pmkcountthread = realarg->pmkct;
 memset(&emptypmk, 0, 32);
@@ -130,7 +135,8 @@ for(c = 0; c < pmkcountthread; c++)
 	{
 	if(memcmp(&emptypmk, zeiger->pmk, 32) == 0)
 		{
-		if(PKCS5_PBKDF2_HMAC_SHA1((const char*)zeiger->psk, zeiger->psklen, (unsigned char*)zeiger->essid, zeiger->essidlen, 4096, 32, zeiger->pmk) == 0)
+		if(mbedtls_pkcs5_pbkdf2_hmac(&sha1_ctx, zeiger->psk, zeiger->psklen, zeiger->essid, zeiger->essidlen, 4096, 32, zeiger->pmk) == 0)
+//		if(PKCS5_PBKDF2_HMAC_SHA1((const char*)zeiger->psk, zeiger->psklen, (unsigned char*)zeiger->essid, zeiger->essidlen, 4096, 32, zeiger->pmk) == 0)
 			{
 			printf("failed to calculate PMK\n");
 			exit(EXIT_FAILURE);
@@ -150,6 +156,11 @@ pthread_t thread[17];
 argument_t args[17];
 uint8_t emptypmk[32];
 
+mbedtls_md_context_t sha1_ctx;
+mbedtls_md_init(&sha1_ctx);
+const mbedtls_md_info_t *info_sha1 = mbedtls_md_info_from_type(MBEDTLS_MD_SHA1);
+mbedtls_md_setup(&sha1_ctx, info_sha1, 1);
+
 cpucount = sysconf( _SC_NPROCESSORS_ONLN );
 if(cpucount > 16)
 	{
@@ -190,7 +201,8 @@ if(ct > 0)
 		{
 		if(memcmp(&emptypmk, zeiger->pmk, 32) == 0)
 			{
-			if(PKCS5_PBKDF2_HMAC_SHA1((const char*)zeiger->psk, zeiger->psklen, (unsigned char*)zeiger->essid, zeiger->essidlen, 4096, 32, zeiger->pmk) == 0)
+			if(mbedtls_pkcs5_pbkdf2_hmac(&sha1_ctx, zeiger->psk, zeiger->psklen, zeiger->essid, zeiger->essidlen, 4096, 32, zeiger->pmk) == 0)
+//			if(PKCS5_PBKDF2_HMAC_SHA1((const char*)zeiger->psk, zeiger->psklen, (unsigned char*)zeiger->essid, zeiger->essidlen, 4096, 32, zeiger->pmk) == 0)
 				{
 				printf("failed to calculate PMK\n");
 				exit(EXIT_FAILURE);
```

It compiles.

ZerBea, 2020-01-24:
Looking good. Thanks for the code snippet.
mbedtls is part of Arch Linux, too so I can test it.
https://www.archlinux.org/packages/community/x86_64/mbedtls/

BTW:
Most of the hcxtools are deprecated on next release of hashcat and JtR due to new hash format WPA-PBKDF2-PMKID+EAPOL.
https://github.com/hashcat/hashcat/issues/1816
https://github.com/magnumripper/JohnTheRipper/issues/4183

I'm going to remove this tools step by step, because we do not need them any longer.
Only hcxpcapngtool, hcxhashtool, hcxpsktool, hcxwltool and wlancap2wpasec will survive.

neheb, 2020-01-24:
Great!

RealEnder, 2020-01-24:
Hi, can you add -D make option to enable compiletime choice between openssl/mbedtls

neheb, 2020-01-24:
It's just a rough test. The crypto selection should be based on what curl is compiled with (it supports OpenSSL, GNU TLS, mbedTLS, and wolfSSL.

RealEnder, 2020-01-24:
Oh yeah, curl part is easy. But for zeroed PMK validation and cimple CPU cracker you have to use the lib directly :)