ZerBea/hcxdumptool #62: Make file build failing with `hcxdumptool.c:4775:16: error: 'ETH_ALEN' undeclared`

namanchikara, 2019-06-17:
I'm on Ubuntu 18.04, Below are the logs of `make`

```
/home/o_o/miniconda/bin/x86_64-conda_cos6-linux-gnu-cc -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -std=gnu99 -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -o hcxpioff hcxpioff.c -Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections
/home/o_o/miniconda/bin/x86_64-conda_cos6-linux-gnu-cc -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -std=gnu99 -DNDEBUG -D_FORTIFY_SOURCE=2 -O2 -o hcxdumptool hcxdumptool.c -Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections
In file included from /home/o_o/miniconda/x86_64-conda_cos6-linux-gnu/sysroot/usr/include/endian.h:61:0,
                 from /home/o_o/miniconda/x86_64-conda_cos6-linux-gnu/sysroot/usr/include/ctype.h:41,
                 from hcxdumptool.c:2:
hcxdumptool.c: In function 'opensocket':
hcxdumptool.c:4664:51: error: 'ETH_P_ALL' undeclared (first use in this function); did you mean 'ETH_TP_MDI'?
 if((fd_socket = socket(AF_PACKET, SOCK_RAW, htons(ETH_P_ALL))) < 0)
                                                   ^
hcxdumptool.c:4664:51: note: each undeclared identifier is reported only once for each function it appears in
hcxdumptool.c:4775:16: error: 'ETH_ALEN' undeclared (first use in this function); did you mean 'ETH_P_ALL'?
 ll.sll_halen = ETH_ALEN;
                ^~~~~~~~
                ETH_P_ALL
Makefile:20: recipe for target 'build' failed
make: *** [build] Error 1

```

namanchikara, 2019-06-17:
Somehow, Running `sudo make` instead of `make` does the trick! Maybe Documentations requires a bit of update?

> Compile
>
> 
> Simply run:
> 
>     `make`
>     `make install (as super user)`

closed 2019-06-17 by namanchikara

ZerBea, 2019-06-17:
No, documentation is ok.
Your environment is broken because you are not able to access your include files as "normal user".

namanchikara, 2019-06-18:
Oh Okay, Got it.