ZerBea/hcxdumptool #63: Filterlist larger than 64 entries

strasharo, 2019-06-30:
In the help output of the tool I noticed that the maximum number of entries is 64. Is possible to increase it to a larger number or there's some technical limitation in the implementation?

ZerBea, 2019-06-30:
Increased maximum filter list entries up to 256 by this commit:
https://github.com/ZerBea/hcxdumptool/commit/7bda4703c8d9de3fc3ee47920a0d76dd5f71d11a

Every incoming packet is checked against the filter list first (addr1, addr2, addr3). In other words, more entries will increase the response time of hcxdumptool. Goal is to respond faster than an access point.
Read more about "theoretical Limits of Wi-Fi Network Scaling" here:
https://homequipment.com/router-capacity/
https://community.cisco.com/t5/other-wireless-mobility-subjects/wireless-access-point-maximum-users/td-p/3042360
https://www.quora.com/How-can-the-maximum-number-of-devices-be-connected-to-a-computer%E2%80%99s-WiFi-hotspot

strasharo, 2019-06-30:
Awesome, thanks a lot. 256 Is more than enough, I just had a roaming hcxdumptool setup and happened to require slightly more than the current limit.

closed 2019-06-30 by strasharo

ZerBea, 2019-06-30:
Good to hear that. If the target is slow, we can increase the value again. But if the target is fast, we will loose. So let's hope that fast targets increased their network scaling, too and will go slow...