ZerBea/hcxdumptool #64: about killing wpa_supplicant

carlhung, 2019-07-02:
when i ran this command:
sudo hcxdumptool -i wlan1 -o test.pcapng --enable_status=1 -c 6

I got this message:
initialization...
warning: wpa_supplicant is running with pid 349 333
interface may not be operational
failed to init socket
hcxdumptool need full (monitor mode and full packet injection running all packet types) and exclusive access to the adapter
that is not the case

therefore, I killed 349 333
but the ssh connection got killed and can't connect the server by using ssh command.

I have two wifi interfaces.
wlan0 for ssh connection

wlan1 for attacking.

how can I keep the connection and to use wlan1 to attack?

ZerBea, 2019-07-02:
Make sure wpa_supplicant has no access to wlan1:
https://unix.stackexchange.com/questions/80550/how-to-prevent-wpa-supplicant-from-connecting-to-a-network

Run hcxdumptool with option --ignore_warning
$ sudo hcxdumptool --ignore_warning -i wlan1 -o test.pcapng --enable_status=1 -c 6
Running this option, hcxdumptool will ignore wpa_supplicant as long as wlan1 is untouched.

ZerBea, 2019-07-03:
Ok

closed 2019-07-03 by ZerBea