ZerBea/hcxdumptool #66: just found authorized handshake EAPOL timeout 3866

khanfar, 2019-07-26:
when try command : hcxdumptool -o hash -i wlomon --filterlist=filter.txt --filtermode=3 --enable_status=1,2,4,8   
i never got PMKID , just got ( found authorization handshake eapol time out 6318 
whats wrong i do ?
i install hash cat in my ubuntu 18 using this command : 
sudo apt install hashcat
thats it for hashcat installation ?? 
when i try to cd hash cat i cant find it ! i type first in terminal echo "bssid">filter.txt after that do command hcxdumptool up , 
Does need first to make echo filter.txt in hashcat first by doing : CD hashcat ???
i cant find hashcat after installing it by command up ! whats im missing ???

ZerBea, 2019-07-26:
1. identify your target.
2. add your target mac to filterlist.txt (use a texteditor - format of mac: 112233445566)
3. run hcxdumptool ($ hcxdumptool -o dump.pcapng -i interface --filterlist=filter.list --filtermode=3 --enable_status=15
we are using a bitmask, so 1+2+4+8 = 15
keep in mind: not every router transmit a PMKID
your driver must support monitor mode and full packet injection
do not use hcxdumptool in combination with airmon-ng

to test if every thing is working like expected and target is in range run rcascan:
hcxdumptool -i interface --do_rcascan
and again: do not(!) run hcxdumdptool on a monitor interface created by airmon-ng! 

4. convert pcapng to hascat format ($ hcxpcaptool -z pmkid.16800 dump.pcapng)
5. run hashcat ($hascat -m 16800 pmkid.16800 wordlist

read help menu and README.md
hashcat require a driver and opencl runtime (read hashcat driver wiki for hashcat related questions)

Closed this, because issue is not related to hcxdumptool, but you can still ask questions here.

closed 2019-07-26 by ZerBea

ZerBea, 2019-07-27:
Some words about the do_rcascan option:
Running this option for at least 30 seconds, hcxdumptool test
that the driver supports all required ioctl() calls and
that the driver support full packet injection.
Also you are informed, how many access points are received and
how many access points are in range.

So it's a good idea to run --do_rcascan first, to retrieve all this information.
There are many tutorials in which airmon-ng and/or airodump-ng is used to get this information.
Be warned that this will lead to unwanted results!