ZerBea/hcxdumptool #68: FreeBSD Support?

danielsokolowski, 2019-08-20:
I apologize if I missed the obvious... could someone advise if hcxdumptool supports FreeBSD and if so where could I find some information on this?

ZerBea, 2019-08-20:
No, you don't miss something. Native, hcxdumptool support only Arch Linux and nothing will change that. I'm tired to wrestle against other operating systems/distributions, their drivers and their default configuration.

So, other Linux distributions are not supported, but they should work, too , depending on the installed drivers (driver must support monitor mode and full packet injection). 
The same applies to FreeBSD. It isn't supported, but it may work, too:
https://forums.freebsd.org/threads/aircrack-ng-monitor-mode-patch-for-freebsd.10042/
So, to get hcxdumptool work on FreeBSD you have to do some hands-on work on the driver and or hcxdumptool.

ZerBea, 2019-08-23:
How about requesting hcxdumptool features on aircrack-ng and/or bettercap?
https://github.com/aircrack-ng/rtl8812au/issues/376#issuecomment-524198833

closed 2019-08-23 by ZerBea