ZerBea/hcxdumptool #69: Not Working

HeIp, 2019-08-31:
Trying to run hcxdumptool on kali, tried various wifi adapters(ALFA) all resulting in similar errors.

root@kali:~# hcxdumptool -i wlan0 -o pmkid.pcapng --enable_status=1
initialization...
warning: NetworkManager is running with pid 480
warning: wpa_supplicant is running with pid 810
warning: failed to set channel 1 (No such device) - removed this channel from scan list
warning: failed to set channel 6 (No such device) - removed this channel from scan list
warning: failed to set channel 11 (No such device) - removed this channel from scan list
warning: failed to set channel 8 (No such device) - removed this channel from scan list
warning: failed to set channel 1 (No such device) - removed this channel from scan list
warning: failed to set channel 9 (No such device) - removed this channel from scan list
warning: failed to set channel 6 (No such device) - removed this channel from scan list
warning: failed to set channel 11 (No such device) - removed this channel from scan list
warning: failed to set channel 1 (No such device) - removed this channel from scan list
warning: failed to set channel 6 (No such device) - removed this channel from scan list
warning: failed to set channel 11 (No such device) - removed this channel from scan list
warning: failed to set channel 7 (No such device) - removed this channel from scan list

start capturing (stop with ctrl+c)
INTERFACE................: wlan0
ERRORMAX.................: 100 errors
FILTERLIST...............: 0 entries
MAC CLIENT...............: acde48ad817e
MAC ACCESS POINT.........: 78f944c5c8a8 (incremented on every new client)
EAPOL TIMEOUT............: 150000
REPLAYCOUNT..............: 64918
ANONCE...................: 2f5ae3d8c91ece0cd1d2e868c4f4dc8e4e0ce9205e748bc70cd69e0b5048bd00

failed to set channel
failed to get interface information: No such device
failed to set interface down: No such device
failed to restore old SIOCSIWMODE: No such device
failed to restore old SIOCSIFFLAGS and to bring interface up: No such device

terminated...
root@kali:~# hcxdumptool -i wlan0 -o pmkid.pcapng --enable_status=1
interface is not suitable
hcxdumptool need full (monitor mode and full packet injection running all packet types) and exclusive access to the adapter
that is not the case

Upon trying to run it again the last message just repeats.

ZerBea, 2019-09-01:
This is not a hcxdumptool issue. It is a driver issue and a system configuration issue:

You must stop this services, because they take access to the device!
warning: NetworkManager is running with pid 480
warning: wpa_supplicant is running with pid 810

You must read $ dmesg, to determine that you are not hit by this kernel issue:
https://bugzilla.kernel.org/show_bug.cgi?id=202541
If there is this message inside
"WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state"
you are hit by the issue and can't use that interface.

Your driver must support monitor mode and full packet injection.
A good how-to, how to check this is here:
https://hashcat.net/forum/thread-8599-post-45845.html#pid45845

Please run the following commands and post output (replace wlan0 with your interface name):
$ uname -r
$ hcxdumptool -I
$ sudo ip link set wlan0 down
$ sudo iw dev wlan0 set type monitor
$ sudo ip link set wlan0 up
$ sudo iw dev wlan0 info

ZerBea, 2019-09-03:
Closed, because issue is not related to hcxdumptool - but you can still ask questions here.

closed 2019-09-03 by ZerBea