ZerBea/hcxdumptool #70: Sending beacon frames with hcxdumptool

DaniloNC, 2019-09-05:
As far as I know most modern devices do not send probe requests to networks present on the PNL by default.

It would be useful if hcxdumptool had support to send beacon frames from a dictionary file, that way clients that had those network on their PNL would try to connect and hcxdumptool would be able to perform a new type of ap less attack. 

airbase-ng has similar functionality.

ZerBea, 2019-09-05:
Good idea. Should be easy to add, because hcxdumptool is already sending broadcast beacons:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L1458
Here we can add a function to use an ESSID from a file, too and transmit one or two beacons.
Sending more beacons doesn't make sense, because we spam the frequency with own beacons.

ZerBea, 2019-09-05:
Pushed an update:
Now hcxdumptool will transmit four kinds of beacons:
broadcast beacons
beacons from internal proberequest list
beacons from internal access point list (clone of an original beacon)
user defined beacons from external list

The behavior can be controlled by this options:
--disable_internal_beacons         : do not transmit beacons from internal beacon list
                                     affected: ap-less
--use_external_beaconlist=<file>   : transmit beacons from this list
                                     maximum ESSID length 32, maximum entries 4095
                                     affected: ap-less

please test and close issue if it works.

DaniloNC, 2019-09-06:
Wow, that was fast!
I have made some tests and it seems to be working great!

Thank You!

ZerBea, 2019-09-06:
I pushed another update:
Now we tell the clients our beacon interval is 1 beacon/second, we use channels 1...13, allow WPA1 and WPA2  connections and have a WPS push button.
1 beacon/second is the limit, because we do not want to spam the frequency with our beacons.
That means, if you have 10 ESSIDs in your list, a single ESSID from this list is transmitted once every 10 seconds.

ZerBea, 2019-09-06:
Latest test results (hcxdumptool beaconing) are amazing:

best handshakes (total)..........: 61 (ap-less: 54)
best PMKIDs (total)..............: 5

best handshakes (total)..........: 87 (ap-less: 77)
best PMKIDs (total)..............: 14

and even though hcxdumptool ran less than 5 minutes:
best handshakes (total)..........: 5 (ap-less: 4)
best PMKIDs (total)..............: 15

I'm using a TP-LINK WN722N V1 for this tests
ID 0cf3:9271 Qualcomm Atheros Communications AR9271 802.11n

I think we can mark this feature request as solved and close it.

closed 2019-09-06 by ZerBea

ZerBea, 2019-09-07:
pushed another update - you will love it:
hcxdumptool: added new option
--reactive_beacon                  : send internal/external beacon on every received proberequest
                                     affected: ap-less

DaniloNC, 2019-09-07:
Sweet. :-)

Something I was noticing, should the beacon frames (or at least the external ones) respect the `--ap_mac` option?

ZerBea, 2019-09-07:
--ap_mac is a dangerous option, because it can lead to a  huge ESSID mess!
If you choose a VENDOR assigned MAC it is possible that an access point with the same MAC
and a different ESSID exists in range. Result is a single MAC with 2 ESSIDs. hashcat will fail to calculate the correct PBKDF2 on one of them.

The beacon frames respect --ap_mac except 2 bits: 
The first octet LSB bit determines if the mac is a unicast (0) or multicast (1)
The first octet Second LSB bit determines if the mac is global unique (0) or locally administered (1)
https://en.wikipedia.org/wiki/MAC_address#/media/File:MAC-48_Address.svg
So we must mask this bits here:
https://github.com/ZerBea/hcxdumptool/blob/master/hcxdumptool.c#L5036
Having those bits set can raise suspicion, the locally administered sometimes are used by enterprise access points though.

ZerBea, 2019-09-07:
hcxpcaptool will inform you if such an ESSID mess is detected if you choose option
$ hcxpcaptool-o test.hccapx --max-essid-changes=5 test.pcapng
xxx handshake(s) written to test.hccapx (warning: ESSID changes detected)

LowSkillGamer, 2019-10-30:
I understand correctly? If I turn on _--reactive_beacon_ then there is a chance to get more ap-less points, but I do not get the BSSID of these points. I'm right?

ZerBea, 2019-10-30:
Yes, that is correct. We make some noise on the channel and hope to get an answer.