ZerBea/hcxdumptool #71: segfault on raspberry pi when using --save_rcascan=

ukscone, 2019-09-07:
Tried both the prebuilt version in the kali arm repos (i know that it says that is for Arch but i've had it running on lots of things & don't think this problem is distro related) and also self-built on kali arm and raspbian (buster) and whenever i press ctrl-c to end the program with --save_rcascan=xxxxxx I get a segfault and an empty file.

ZerBea, 2019-09-07:
Thanks for reporting that issue:
==3800==ERROR: AddressSanitizer: heap-use-after-free on address 0x7fe67de2d81c at pc 0x7fe681007484 bp 0x7ffd5badc990 sp 0x7ffd5badc138
READ of size 32 at 0x7fe67de2d81c thread T0
    #0 0x7fe681007483 in __interceptor_memcmp /build/gcc/src/gcc/libsanitizer/sanitizer_common/sanitizer_common_interceptors.inc:825
    1 0x55de0df1a225 in sort_aplist_by_essid include/hcxdumptool.h:124
    2 0x7fe680da3066 in msort_with_tmp.part.0 (/usr/lib/libc.so.6+0x3c066)
    3 0x7fe680da2f14 in msort_with_tmp.part.0 (/usr/lib/libc.so.6+0x3bf14)
    4 0x7fe680da2f14 in msort_with_tmp.part.0 (/usr/lib/libc.so.6+0x3bf14)
    5 0x7fe680da2f14 in msort_with_tmp.part.0 (/usr/lib/libc.so.6+0x3bf14)
    6 0x7fe680da3323 in qsort_r (/usr/lib/libc.so.6+0x3c323)

It should be fixed by this commit:
https://github.com/ZerBea/hcxdumptool/issues/71

Please test and let me know if it is working for you. In that case, I'll release a new version.

ukscone, 2019-09-07:
will have a proper test in a little bit but quick couple of go's and it all seems fine. thanks
`INFO: cha=2, rx=599, rx(dropped)=0, tx=64, err=0, aps=35 (8 in range)
------------------------------------------------------------------------------
^C $HEX[] $HEX[]
terminated...`

ZerBea, 2019-09-08:
Ok, that's looking good. I'll wait a little bit before I push the new release. I have made many changes/inprovements (mainly on beacon behavior) since you reported this issue. They should be tested, too.

closed 2019-09-26 by ZerBea