ZerBea/hcxdumptool #72: raspberry pi 4 with kali linux, not work with integrated wifi card.

LowSkillGamer, 2019-09-26:
raspberry pi 4 
kali linux 2019.3a
brcmfmac (Broadcom 43430)

I try install hcxdumptool from git and from kali rep, but any version hcxdumptool does not work with integrated wifi card.
It is easy to switch to monitor mode and airodump-ng with aireplay-ng (injection test) works fine. But hcxdumptool not work. hcxdumptool just does not see any points and any packages.

It perfectly works with an external wifi adapter.
Maybe I'm doing something wrong? Or is it all about hcxdumptool?

LowSkillGamer, 2019-09-26:
I noticed that if you put wlan0 itself in monitoring mode, the interface does not find points in airodump-ng at all, and if you use the "monstart" command and wlan0mon is created in parallel. That wlan0mon works with airodump-ng. This is so if someone suddenly wants to check or solve a problem.

ZerBea, 2019-09-26:
The Raspberry Pi's embedded WiFi chipset does not support promiscuous mode, because Broadcom never released a patch to enable monitor mode.
As far as I know, KALI is running a wrapper (NEXMON) and this wrapper doesn't support all ioctl() system calls. 

Please add output of 
$ hcxdumptool -I
$ hcxdumptool -i "your interface" --check_driver

LowSkillGamer, 2019-09-26:
[image]
[image]

ZerBea, 2019-09-26:
First screen shot:
It looks like ioctl() system calls are working. Now we check if packet injection is working:
Please add the output of
$ hcxdumptool -i wlan0 --do_rcascan
run it at least for 2 minutes.

second screen shot:
Do not run hcxdumptool on monitor interfaces (wlan0mon) created by third party tools. That will not work.

ZerBea, 2019-09-26:
Pushed an update to detect the presence of a valid radiotap header. Please use latest git head for
$ hcxdumptool -i wlan0 --do_rcascan

ZerBea, 2019-09-27:
I think I know why we do not receive packets. NETLINK (NL80211_IFTYPE_MONITOR) is in use: https://aur.archlinux.org/cgit/aur.git/tree/add-nexmon.patch?h=linux-aarch64-raspberrypi-4.14
hcxdumptool doesn't work on virtual NETLINK interfaces.

LowSkillGamer, 2019-09-27:
do_rcascan on wlan0
[image]

iw dev
[image]

I try capture packets via tshark and tcpdump, pcapng and cap was empty.

And as I said before, airodump-ng only works when wlan0mon is created.
[image]

It seems you are right, this is not a real monitor mode.

ZerBea, 2019-09-27:
Correct. We have two problems here:
1) wlan0 isn't in a real monitor mode. Instead a wrapper is in use (NEXMON).
2) By NEXMON created interface is a virtual (fake) interface and it use AF_NETLINK

hcxdumptool  doesn't use NETLINK protocol because it has too much overhead and is too slow to perform some attack vectors.
We can close this issue report here, because it is related to the Broadcom driver.

closed 2019-09-27 by ZerBea