ZerBea/hcxdumptool #73: openssl evp.h error

help789, 2019-09-27:


help789, 2019-09-27:
[image]

help789, 2019-09-27:
sir what should i do

help789, 2019-09-27:
after make command i get this error

help789, 2019-09-27:
@ZerBea sir kindly reply sooner

LowSkillGamer, 2019-09-27:
> sir what should i do
Try install libssl1.1 and libssl-dev

sudo apt install libssl1.1 libssl-dev

ZerBea, 2019-09-28:
Step 1:
read README.md section requirements:
Requirements
libopenssl and openssl-dev installed

Step 2:
satisfy requirements /develop files) if they are not installed by your distributions as default. All Debian related distributions (KALI is one of them)  doesn't install develop files (-dev)!
Use the package management system of your distribution to install the dependencies.
BTW:
That and over thousands other things are the reason for me to use Arch Linux.

@LowSkillGamer thaks for your response

You canstill ask questions hrere, but I close this issue, because it is related to missing files of your distribution and not to hcxdumptool.

closed 2019-09-28 by ZerBea

ZerBea, 2020-03-12:
https://github.com/ZerBea/hcxdumptool/issues/101#issuecomment-598068155

akshat12233, 2020-05-11:
> > sir what should i do
> 
> Try install libssl1.1 and libssl-dev
> 
> **sudo apt install libssl1.1 libssl-dev**
> 
> If already installed, use
> **sudo apt install --reinstall libssl1.1 libssl-dev**

by using the commands in my terminal i m getting error 
which is 
unable to locate pacakage libssl-dev
please help....

strasharo, 2020-05-11:
What's the distribution you're using?

akshat12233, 2020-05-11:
I m using Kali linux please help

ZerBea, 2020-05-11:
The required package  is part of KALI:
https://pkg.kali.org/pkg/openssl
The recommended way to install it is this:
https://installlion.com/kali/kali/main/l/libssl-dev/install/index.html
If it doesn't work, your package management system is not updated or more likely broken and need to be updated or fixed:
https://superuser.com/questions/1410570/cant-install-anything-on-kali-linux-unable-to-locate-package

BTW:
KALI is not a distribution to be used by Linux beginners.
https://www.infoworld.com/article/3060814/should-beginners-install-kali-linux-on-their-computers.html

akshat12233, 2020-05-11:
Thanx bro i wil switch

ZerBea, 2020-05-11:
This will help new Linux users to pick their first distribution:
https://itsfoss.com/best-linux-beginners/

akshat12233, 2020-05-11:
Thankyou