ZerBea/hcxdumptool #76: gpsd setup

pakouri, 2019-10-31:
Hello!

I've noticed you dropped gpsd support in this version and opted to get data directly from a device. The issue would be that I never used a gps device I was using gpsd and sending data from my phone over to the daemon via a WiFi connection between the phone and my raspberry.

If I understand this correctly I won't be able to use my setup with this newer version. Is there a way to still make it work without needing to buy a gps device?

Have a good one!

ZerBea, 2019-10-31:
How are the GPS data send from the phone to the device? Which protocol is used?

ZerBea, 2019-10-31:
Read about sharing GPD data here:
https://www.jillybunch.com/sharegps/nmea-bluetooth-linux.html
hcxdumptool should be able to understand this

ZerBea, 2019-10-31:
Share GPS is an application for real-time location data sharing via a variety of methods. It also supports saving tracks via KML/KMZ, allowing the user to record footprints while the app is running in the background. Share GPS supports sending standard NMEA data over Bluetooth, USB, and TCP/IP.

ZerBea, 2019-10-31:
In this case your phone is acting as a GPS device and providing NMEA sentences. You can read them directly via cat /dev/rfcomm0 without running GPSD.
rfcoom can be found in nearly every distribution:
https://linux.die.net/man/1/rfcomm

ZerBea, 2019-10-31:
Please wait with the tests. Maybe I can add both options - we are on an early alfa version:
--use-gsp-device=<device>
--use gpsd
I'll check it and push an update, soon.

ZerBea, 2019-11-01:
By this commit:
https://github.com/ZerBea/hcxdumptool/commit/71aaa8b3f1cdcc4d515585b7e67ad2c991015858
we can get GPS data directly from the device (--use-gsp-device=device) or retrieve them running GPSD (--use_GPSD).
Please test and close issue, if it works.

BTW:
Old hcxpcaptool isn't able to handle NMEA sentences. I'm working on hcxpcaptool v.6.0.0 which can handle it. If everything is working like expected, I'll release both of them.

ZerBea, 2019-11-01:
I'm going to push another update to support GPSD again, but I'm not happy with this.
Sometime GPSD hangs:
$ gpsd -D4 /dev/ttyACM0
$ gpspipe -r
{"class":"VERSION","release":"3.19","rev":"3.19","proto_major":3,"proto_minor":14}
{"class":"DEVICES","devices":[]}
{"class":"WATCH","enable":true,"json":false,"nmea":true,"raw":0,"scaled":false,"timing":false,"split24":false,"pps":false}
... no GPS date here ...

That causes hcxdumptool to hang, too!!!
$ sudo hcxdumptool -i wlp3s0f0u2 --filtermode=1 --filterlist_ap=$HOME/Source/raspberry/filterlistap --filterlist_client=$HOME/Source/raspberry/filterlistap -o wpatst.pcapng --enable_status=127  --use_gpsd
initialization...
activating GPS...
connecting to GPSD...
^C
terminating...
No GPS data from GPSD, here and we must terminate!

I really suggest to read the data from the device directly.
That is working:
$ sudo hcxdumptool -i wlp3s0f0u2 --filtermode=1 --filterlist_ap=filterlistap --filterlist_client=filterlistap -o wpatst.pcapng --enable_status=127  --use_gps_device=/dev/ttyACM0
initialization...
activating GPS...

start capturing (stop with ctrl+c)
NMEA 0183 RMC SENTENCE..: 162227.00,A,5010.74927,N,00649.57091,E,0.......
INTERFACE NAME..........: wlp3s0f0u2
...