ZerBea/hcxdumptool #79: Question about very similar WiFi adapters (chipsets)

mato200, 2019-11-05:
Sry for posting a question here.

I am looking to buy my 1st adapter of this kind and i have 2 options:

RT3070
https://web.archive.org/web/20180201130205/https://wikidevi.com/wiki/Edimax_EW-7711UAn_v1

MT7601U
https://web.archive.org/web/20180201133258/https://wikidevi.com/wiki/Edimax_EW-7711UAn_v2

or

RT3070
https://web.archive.org/web/20180201130942/https://wikidevi.com/wiki/D-Link_DWA-127

MT7601U
https://web.archive.org/web/20180201142753/https://wikidevi.com/wiki/D-Link_DWA-127_rev_B1

They are extremely similar - same chipsets on both versions.
I have read that you support the Edimax one.
Will D-Link work as well because it uses the same chipset ?

ZerBea, 2019-11-05:
This is not because hcxdumptool support a driver. Instead the driver must support monitor mode and packet injection on the chipset. That isn't the case on all drivers.
Both drivers (rt2800usb and mt76) support monitor mode and full packet injection and both are working fine (independent of the VENDOR).
I'm running TENDA W311U+ on kernel 4.19.81
ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
ieee80211 phy7: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
as well as EDIMAX EW-7711UAN v2 on kernel 4.19.81
D 7392:7710 Edimax Technology Co., Ltd
mt7601u 5-3.1.2:1.0: ASIC revision: 76010001 MAC revision: 76010500
as well as some cheap nano devices (ALLNET and SEMPRE - both mt 7601u)

and TP-LINK Archer T2UH on kernel 5.3
ID 148f:761a Ralink Technology, Corp. MT7610U ("Archer T2U" 2.4G+5G WLAN Adapter)
usbcore: registered new interface driver mt76x0u

Kernel 4.19 doesn't support MT7610U and we have to wait for the next LTS Raspberry Pi kernel (perhaps 5.4).
Unfortunately some of my devices are hit by this kernel issue (kernel >= 4.20):
https://bugzilla.kernel.org/show_bug.cgi?id=202541
and we have to wait for a fix, here.

This are the device IDs lists. I'm sure, you'll find all mentioned devices inside of this lists:

Supported device IDs mt7601u:
linux-5.3.8/drivers/net/wireless/mediatek/mt7601u/usb.c
static const struct usb_device_id mt7601u_device_table[] = {
	{ USB_DEVICE(0x0b05, 0x17d3) },
	{ USB_DEVICE(0x0e8d, 0x760a) },
	{ USB_DEVICE(0x0e8d, 0x760b) },
	{ USB_DEVICE(0x13d3, 0x3431) },
	{ USB_DEVICE(0x13d3, 0x3434) },
	{ USB_DEVICE(0x148f, 0x7601) },
	{ USB_DEVICE(0x148f, 0x760a) },
	{ USB_DEVICE(0x148f, 0x760b) },
	{ USB_DEVICE(0x148f, 0x760c) },
	{ USB_DEVICE(0x148f, 0x760d) },
	{ USB_DEVICE(0x2001, 0x3d04) },
	{ USB_DEVICE(0x2717, 0x4106) },
	{ USB_DEVICE(0x2955, 0x0001) },
	{ USB_DEVICE(0x2955, 0x1001) },
	{ USB_DEVICE(0x2a5f, 0x1000) },
	{ USB_DEVICE(0x7392, 0x7710) },
	{ 0, }

Supported device IDs  rt2800usb:
linux-5.3.8/drivers/net/wireless/ralink/rt2x00/rt2800usb.c
static const struct usb_device_id rt2800usb_device_table[] = {
	/* Abocom */
	{ USB_DEVICE(0x07b8, 0x2870) },
	{ USB_DEVICE(0x07b8, 0x2770) },
	{ USB_DEVICE(0x07b8, 0x3070) },
	{ USB_DEVICE(0x07b8, 0x3071) },
	{ USB_DEVICE(0x07b8, 0x3072) },
	{ USB_DEVICE(0x1482, 0x3c09) },
	/* AirTies */
	{ USB_DEVICE(0x1eda, 0x2012) },
	{ USB_DEVICE(0x1eda, 0x2210) },
	{ USB_DEVICE(0x1eda, 0x2310) },
	/* Allwin */
	{ USB_DEVICE(0x8516, 0x2070) },
	{ USB_DEVICE(0x8516, 0x2770) },
	{ USB_DEVICE(0x8516, 0x2870) },
	{ USB_DEVICE(0x8516, 0x3070) },
	{ USB_DEVICE(0x8516, 0x3071) },
	{ USB_DEVICE(0x8516, 0x3072) },
	/* Alpha Networks */
	{ USB_DEVICE(0x14b2, 0x3c06) },
	{ USB_DEVICE(0x14b2, 0x3c07) },
	{ USB_DEVICE(0x14b2, 0x3c09) },
	{ USB_DEVICE(0x14b2, 0x3c12) },
	{ USB_DEVICE(0x14b2, 0x3c23) },
	{ USB_DEVICE(0x14b2, 0x3c25) },
[... 356 lines cut ...]
	{ USB_DEVICE(0x148F, 0x7610) },	/* MT7610U */
	{ USB_DEVICE(0x13B1, 0x003E) },	/* Linksys AE6000 */
	{ USB_DEVICE(0x0E8D, 0x7610) },	/* Sabrent NTWLAC */
	{ USB_DEVICE(0x7392, 0xa711) },	/* Edimax 7711mac */
	{ USB_DEVICE(0x7392, 0xb711) },	/* Edimax / Elecom  */
	{ USB_DEVICE(0x148f, 0x761a) },	/* TP-Link TL-WDN5200 */
	{ USB_DEVICE(0x148f, 0x760a) },	/* TP-Link unknown */
	{ USB_DEVICE(0x0b05, 0x17d1) },	/* Asus USB-AC51 */
	{ USB_DEVICE(0x0b05, 0x17db) },	/* Asus USB-AC50 */
	{ USB_DEVICE(0x0df6, 0x0075) },	/* Sitecom WLA-3100 */
	{ USB_DEVICE(0x2019, 0xab31) },	/* Planex GW-450D */
	{ USB_DEVICE(0x2001, 0x3d02) },	/* D-LINK DWA-171 rev B1 */
	{ USB_DEVICE(0x0586, 0x3425) },	/* Zyxel NWD6505 */
	{ USB_DEVICE(0x07b8, 0x7610) },	/* AboCom AU7212 */
	{ USB_DEVICE(0x04bb, 0x0951) },	/* I-O DATA WN-AC433UK */
	{ USB_DEVICE(0x057c, 0x8502) },	/* AVM FRITZ!WLAN USB Stick AC 430 */
	{ USB_DEVICE(0x293c, 0x5702) },	/* Comcast Xfinity KXW02AAA  */
	{ USB_DEVICE(0x20f4, 0x806b) },	/* TRENDnet TEW-806UBH  */
	{ USB_DEVICE(0x7392, 0xc711) }, /* Devolo Wifi ac Stick */
	{ USB_DEVICE(0x0df6, 0x0079) }, /* Sitecom Europe B.V. ac  Stick */
	{ USB_DEVICE(0x2357, 0x0105),
	  .driver_info = 1,	     }, /* TP-LINK Archer T1U */
	{ USB_DEVICE_AND_INTERFACE_INFO(0x0E8D, 0x7630, 0xff, 0x2, 0xff)}, /* MT7630U */
	{ USB_DEVICE_AND_INTERFACE_INFO(0x0E8D, 0x7650, 0xff, 0x2, 0xff)}, /* MT7650U */
	{ 0, }

ZerBea, 2019-11-06:
If the driver support monitor mode and packet injection via AF_PACKET (not AF_NETLINK !)
and your device is in the ID table of the driver code, hcxdumptool will work regardless of the manufacturer.

closed 2020-01-23 by ZerBea