ZerBea/hcxdumptool #80: hcxdumptool freezes after few seconds

LowSkillDeveloper, 2019-12-02:
Every few days I update all the utilities in my system. And after the last update, hcxdumptool stopped working.

That is, when I try to start as usual
`hcxdumptool -i wlan1 --reactive --enable_status 31 -o manual_a9.pcapng`

It works the first 5-10 seconds and freezes. I cannot close using Ctrl + C and the iwconfig command is not responding. Everyone wifi commands stops normal working. Everything becomes normal again when I disconnect the wifi adapter.

Having rolled back to the version from the kali repository, everything works fine there.

I checked these commands
```
hcxdumptool -I
hcxdumptool -i wlan1 --check_driver
hcxdumptool -i wlan1 --do_rcascan
```
Everything is fine here.

ZerBea, 2019-12-02:
Please add the output of 
$ hcxdumptool -I
and also the output of
$ dmesg

BTW:
iwconfig is deprecated and replaced by iw:
https://www.tecmint.com/deprecated-linux-networking-commands-and-their-replacements/
since a long time:
https://dougvitale.wordpress.com/2011/12/21/deprecated-linux-networking-commands-and-their-replacements/

LowSkillDeveloper, 2019-12-02:
This is on my RPi 4
[image]

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 4.19.81-Re4son-v8l+ (root@kali-pi-dev-v8) (gcc version 8.2.0 (Debian 8.2.0-12)) #1 SMP PREEMPT Wed Nov 6 07:24:49 UTC 2019
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] cma: Reserved 64 MiB at 0x0000000030000000
[    0.000000] On node 0 totalpages: 983040
[    0.000000]   DMA32 zone: 3328 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 212992 pages, LIFO batch:63
[    0.000000]   Normal zone: 12032 pages used for memmap
[    0.000000]   Normal zone: 770048 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xa0/0x464 with crng_init=0
[    0.000000] percpu: Embedded 23 pages/cpu s56728 r8192 d29288 u94208
[    0.000000] pcpu-alloc: s56728 r8192 d29288 u94208 alloc=23*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] CPU features: enabling workaround for EL2 vector hardening
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 967680
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:1B:41:FC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fix_enable=2 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait rootflags=noload net.ifnames=0
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] software IO TLB: mapped [mem 0x2aff5000-0x2eff5000] (64MB)
[    0.000000] Memory: 3713760K/3932160K available (8380K kernel code, 1070K rwdata, 2652K rodata, 1024K init, 1124K bss, 152864K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 29565 entries in 116 pages
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=64 to nr_cpu_ids=4.
[    0.000000] 	Tasks RCU enabled.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000218] Console: colour dummy device 80x25
[    0.000581] console [tty1] enabled
[    0.000631] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000668] pid_max: default: 32768 minimum: 301
[    0.000961] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.001025] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes)
[    0.002022] Disabling memory control group subsystem
[    0.024038] ASID allocator initialised with 32768 entries
[    0.032035] rcu: Hierarchical SRCU implementation.
[    0.040534] EFI services will not be available.
[    0.048091] smp: Bringing up secondary CPUs ...
[    0.080286] Detected PIPT I-cache on CPU1
[    0.080347] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.112370] Detected PIPT I-cache on CPU2
[    0.112413] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.144488] Detected PIPT I-cache on CPU3
[    0.144529] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.144668] smp: Brought up 1 node, 4 CPUs
[    0.144754] SMP: Total of 4 processors activated.
[    0.144776] CPU features: detected: 32-bit EL0 Support
[    0.145377] CPU: All CPU(s) started at EL2
[    0.145426] alternatives: patching kernel code
[    0.146699] devtmpfs: initialized
[    0.154625] Enabled cp15_barrier support
[    0.154674] Enabled setend support
[    0.155047] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.155088] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.156996] pinctrl core: initialized pinctrl subsystem
[    0.157644] DMI not present or invalid.
[    0.158068] NET: Registered protocol family 16
[... 254 lines cut ...]
[   70.586423] usbcore: registered new interface driver ath9k_htc
[   70.889643] usb 1-1.2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   71.150943] ath9k_htc 1-1.2:1.0: ath9k_htc: HTC initialized with 33 credits
[   71.691657] ath9k_htc 1-1.2:1.0: ath9k_htc: FW Version: 1.4
[   71.691692] ath9k_htc 1-1.2:1.0: FW RMW support: On
[   71.691714] ath: EEPROM regdomain: 0x0
[   71.691723] ath: EEPROM indicates default country code should be used
[   71.691730] ath: doing EEPROM country->regdmn map search
[   71.691742] ath: country maps to regdmn code: 0x3a
[   71.691752] ath: Country alpha2 being used: US
[   71.691760] ath: Regpair used: 0x3a
[   71.726250] ieee80211 phy1: Atheros AR9271 Rev:1
[   71.741112] brcmfmac: brcmf_cfg80211_reg_notifier: not an ISO3166 code (0x55 0x53)
[... cut ...]
[   73.172598] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   73.266435] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   87.924297] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   87.942586] device wlan1 entered promiscuous mode
[  157.529063] usb 1-1.2: USB disconnect, device number 3
[  158.017334] ath: phy1: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xa038160e
[  158.028689] device wlan1 left promiscuous mode
[  158.112001] usb 1-1.2: ath9k_htc: USB layer deinitialized

```

LowSkillDeveloper, 2019-12-02:
This is on my Virtualbox on PC
```

[    0.000000] Linux version 5.3.0-kali2-amd64 (devel@kali.org) (gcc version 9.2.1 20191109 (Debian 9.2.1-19)) #1 SMP Debian 5.3.9-3kali1 (2019-11-20)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-5.3.0-kali2-amd64 root=UUID=9a39da71-7b4e-43d8-903d-f971653ba1f4 ro initrd=/install/gtk/initrd.gz quiet splash
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000dffeffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dfff0000-0x00000000dfffffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000019fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.5 present.
[    0.000000] DMI: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000000] kvm-clock: cpu 0, msr 136c14001, primary cpu clock
[    0.000000] kvm-clock: using sched offset of 36388832198673 cycles
[    0.000001] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000003] tsc: Detected 2808.002 MHz processor
[    0.001606] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.001607] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.001610] last_pfn = 0x1a0000 max_arch_pfn = 0x400000000
[    0.001619] MTRR default type: uncachable
[    0.001619] MTRR variable ranges disabled:
[    0.001620] Disabled
[    0.001621] x86/PAT: MTRRs disabled, skipping PAT initialization too.
[    0.001622] CPU MTRRs all blank - virtualized system.
[    0.001624] x86/PAT: Configuration [0-7]: WB  WT  UC- UC  WB  WT  UC- UC  
[    0.001627] last_pfn = 0xdfff0 max_arch_pfn = 0x400000000
[    0.085823] BRK [0x136e01000, 0x136e01fff] PGTABLE
[    0.085825] BRK [0x136e02000, 0x136e02fff] PGTABLE
[    0.085825] BRK [0x136e03000, 0x136e03fff] PGTABLE
[    0.085843] BRK [0x136e04000, 0x136e04fff] PGTABLE
[    0.085844] BRK [0x136e05000, 0x136e05fff] PGTABLE
[    0.085889] BRK [0x136e06000, 0x136e06fff] PGTABLE
[    0.085897] BRK [0x136e07000, 0x136e07fff] PGTABLE
[    0.085905] BRK [0x136e08000, 0x136e08fff] PGTABLE
[    0.085920] BRK [0x136e09000, 0x136e09fff] PGTABLE
[    0.085962] RAMDISK: [mem 0x30167000-0x340aafff]
[    0.085970] ACPI: Early table checksum verification disabled
[    0.085972] ACPI: RSDP 0x00000000000E0000 000024 (v02 VBOX  )
[    0.085975] ACPI: XSDT 0x00000000DFFF0030 00003C (v01 VBOX   VBOXXSDT 00000001 ASL  00000061)
[    0.085978] ACPI: FACP 0x00000000DFFF00F0 0000F4 (v04 VBOX   VBOXFACP 00000001 ASL  00000061)
[    0.085981] ACPI: DSDT 0x00000000DFFF0480 0022EA (v02 VBOX   VBOXBIOS 00000002 INTL 20100528)
[    0.085982] ACPI: FACS 0x00000000DFFF0200 000040
[    0.085984] ACPI: FACS 0x00000000DFFF0200 000040
[    0.085985] ACPI: APIC 0x00000000DFFF0240 00006C (v02 VBOX   VBOXAPIC 00000001 ASL  00000061)
[    0.085986] ACPI: SSDT 0x00000000DFFF02B0 0001CC (v01 VBOX   VBOXCPUT 00000002 INTL 20100528)
[    0.085991] ACPI: Local APIC address 0xfee00000
[    0.086165] No NUMA configuration found
[    0.086166] Faking a node at [mem 0x0000000000000000-0x000000019fffffff]
[    0.086168] NODE_DATA(0) allocated [mem 0x19fff9000-0x19fffdfff]
[    0.086185] Zone ranges:
[    0.086186]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.086187]   DMA32    [mem 0x0000000001000000-0x00000000ffffffff]
[    0.086187]   Normal   [mem 0x0000000100000000-0x000000019fffffff]
[    0.086188]   Device   empty
[    0.086188] Movable zone start for each node
[    0.086189] Early memory node ranges
[    0.086189]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[... 426 lines cut ...]
[   55.908580] usb 1-2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[   55.908581] usb 1-2: Product: UB93
[   55.908582] usb 1-2: Manufacturer: ATHEROS
[   55.908583] usb 1-2: SerialNumber: 12345
[   55.968620] usb 1-2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.dev.0.fw requested
[   55.968731] usbcore: registered new interface driver ath9k_htc
[   55.970813] usb 1-2: firmware: direct-loading firmware ath9k_htc/htc_9271-1.dev.0.fw
[   56.259048] usb 1-2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.dev.0.fw, size: 51008
[   56.512815] ath9k_htc 1-2:1.0: ath9k_htc: HTC initialized with 33 credits
[   56.789472] ath9k_htc 1-2:1.0: ath9k_htc: FW Version: 1.4
[   56.789474] ath9k_htc 1-2:1.0: FW RMW support: On
[   56.789475] ath: EEPROM regdomain: 0x0
[   56.789475] ath: EEPROM indicates default country code should be used
[   56.789476] ath: doing EEPROM country->regdmn map search
[   56.789476] ath: country maps to regdmn code: 0x3a
[   56.789477] ath: Country alpha2 being used: US
[   56.789477] ath: Regpair used: 0x3a
[   56.796614] ieee80211 phy0: Atheros AR9271 Rev:1
[   87.158832] device wlan0 entered promiscuous mode
[   90.710576] ath: phy0: Unable to set channel
[  132.137724] usb 1-2: USB disconnect, device number 3
[  132.249860] device wlan0 left promiscuous mode
[  132.307754] usb 1-2: ath9k_htc: USB layer deinitialized

```

LowSkillDeveloper, 2019-12-02:
Lol, looking at the log, I realized that the problem with wireless-regdb. I tweaked the TX Power a bit.

But it’s strange why there is no problem on the old version from the kali repository. And wifite with bettercap work fine.

LowSkillDeveloper, 2019-12-02:
I tried another adapter on same chipset. And this not works.
but wifite and bettercap works fine.

And if you start hcxdumptool first, and then even if you manage to stop it, then wifite, etc. Do not see anymore networks. 
This also applies to hcxdumptool. At the first start, it shows several PROBE RESPONSE. The next one has nothing at all.

LowSkillDeveloper, 2019-12-02:
hcxdumptool 5.2.2 works fine.

ZerBea, 2019-12-02:
Thanks for the information. dmesg show you two driver errors 

first one from here:
https://github.com/ZerBea/hcxdumptool/issues/80#issuecomment-560548942:
[   87.942586] device wlan1 entered promiscuous mode
[  157.529063] usb 1-1.2: USB disconnect, device number 3
[  158.017334] ath: phy1: Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL=0xa038160e
[  158.028689] device wlan1 left promiscuous mode

second one from here:
https://github.com/ZerBea/hcxdumptool/issues/80#issuecomment-560549179
[   87.158832] device wlan0 entered promiscuous mode
[   90.710576] ath: phy0: Unable to set channel
[  132.137724] usb 1-2: USB disconnect, device number 3
[  132.249860] device wlan0 left promiscuous mode

Make sure that NetworkManager doesn't have access to the device.
hcxdumptool showed you a warning that it interfers with NetworkManager.
We are not able to set a channel, because NetworkManager will prevent this
[   90.710576] ath: phy0: Unable to set channel

Read more about how to prevent this, here:
How can I make NetworkManager ignore my wireless card?
https://askubuntu.com/questions/21914/how-can-i-make-networkmanager-ignore-my-wireless-card

The other issue is already reported on kernel.org:
https://bugzilla.kernel.org/show_bug.cgi?id=198701
and we have to wait for a fix.

ZerBea, 2019-12-02:
Here are my logs (NetworkManager doesn't have access to the device):
$ hcxdumptool -v
hcxdumptool 6.0.0 (C) 2019 ZeroBeat

$ hcxdumptool -I
wlan interfaces:
f81a67077d0e wlp39s0f3u3u1u2 (ath9k_htc)

dmesg log:
[18802.485520] usb 5-3.1.2: new high-speed USB device number 7 using xhci_hcd
[18802.695446] usb 5-3.1.2: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[18802.695451] usb 5-3.1.2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[18802.695454] usb 5-3.1.2: Product: USB2.0 WLAN
[18802.695457] usb 5-3.1.2: Manufacturer: ATHEROS
[18802.695459] usb 5-3.1.2: SerialNumber: 12345
[18802.888604] usb 5-3.1.2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[18802.888663] usbcore: registered new interface driver ath9k_htc
[18803.180475] usb 5-3.1.2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[18803.430744] ath9k_htc 5-3.1.2:1.0: ath9k_htc: HTC initialized with 33 credits
[18803.660868] ath9k_htc 5-3.1.2:1.0: ath9k_htc: FW Version: 1.4
[18803.660872] ath9k_htc 5-3.1.2:1.0: FW RMW support: On
[18803.660874] ath: EEPROM regdomain: 0x809c
[18803.660875] ath: EEPROM indicates we should expect a country code
[18803.660876] ath: doing EEPROM country->regdmn map search
[18803.660877] ath: country maps to regdmn code: 0x52
[18803.660879] ath: Country alpha2 being used: CN
[18803.660880] ath: Regpair used: 0x52
[18803.664696] ieee80211 phy3: Atheros AR9271 Rev:1
[18803.667298] ath9k_htc 5-3.1.2:1.0 wlp39s0f3u3u1u2: renamed from wlan0
[18803.673533] audit: type=1130 audit(1575322972.184:419): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[19082.951881] device wlp39s0f3u3u1u2 entered promiscuous mode
[19082.951967] audit: type=1700 audit(1575323251.467:425): dev=wlp39s0f3u3u1u2 prom=256 old_prom=0 auid=1000 uid=0 gid=0 ses=2
[19096.731559] device wlp39s0f3u3u1u2 left promiscuous mode

and hcxdumptool works like expected - until I get hit by the diver issue.

BTW
This caused the freeze:
"Timeout while waiting for nf to load: AR_PHY_AGC_CONTROL"
Just do a google search and you will find many, many issue reports.

LowSkillDeveloper, 2019-12-02:
Okay, as I understand, we can only wait for the driver will be fixed?

And I recently ordered myself a Ralink RT3070. It will work fine with him?

ZerBea, 2019-12-03:
If you "blacklisted" the device by "NetworkManager config" and the driver issue is still present, we must wait for the kernel driver fix.

The RT3070 uses the rt2800usb driver. It will work fine if you make sure that NetworkManager can't access the device (by adding the device mac to NetworkManager config).
It will not work if you connect the device to an USB3 port especially on an AMD RYZEN motherboard. In that case you will run into this kernel issue (not fixed, yet):
https://bugzilla.kernel.org/show_bug.cgi?id=202541

Here is an example of this issue:
ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
dmesg log:
[   43.907984] usb 1-2: new high-speed USB device number 6 using xhci_hcd
[   44.064919] usb 1-2: New USB device found, idVendor=148f, idProduct=3070, bcdDevice= 1.01
[   44.064928] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   44.064934] usb 1-2: Product: 802.11 n WLAN
[   44.064940] usb 1-2: Manufacturer: Ralink
[   44.064944] usb 1-2: SerialNumber: 1.0
[   44.725138] usb 1-2: reset high-speed USB device number 6 using xhci_hcd
[   44.876642] ieee80211 phy1: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
[   44.893423] ieee80211 phy1: rt2x00_set_rf: Info - RF chipset 0005 detected
[   44.894231] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   44.904724] usbcore: registered new interface driver rt2800usb
[   44.958162] audit: type=1130 audit(1575358746.711:35): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[   45.025041] rt2800usb 1-2:1.0 wlp0s20f0u2: renamed from wlan0
[   45.092238] ieee80211 phy1: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[   45.126266] ieee80211 phy1: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
[   45.435450] ieee80211 phy1: rt2x00usb_vendor_request: Error - Vendor Request 0x06 failed for offset 0x0404 with error -71
[   46.467853] ieee80211 phy1: rt2800_wait_csr_ready: Error - Unstable hardware
[   46.467866] ieee80211 phy1: rt2800usb_set_device_state: Error - Device failed to enter state 4 (-5)

The same device, connected to an USB2 port of the same notebook is working fine:
[ 1839.849738] usb 1-3: new high-speed USB device number 9 using xhci_hcd
[ 1840.008305] usb 1-3: New USB device found, idVendor=148f, idProduct=3070, bcdDevice= 1.01
[ 1840.008315] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1840.008321] usb 1-3: Product: 802.11 n WLAN
[ 1840.008326] usb 1-3: Manufacturer: Ralink
[ 1840.008331] usb 1-3: SerialNumber: 1.0
[ 1840.137020] usb 1-3: reset high-speed USB device number 9 using xhci_hcd
[ 1840.288016] ieee80211 phy4: rt2x00_set_rt: Info - RT chipset 3070, rev 0201 detected
[ 1840.300254] ieee80211 phy4: rt2x00_set_rf: Info - RF chipset 0005 detected
[ 1840.300882] ieee80211 phy4: Selected rate control algorithm 'minstrel_ht'
[ 1840.321054] rt2800usb 1-3:1.0 wlp0s20f0u3: renamed from wlan0
[ 1869.514883] ieee80211 phy4: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2870.bin'
[ 1869.514906] ieee80211 phy4: rt2x00lib_request_firmware: Info - Firmware detected - version: 0.36
[ 1869.769893] device wlp0s20f0u3 entered promiscuous mode
[ 1869.770056] audit: type=1700 audit(1575360571.519:168): dev=wlp0s20f0u3 prom=256 old_prom=0 auid=1000 uid=0 gid=0 ses=2
[ 1876.139841] device wlp0s20f0u3 left promiscuous mode
[ 1876.139872] audit: type=1700 audit(1575360577.889:169): dev=wlp0s20f0u3 prom=0 old_prom=256 auid=1000 uid=0 gid=0 ses=2
[ 1876.161908] audit: type=1106 audit(1575360577.909:170): pid=1404 uid=0 auid=1000 ses=2 msg='op=PAM:session_close grantors=pam_limits,pam_unix,pam_permit acct="root" exe="/usr/bin/sudo" hostname=? addr=? terminal=/dev/pts/1 res=success'

As of today and kernel 5.3.13 most of the drivers are not(!) working out of the box due to several driver issues (especially under heavy workload), depending on the hardware configuration (e.g. USB3, VENDOR). Or they don't support monitor mode.
If issues are reported on 
https://bugzilla.kernel.org
and/or
https://github.com/openwrt/mt76/issues
and for rtl8812au on
https://github.com/aircrack-ng/rtl8812au/issues

Some of them are fixed in latest kernel versions:
https://bugzilla.kernel.org/show_bug.cgi?id=202241
https://bugzilla.kernel.org/show_bug.cgi?id=202243
https://bugzilla.kernel.org/show_bug.cgi?id=205305
https://github.com/openwrt/mt76/issues/216#issuecomment-500999516
but many, many of them are still unfixed.

It it is very fragile and really hard work to get a driver working like expected (monitor mode inclusive full packet injection). A single update/commit can destroy the driver. Here is a good example
https://github.com/aircrack-ng/rtl8812au/issues/499

That is the reason, why I removed several adapters (formerly known as working) from the list of working devices:
https://github.com/ZerBea/hcxdumptool/wiki/WiFi-Adapters

ZerBea, 2019-12-03:
Some words about tx power, beside this ones here:
https://metis.fi/en/2017/10/txpower/

It is a fairytale that increasing tx power will lead to more results!
https://en.wikipedia.org/wiki/DBm
"A power level of 0 dBm corresponds to a power of 1 milliwatt. A 10 dB increase in level is equivalent to a 10-fold increase in power. A 3 dB increase in level is approximately equivalent to doubling the power, which means that a level of 3 dBm corresponds roughly to a power of 2 mW. Similarly, for each 3 dB decrease in level, the power is reduced by about one half, making −3 dBm correspond to a power of about 0.5 mW. "

A good antenna is the best hf amplifier:
https://www.arrl.org/files/file/Technology/tis/info/pdf/9811054.pdf

Increasing tx power will make the signal crappy! A spectrum analyzer will show you this.

... and thousands of more good reasons.

ZerBea, 2019-12-04:
Just compiled kernel 5.4 and the driver issue is still present:
[ 1907.925136] usb 1-2: new high-speed USB device number 5 using xhci_hcd
[ 1908.182905] usb 1-2: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[ 1908.182910] usb 1-2: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[ 1908.182912] usb 1-2: Product: USB2.0 WLAN
[ 1908.182914] usb 1-2: Manufacturer: ATHEROS
[ 1908.182916] usb 1-2: SerialNumber: 12345
[ 1908.320074] usb 1-2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[ 1908.320126] usbcore: registered new interface driver ath9k_htc
[ 1909.396039] usb 1-2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[ 1909.646042] ath9k_htc 1-2:1.0: ath9k_htc: HTC initialized with 33 credits
[ 1909.873443] ath9k_htc 1-2:1.0: ath9k_htc: FW Version: 1.4
[ 1909.873446] ath9k_htc 1-2:1.0: FW RMW support: On
[ 1909.873448] ath: EEPROM regdomain: 0x809c
[ 1909.873449] ath: EEPROM indicates we should expect a country code
[ 1909.873450] ath: doing EEPROM country->regdmn map search
[ 1909.873451] ath: country maps to regdmn code: 0x52
[ 1909.873453] ath: Country alpha2 being used: CN
[ 1909.873454] ath: Regpair used: 0x52
[ 1909.877347] ieee80211 phy0: Atheros AR9271 Rev:1
[ 1909.879935] ath9k_htc 1-2:1.0 wlp3s0f0u2: renamed from wlan0
[ 2010.097396] device wlp3s0f0u2 entered promiscuous mode

The device entered promiscuous mode until we get hit by the xhci issue:
[ 2135.948226] device wlp3s0f0u2 left promiscuous mode
[ 2138.758824] usb 1-2: USB disconnect, device number 5
[ 2138.759204] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[ 2138.759316] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[ 2138.850646] audit: type=1130 audit(1575446536.167:90): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=systemd-rfkill comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[ 2138.882373] usb 1-2: ath9k_htc: USB layer deinitialized

Unfortunately 5.4 is a LTS kernel.

strasharo, 2019-12-04:
Any distro which has a non-buggy/fixed kernel by default currently?

ZerBea, 2019-12-04:
I don't suppose. Not even Arch (rolling release) has a fix.
Unfortunately some of this issues are difficult to prove. I'm hunting for the cause of that xhci issue for several weeks - without  success.
There is no warning and no error message. And if we have a warning or an error message it is different from time to time.
So we have to wait....
https://bugzilla.kernel.org/show_bug.cgi?id=202541#c102

strasharo, 2019-12-09:
Which is the latest kernel without the issue, 4.19 ?

ZerBea, 2019-12-09:
4.19.86 is running fine here:
https://archlinuxarm.org/packages/armv7h/linux-raspberrypi
as well as
https://www.archlinux.org/packages/core/x86_64/linux-lts/
Unfortunately 5.4.2 provides some nice features, I do not want to miss.

ZerBea, 2019-12-10:
Looks like we can expect an ath9k_htc fix on kernel 5.5. Two patches are merged:
https://bugzilla.kernel.org/show_bug.cgi?id=198701#c5
If it works like expected, we can expect that both patches are ported back to LTS-Kernels.

ZerBea, 2020-01-14:
I can reproduce that freeze:
https://hashcat.net/forum/thread-6661-post-47168.html#pid47168

Possible fix for that kernel issue: (another two patches commited):
https://hashcat.net/forum/thread-6661-post-47187.html#pid47187

ZerBea, 2020-01-23:
Now removed TP-LINK TL-WN722N v1 from the "device known as working list" due to driver issues.
https://github.com/ZerBea/hcxdumptool/commit/ca98009709424a9ea3eb8f7633efa33b9b0fec2c

closed 2020-01-23 by ZerBea

gonzabrusco, 2020-11-01:
Hello. This happened to me with latest Kali 2020.3. But when I updated to the kernel 5.8 it stopped happening. Maybe this got fixed?

ZerBea, 2020-11-01:
@gonzabrusco hello.
That depend on the driver. Unfortunately 99% of all reported issues are related to KALI, the driver or the firmware, eg.:
https://bugzilla.kernel.org/show_bug.cgi?id=207397

Please comment output of 
$ hcxdumptool -I
$ sudo hcxdumptool --check_driver
$ sudo hcxdumptool --check_injection
$ dmesg (all lines after you plugged in the device and start hcxdumptool)
Maybe we can find out, what changed (especially dmesg log will tell us this).

BTW
Kernel 5.8 reached EOL:
https://www.kernel.org/

gonzabrusco, 2020-11-03:
```
kali@kali:~$ sudo hcxdumptool -I
wlan interfaces:
e8de27a11847 wlan0 (ath9k_htc)

kali@kali:~$ sudo hcxdumptool -i wlan0 --check_driver
initialization...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...

kali@kali:~$ sudo hcxdumptool -i wlan0  --check_injection
initialization...
starting packet injection test (that can take up to two minutes)...
packet injection is working!

terminating...

DMESG OUTPUT:
[   41.393598] usb 1-1: new high-speed USB device number 2 using ehci-pci
[   41.767035] usb 1-1: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[   41.767042] usb 1-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[   41.767047] usb 1-1: Product: USB2.0 WLAN
[   41.767050] usb 1-1: Manufacturer: ATHEROS
[   41.767052] usb 1-1: SerialNumber: 12345
[   41.829607] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   41.830094] cfg80211: Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
[   41.830539] cfg80211: Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
[   41.830981] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   41.832367] platform regulatory.0: firmware: direct-loading firmware regulatory.db
[   41.832590] platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s
[   41.875475] usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[   41.877637] usbcore: registered new interface driver ath9k_htc
[   41.877675] usb 1-1: firmware: direct-loading firmware ath9k_htc/htc_9271-1.4.0.fw
[   42.178778] usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[   42.449469] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[   43.218798] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
[   43.218803] ath9k_htc 1-1:1.0: FW RMW support: On
[   43.218807] ath: EEPROM regdomain: 0x809c
[   43.218808] ath: EEPROM indicates we should expect a country code
[   43.218810] ath: doing EEPROM country->regdmn map search
[   43.218812] ath: country maps to regdmn code: 0x52
[   43.218814] ath: Country alpha2 being used: CN
[   43.218815] ath: Regpair used: 0x52
[   43.264732] ieee80211 phy0: Atheros AR9271 Rev:1
[   84.840218] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   84.869244] device wlan0 entered promiscuous mode
[   86.424281] device wlan0 left promiscuous mode
[  120.001071] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  120.012694] device wlan0 entered promiscuous mode
[  135.274403] device wlan0 left promiscuous mode
```

ZerBea, 2020-11-03:
Great, thanks. Everything is looking fine, now.
Driver test and injection test passing without issues.
Firmware loaded without issues: 1.4.0.fw
Regdomain is set to CN
Device is going into monitor mode: device wlan0 entered promiscuous mode
Device is leaving monitor mode when hcxdumptool finished: device wlan0 left promiscuous mode
Looks like the issues are fixed by kernel >= 5.8

ZerBea, 2020-11-03:
After some tests on 
```
$ uname -r
5.9.3-arch1-1
```
I still can't recommend an ath9k_htc interface. We are running into a kernel issue that caused the driver after a while to freeze:
```
[ 6326.578280] usb 1-2: Product: USB2.0 WLAN
[ 6326.578282] usb 1-2: Manufacturer: ATHEROS
[ 6326.578284] usb 1-2: SerialNumber: 12345
[ 6326.718208] usb 1-2: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[ 6326.718261] usbcore: registered new interface driver ath9k_htc
[ 6327.794881] usb 1-2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[ 6328.044267] ath9k_htc 1-2:1.0: ath9k_htc: HTC initialized with 33 credits
[ 6328.280918] ath9k_htc 1-2:1.0: ath9k_htc: FW Version: 1.4
[ 6328.280921] ath9k_htc 1-2:1.0: FW RMW support: On
[ 6328.280922] ath: EEPROM regdomain: 0x809c
[ 6328.280923] ath: EEPROM indicates we should expect a country code
[ 6328.280924] ath: doing EEPROM country->regdmn map search
[ 6328.280925] ath: country maps to regdmn code: 0x52
[ 6328.280926] ath: Country alpha2 being used: CN
[ 6328.280926] ath: Regpair used: 0x52
[ 6328.284655] ieee80211 phy0: Atheros AR9271 Rev:1
[ 6328.288461] ath9k_htc 1-2:1.0 wlp3s0f0u2: renamed from wlan0
...
[ 6344.644087] device wlp3s0f0u2 entered promiscuous mode
...
[ 6488.849536] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[ 6488.849711] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[ 6488.850873] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[ 6489.171580] device wlp3s0f0u2 left promiscuous mode
```
This is a known issue and it is still unfixed:
https://bugzilla.kernel.org/show_bug.cgi?id=202541

gonzabrusco, 2020-11-03:
Fair enough. But reading the link you sent I come to the conclusion that this bug is not related to this adapter in particular. It seems like a bigger problem affecting several devices.

ZerBea, 2020-11-03:
Yes, you're right. That is an xhci (USB host) issue and it affect several devices.
The ath9k issue (driver and firmware) seems to be solved.

gonzabrusco, 2020-12-14:
Can I run hcxdumptool with kernel 4.19 ? To avoid xhci errors...

ZerBea, 2020-12-14:
There is no "general" answer. That depend on the chipset of the USB controller and the chipset of the WiFi device. Some are working, others not.
e.g.:
this one is not working:
`USB controller: ASMedia Technology Inc. ASM2142 USB 3.1 Host Controller in combination with ID 148f:5370 Ralink Technology, Corp. RT5370 Wireless Adapter
`
while this one is working:
`USB controller: ASMedia Technology Inc. ASM2142 USB 3.1 Host Controller in combination with ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
`
and this one is working
`USB controller: ASMedia Technology Inc. ASM2142 USB 3.1 Host Controller in combination with ID 7392:7710 Edimax Technology Co., Ltd Edimax Wi-Fi
`
while this one is not working on a Raspberry PI:
`ID 0b05:17d1 ASUSTek Computer, Inc. AC51 802.11a/b/g/n/ac Wireless Adapter [Mediatek MT7610U]
`

I prefer MT76 and RT2870/RT3070 chipsets, because they are working on all my systems (Desktop, Notebook and Raspberry PI
 "out of the box".

ZerBea, 2020-12-14:
And to make it clear: We are talking about kernel issues. There is no workaround for hcxdumptool to prevent this issue.

gonzabrusco, 2020-12-14:
I understand. My question was oriented to the workings of hcxdumptool. Is it compatible with Kernel 4.19? Because on the main readme you specify 5.4. Will it compile anyway? Thanks

ZerBea, 2020-12-14:
Ok, now I understand, too. Please forgive me. I'm old and my brain is a little bit slow.
Kernel 4.19 LTS is working fine - but you should know that some new drivers (and features) are not backported, yet.
Running new hardware, it is not a good idea to use an older kernel.
Luckily 5.10 was released today. It will be the new LTS kernel and I expect to see it on Arch Linux, soon. That include Arch Linux Arm - RaspBerry PI.

ZerBea, 2020-12-14:
In README.md I mentioned 5.4, because it is the first kernel with working mt76x0 drivers.

gonzabrusco, 2020-12-20:
Thanks @ZerBea 
Can you help me debug what's happening on my end? Because it seems something is crashing but I don't know what. Dmesg and hcxdumptool stop responding after some minutes. Nevertheless check_injection and check_driver work perfectly. I'm using latest Kali on top VMware with TP-LINK TL-WN722N v1. Don't think it a software problem because I can run airodump all night long and it does not crash.

ZerBea, 2020-12-20:
First of all: you can't compare airodump-ng with hcxdumptool because
airodump-ng is passive (doesn't transmsit)
and
hcxdumptool is active (transmsit).

To identify the issue open 2 terminals.
Than plug in your the TP-LINK TL-WN722N v1
Now start hcxdumptool in terminal 1:
$ hcxdumptool -i your_interface --enable_status=15
if hcxdumptool stops working run dmesg in terminal 2:
$ dmesg

We only need the line from dmesg log after you plugged in the device.
Output looks like this:
```
[15767.473292] usb 5-1.4: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[15767.724426] ath9k_htc 5-1.4:1.0: ath9k_htc: HTC initialized with 33 credits
[15767.953920] ath9k_htc 5-1.4:1.0: ath9k_htc: FW Version: 1.4
[15767.953923] ath9k_htc 5-1.4:1.0: FW RMW support: On
[15767.953924] ath: EEPROM regdomain: 0x809c
[15767.953925] ath: EEPROM indicates we should expect a country code
[15767.953926] ath: doing EEPROM country->regdmn map search
[15767.953927] ath: country maps to regdmn code: 0x52
[15767.953928] ath: Country alpha2 being used: CN
[15767.953929] ath: Regpair used: 0x52
[15767.957699] ieee80211 phy0: Atheros AR9271 Rev:1
[15767.967033] ath9k_htc 5-1.4:1.0 wlp39s0f3u1u4: renamed from wlan0
...
[15944.530389] device wlp39s0f3u1u4 entered promiscuous mode    <---<< at this point hcxdumptool is going into monitor mode
...
[15944.548822] device wlp39s0f3u1u4 left promiscuous mode <---<< at this point hcxdumptool left monitor mode
```

The lines between "entered promiscuous mode" and  "left promiscuous mode" are important, because they tell us what happened.

gonzabrusco, 2020-12-20:
The problem is that when hcxdumptool stops working, DMESG also stops responding. I have to eject the TL-WN722 to make it work again.
I'm using latest version of hcxdumptool compiled from the github.

This is the result (I freezes and I have to pull the usb wifi adapter at the end):

```
┌──(kali㉿kali)-[~/Desktop]
└─$ sudo hcxdumptool -i wlan0 --check_injection                                                                                                        1 ⨯
initialization...
starting packet injection test (that can take up to two minutes)...
packet injection is working!
ratio: 217 to 117 

terminating...
                                                                                                                                                           
┌──(kali㉿kali)-[~/Desktop]
└─$ sudo hcxdumptool -i wlan0 --check_driver   
initialization...
starting driver test...
driver tests passed...
all required ioctl() system calls are supported by driver

terminating...
                                                                                                                                                                                                                                 
┌──(kali㉿kali)-[~/Desktop]
└─$ sudo hcxdumptool -o dump.pcapng -i wlan0 --enable_status=15                                                                                        1 ⨯
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE HARDWARE MAC....: e8de27a11847
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.9.0-kali1-amd64
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 001761182dd7 (BROADCAST HIDDEN)
ACCESS POINT (ROGUE)......: 001761182dd8 (BROADCAST OPEN)
ACCESS POINT (ROGUE)......: 001761182dd9 (incremented on every new client)
CLIENT (ROGUE)............: b025aa9831d0
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 65489
ANONCE....................: a3d387c6f33fe998fcc99fe6816a7a7dd43570dbc5a1d45c2e259d012e2a9447
SNONCE....................: aa1620fe493d39abf3d377b3a18ba91fd0db7c641bb0b951a5cd8b1f6ba827e8

08:36:19   1 ffffffffffff 28be9b9e170f Carilo A9 [BEACON]
08:36:19   1 ffffffffffff 24a43ca00dee CARILO_HOUSE [BEACON]
08:36:19   1 ffffffffffff 922aa8b3bfb2 [HIDDEN BEACON]
08:36:19   1 ffffffffffff 6014b39a4a60 Fibertel WiFi187 2.4GHz [BEACON]
08:36:19   1 ffffffffffff 822aa8b3bfb2 Seabreeze [BEACON]
08:36:19   1 ffffffffffff 788a2024420d Edificio Playas [BEACON]
08:36:19   1 ffffffffffff 14cc20b54db3 Renata 2.4GHz [BEACON]
08:36:19   1 34f39a07a7ec 14cc20b54db3 Renata 2.4GHz [PROBERESPONSE]
08:36:19   1 34f39a07a7ec 14cc20b54db3 Renata 2.4GHz [AUTHENTICATION]
08:36:19   1 34f39a07a7ec 14cc20b54db3 Renata 2.4GHz [REASSOCIATION]
08:36:19   1 34f39a07a7ec 14cc20b54db3 Renata 2.4GHz [EAPOL:M2M3 EAPOLTIME:1687 RC:2 KDV:2]
08:36:19   1 ffffffffffff 6014b3d7d3d0 Fibertel WiFi935 2.4GHz [BEACON]
08:36:19   1 34f39a07a7ec 14cc20b54db3 Renata 2.4GHz [EAPOL:M1M2 EAPOLTIME:786 RC:1 KDV:2]
08:36:19   1 34f39a07a7ec 14cc20b54db3 Renata 2.4GHz [EAPOL:M2M3 EAPOLTIME:2499 RC:2 KDV:2]
08:36:19   1 34f39a07a7ec 14cc20b54db3 Renata 2.4GHz [EAPOL:M3M4ZEROED EAPOLTIME:684 RC:2 KDV:2]
08:36:19   1 ffffffffffff ac84c6b3e988 Torrecillas VII [BEACON]
08:36:19   1 ffffffffffff ac84c6b3c642 Torre Dptos 8 y 9  [BEACON]
[... 53 lines cut ...]
[  150.652106] usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[  150.936301] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[  152.284272] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
[  152.284277] ath9k_htc 1-1:1.0: FW RMW support: On
[  152.284281] ath: EEPROM regdomain: 0x809c
[  152.284282] ath: EEPROM indicates we should expect a country code
[  152.284284] ath: doing EEPROM country->regdmn map search
[  152.284285] ath: country maps to regdmn code: 0x52
[  152.284288] ath: Country alpha2 being used: CN
[  152.284289] ath: Regpair used: 0x52
[  152.326759] ieee80211 phy0: Atheros AR9271 Rev:1
[  177.967183] device wlan0 entered promiscuous mode
[  201.745498] device wlan0 left promiscuous mode
[  217.211528] device wlan0 entered promiscuous mode
[  217.885467] device wlan0 left promiscuous mode
[  241.211218] device wlan0 entered promiscuous mode
[  291.270322] ath: phy0: Unable to set channel
[  416.400041] usb 1-1: USB disconnect, device number 2
[  416.722291] ath: phy0: Chip reset failed
[  416.722295] ath: phy0: Unable to reset channel (2437 Mhz) reset status -22
[  416.722398] ath: phy0: Unable to set channel
[  416.727415] device wlan0 left promiscuous mode
```

Looks like the problem is "ath: phy0: Unable to set channel"

ZerBea, 2020-12-20:
Something disconnect your USB device permanently and hcxdumptool retry to set monitor mode
```
[  177.967183] device wlan0 entered promiscuous mode
[  201.745498] device wlan0 left promiscuous mode
[  217.211528] device wlan0 entered promiscuous mode
[  217.885467] device wlan0 left promiscuous mode
[  241.211218] device wlan0 entered promiscuous mode
```
At least hcxdumptool give up and print an error message that your network went down.

Now you have to find out, what exactly disconnect your device USB device. I assume it is your VM or a tool which take access to the device.

gonzabrusco, 2020-12-20:
Those are because I run the check_driver and check_injection before running your command.

gonzabrusco, 2020-12-20:
This is the result again without running those commands before:

```
[  136.376382] usb 1-1: new high-speed USB device number 2 using ehci-pci
[  136.759571] usb 1-1: New USB device found, idVendor=0cf3, idProduct=9271, bcdDevice= 1.08
[  136.759573] usb 1-1: New USB device strings: Mfr=16, Product=32, SerialNumber=48
[  136.759575] usb 1-1: Product: USB2.0 WLAN
[  136.759576] usb 1-1: Manufacturer: ATHEROS
[  136.759576] usb 1-1: SerialNumber: 12345
[  136.791615] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[  136.791846] cfg80211: Loaded X.509 cert 'benh@debian.org: 577e021cb980e0e820821ba7b54b4961b8b4fadf'
[  136.792011] cfg80211: Loaded X.509 cert 'romain.perier@gmail.com: 3abbc6ec146e09d1b6016ab9d6cf71dd233f0328'
[  136.792155] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[  136.792976] platform regulatory.0: firmware: direct-loading firmware regulatory.db
[  136.793179] platform regulatory.0: firmware: direct-loading firmware regulatory.db.p7s
[  136.821121] usb 1-1: ath9k_htc: Firmware ath9k_htc/htc_9271-1.4.0.fw requested
[  136.821455] usbcore: registered new interface driver ath9k_htc
[  136.822221] usb 1-1: firmware: direct-loading firmware ath9k_htc/htc_9271-1.4.0.fw
[  137.147884] usb 1-1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[  137.429543] ath9k_htc 1-1:1.0: ath9k_htc: HTC initialized with 33 credits
[  138.526682] ath9k_htc 1-1:1.0: ath9k_htc: FW Version: 1.4
[  138.526686] ath9k_htc 1-1:1.0: FW RMW support: On
[  138.526690] ath: EEPROM regdomain: 0x809c
[  138.526691] ath: EEPROM indicates we should expect a country code
[  138.526693] ath: doing EEPROM country->regdmn map search
[  138.526694] ath: country maps to regdmn code: 0x52
[  138.526697] ath: Country alpha2 being used: CN
[  138.526698] ath: Regpair used: 0x52
[  138.558945] ieee80211 phy0: Atheros AR9271 Rev:1
[  161.020664] device wlan0 entered promiscuous mode
[  431.116346] usb 1-1: USB disconnect, device number 2
[  431.534080] device wlan0 left promiscuous mode
[  431.591706] usb 1-1: ath9k_htc: USB layer deinitialized
```
```

┌──(kali㉿kali)-[~]
└─$ sudo hcxdumptool -o dump.pcapng -i wlan0 --enable_status=15 
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE HARDWARE MAC....: e8de27a11847
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.9.0-kali1-amd64
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 706f8122668f (BROADCAST HIDDEN)
ACCESS POINT (ROGUE)......: 706f81226690 (BROADCAST OPEN)
ACCESS POINT (ROGUE)......: 706f81226691 (incremented on every new client)
CLIENT (ROGUE)............: fcc233d3e44b
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 62538
ANONCE....................: 4cf03b79e5c90c168da58cf61f792037259a26b65d8b668ddfdb6fc205683993
SNONCE....................: 282dfed6c87e65a43ee73dd6335a7d6da7df930c80e821f3b26c520e3c269f5e

08:52:43   1 ffffffffffff 6014b3d7d3d0 Fibertel WiFi935 2.4GHz [BEACON]
08:52:43   1 ffffffffffff 28be9b9e170f Carilo A9 [BEACON]
08:52:43   1 ffffffffffff 6872513abd1d CARILO_HOUSE [BEACON]
08:52:43   1 ffffffffffff b0fc3686f570 Fibertel DRI C9 [BEACON]
08:52:43   1 ffffffffffff c025e9efa246 TP-LINK_A246 [BEACON]
[... 46 lines cut ...]
08:53:08  11 4ec18f57a7cf 6872513abca8 CARILO_HOUSE [PROBERESPONSE]
08:53:09  11 620f1abb069a 24a43ca00dee CARILO_HOUSE [PROBERESPONSE]
08:53:09  11 de60fdba3a5a 00156d9ee135 CARILO_HOUSE [PROBERESPONSE]
08:53:10  11 0022fe016108 6872513abc24 CARILO_HOUSE [PROBERESPONSE]
08:53:11  11 84100d4447fc 822aa8b3bfb2 Seabreeze [PROBERESPONSE]
08:53:13   2 ffffffffffff 6872515464a0 La Hosteria [BEACON]
08:53:13   2 6466b3221a29 6872515464a0 La Hosteria [PROBERESPONSE]
08:53:20   1 dc9fdb3403d0 00156d7ce740 Torrecillas [PROBERESPONSE]
08:53:38   9 dc9fdb3403d0 c025e9efa246 TP-LINK_A246 [PROBERESPONSE]
08:53:40   1 ffffffffffff 7683c2342926 [HIDDEN BEACON]
08:53:41   1 3cdcbce32015 68ff7b5a8936 laposta wifi [PROBERESPONSE]
08:53:57  10 ffffffffffff 50d4f793cc04 Sea Point [BEACON]
08:54:01   1 60f1894d08a0 6014b3d7d3d0 Fibertel WiFi935 2.4GHz [PROBERESPONSE]
08:54:01   1 60f1894d08a0 10feed0d32c2 Lemuhue WFPatio [PROBERESPONSE]
08:54:01   1 30074d139e2d 90671c865ee4 ARNET DRI [PROBERESPONSE]
08:54:03   1 ffffffffffff 7a8a20244331 [HIDDEN BEACON]
^C
terminating...
failed to get interface information: No such device
failed to set interface down: No such device
failed to restore old SIOCSIWMODE: No such device
failed to restore old SIOCSIFFLAGS and to bring interface up: No such device
```

It freezed again. So I had to pull the usb adapter again to be able to print dmesg.

ZerBea, 2020-12-20:
I can reproduce that running my TP-LINK TL--WN722N. After a while and under heavy load (as hcxdudmptool produce it), the driver died.
I'll do some further going investigations to find out why the driver died.

gonzabrusco, 2020-12-20:
I changed my virtual machine setting from usb 2.0 to usb 3.0 and now it's working (for now). I will let you know if it fails again. Thanks!

ZerBea, 2020-12-20:
That may one problem, but there is another one.

gonzabrusco, 2020-12-20:
It finally freezed again. But after much longer time. I don't know if it was a coincidence. This is dmesg (after pulling the usb adaptar):

```
[ 4601.294806] ath: phy0: Unable to set channel
[ 4835.221456] INFO: task vmtoolsd:510 blocked for more than 120 seconds.
[ 4835.221466]       Tainted: G            E     5.9.0-kali1-amd64 #1 Debian 5.9.1-1kali2
[ 4835.221469] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 4835.221472] task:vmtoolsd        state:D stack:    0 pid:  510 ppid:     1 flags:0x00000000
[ 4835.221479] Call Trace:
[ 4835.221494]  __schedule+0x281/0x8a0
[ 4835.221502]  schedule+0x4a/0xb0
[ 4835.221507]  schedule_preempt_disabled+0xa/0x10
[ 4835.221512]  __mutex_lock.constprop.0+0x13a/0x480
[ 4835.221518]  ? __netlink_lookup+0xb4/0x120
[ 4835.221523]  __netlink_dump_start+0xba/0x2d0
[ 4835.221529]  ? rtnl_fill_ifinfo+0x1290/0x1290
[ 4835.221534]  rtnetlink_rcv_msg+0x231/0x360
[ 4835.221539]  ? rtnl_fill_ifinfo+0x1290/0x1290
[ 4835.221544]  ? rtnl_calcit.isra.0+0x110/0x110
[ 4835.221548]  netlink_rcv_skb+0x47/0x110
[ 4835.221553]  netlink_unicast+0x1f9/0x2c0
[ 4835.221558]  netlink_sendmsg+0x243/0x480
[ 4835.221565]  sock_sendmsg+0x5e/0x60
[ 4835.221570]  __sys_sendto+0xee/0x150
[ 4835.221581]  ? exit_to_user_mode_prepare+0x32/0x140
[ 4835.221586]  __x64_sys_sendto+0x25/0x30
[ 4835.221591]  do_syscall_64+0x33/0x80
[ 4835.221597]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 4835.221603] RIP: 0033:0x7f4f558d535c
[ 4835.221605] Code: Bad RIP value.
[ 4835.221607] RSP: 002b:00007ffe28126780 EFLAGS: 00000293 ORIG_RAX: 000000000000002c
[ 4835.221611] RAX: ffffffffffffffda RBX: 00007ffe28127910 RCX: 00007f4f558d535c
[ 4835.221613] RDX: 0000000000000014 RSI: 00007ffe28127850 RDI: 0000000000000009
[ 4835.221615] RBP: 0000000000000000 R08: 00007ffe28127810 R09: 000000000000000c
[ 4835.221617] R10: 0000000000000000 R11: 0000000000000293 R12: 00007ffe28127810
[ 4835.221618] R13: 00007ffe28127850 R14: 000055d19727cc30 R15: 00007ffe281267c0
[ 4835.221657] INFO: task kworker/2:2:1468 blocked for more than 120 seconds.
[ 4835.221661]       Tainted: G            E     5.9.0-kali1-amd64 #1 Debian 5.9.1-1kali2
[ 4835.221664] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 4835.221668] task:kworker/2:2     state:D stack:    0 pid: 1468 ppid:     2 flags:0x00004000
[ 4835.221680] Workqueue: ipv6_addrconf addrconf_verify_work
[ 4835.221682] Call Trace:
[ 4835.221688]  __schedule+0x281/0x8a0
[ 4835.221694]  schedule+0x4a/0xb0
[ 4835.221698]  schedule_preempt_disabled+0xa/0x10
[ 4835.221703]  __mutex_lock.constprop.0+0x13a/0x480
[ 4835.221709]  ? __switch_to_asm+0x36/0x70
[ 4835.221715]  addrconf_verify_work+0xa/0x20
[ 4835.221721]  process_one_work+0x1b4/0x370
[ 4835.221726]  worker_thread+0x53/0x3e0
[ 4835.221730]  ? process_one_work+0x370/0x370
[ 4835.221733]  kthread+0x11b/0x140
[ 4835.221736]  ? __kthread_bind_mask+0x60/0x60
[ 4835.221741]  ret_from_fork+0x22/0x30
[ 4956.053234] INFO: task vmtoolsd:510 blocked for more than 241 seconds.
[ 4956.053244]       Tainted: G            E     5.9.0-kali1-amd64 #1 Debian 5.9.1-1kali2
[ 4956.053247] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 4956.053251] task:vmtoolsd        state:D stack:    0 pid:  510 ppid:     1 flags:0x00000000
[ 4956.053257] Call Trace:
[ 4956.053273]  __schedule+0x281/0x8a0
[ 4956.053280]  schedule+0x4a/0xb0
[ 4956.053285]  schedule_preempt_disabled+0xa/0x10
[ 4956.053346]  __mutex_lock.constprop.0+0x13a/0x480
[ 4956.053370]  ? __netlink_lookup+0xb4/0x120
[ 4956.053375]  __netlink_dump_start+0xba/0x2d0
[ 4956.053382]  ? rtnl_fill_ifinfo+0x1290/0x1290
[ 4956.053386]  rtnetlink_rcv_msg+0x231/0x360
[ 4956.053392]  ? rtnl_fill_ifinfo+0x1290/0x1290
[ 4956.053397]  ? rtnl_calcit.isra.0+0x110/0x110
[ 4956.053401]  netlink_rcv_skb+0x47/0x110
[... 16 lines cut ...]
[ 4956.053514] INFO: task kworker/2:2:1468 blocked for more than 241 seconds.
[ 4956.053519]       Tainted: G            E     5.9.0-kali1-amd64 #1 Debian 5.9.1-1kali2
[ 4956.053522] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 4956.053524] task:kworker/2:2     state:D stack:    0 pid: 1468 ppid:     2 flags:0x00004000
[ 4956.053536] Workqueue: ipv6_addrconf addrconf_verify_work
[ 4956.053538] Call Trace:
[ 4956.053544]  __schedule+0x281/0x8a0
[ 4956.053550]  schedule+0x4a/0xb0
[ 4956.053554]  schedule_preempt_disabled+0xa/0x10
[ 4956.053559]  __mutex_lock.constprop.0+0x13a/0x480
[ 4956.053566]  ? __switch_to_asm+0x36/0x70
[ 4956.053572]  addrconf_verify_work+0xa/0x20
[ 4956.053577]  process_one_work+0x1b4/0x370
[ 4956.053583]  worker_thread+0x53/0x3e0
[ 4956.053586]  ? process_one_work+0x370/0x370
[ 4956.053590]  kthread+0x11b/0x140
[ 4956.053593]  ? __kthread_bind_mask+0x60/0x60
[ 4956.053597]  ret_from_fork+0x22/0x30
[ 5036.258317] xhci_hcd 0000:03:00.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 7 comp_code 4
[ 5036.258372] xhci_hcd 0000:03:00.0: Looking for event-dma 000000012b464080 trb-start 0000000134dd1090 trb-end 0000000134dd1090 seg-start 0000000134dd1000 seg-end 0000000134dd1ff0
[ 5036.260219] usb 2-2.1: USB disconnect, device number 5
[ 5036.275327] xhci_hcd 0000:03:00.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[ 5036.329246] device wlan0 left promiscuous mode
[ 5036.379322] usb 2-2.1: ath9k_htc: USB layer deinitialized
```

ZerBea, 2020-12-20:
After some investigations, it looks like we are running into a (well know) driver issue:
```
[ 7380.126863] device wlp5s0f4u2 entered promiscuous mode
[ 7388.146435] kauditd_printk_skb: 4 callbacks suppressed
[ 7556.470126] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7556.975375] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7557.480254] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7557.984752] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7558.491501] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7559.079128] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7559.585129] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7560.089505] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7560.595753] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7561.100137] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7561.604635] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7562.108259] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7562.682879] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7563.189002] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7563.693506] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7564.199255] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7564.786885] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7565.292644] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7565.796886] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7566.301008] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
[ 7566.805258] ath: phy27: DMA failed to stop in 10 ms AR_CR=0x00000024 AR_DIAG_SW=0x02000020 DMADBG_7=0x00028040
```
After doing some DuckDuckGo searches, I noticed that the issue is well known for a long time.
I try to add some code to hcxdumptool to detect this driver issue and to terminate hcxdumptool on this error.

ZerBea, 2020-12-20:
By latest commits
https://github.com/ZerBea/hcxdumptool/commit/539251196b2eae7d3647dd2fa5b27a6c5f24d388
and
https://github.com/ZerBea/hcxdumptool/commit/647874025b5d223b3d1652aae2e331c5f8f7fd7b
I added some function to detect if the driver does not respond (especially on ath9k devices like TP-LINK TL-WN722N v1 - green LED doesn't flash any longer). Either hcxdumptool stops (if max errors reached) or you can terminate it by pressing ctrl+c.
That is all I can do until the driver receive a fix!

ZerBea, 2020-12-21:
```
21.12.2020
==========
removed TP-LINK TL-WN722N v1 (ath9k driver) from list of working devices due to driver issue
```
see changelog for more details

ZerBea, 2020-12-21:
Now the issue is moved to bugzilla.kernel.org:
https://bugzilla.kernel.org/show_bug.cgi?id=207397

@gonzabrusco can you confirm that hcxdumptool (latest git head) no longer freezes when the ath9k driver died?

ZerBea, 2020-12-21:
I don't expect a "fast solution":
https://duckduckgo.com/?q=ath9k+freeze&t=h_&ia=web

gonzabrusco, 2020-12-21:
I tried it a few times. It always hangs. Sorry. Only one time it showed me this: **"driver is busy: failed to transmit proberesponse"** but it hanged nevertheless. I'm using this as OS on WMWare with Windows 10 as host: https://www.offensive-security.com/kali-linux-vm-vmware-virtualbox-image-download/ (Kali Linux VMware 64-Bit (7z))

Sometimes it works for a couple of seconds before it hangs, sometimes it works for minutes. But always the led stops blinking and it freezes. The error message you added does not always shows up. Dmesg stops responding so I have to pull the adapter to see it. But it does not show anything relevant really. I'm not sure why it is freezing.

```
┌──(kali㉿kali)-[~/Desktop/hcxdumptool]
└─$ ./hcxdumptool -v                              
hcxdumptool 6.1.4-13-g41ca5ae (C) 2020 ZeroBeat
                                                                                                                                                           
┌──(kali㉿kali)-[~/Desktop/hcxdumptool]
└─$ sudo ./hcxdumptool -i wlan0 --enable_status=15
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE........: N/A
INTERFACE NAME............: wlan0
INTERFACE HARDWARE MAC....: e8de27a11847
DRIVER....................: ath9k_htc
DRIVER VERSION............: 5.9.0-kali1-amd64
DRIVER FIRMWARE VERSION...: 1.4
openSSL version...........: 1.1
ERRORMAX..................: 100 errors
BPF code blocks...........: 0
FILTERLIST ACCESS POINT...: 0 entries
FILTERLIST CLIENT.........: 0 entries
FILTERMODE................: unused
WEAK CANDIDATE............: 12345678
ESSID list................: 0 entries
ACCESS POINT (ROGUE)......: 0c811201d239 (BROADCAST HIDDEN)
ACCESS POINT (ROGUE)......: 0c811201d23a (BROADCAST OPEN)
ACCESS POINT (ROGUE)......: 0c811201d23b (incremented on every new client)
CLIENT (ROGUE)............: f0a225ee726e
EAPOLTIMEOUT..............: 20000 usec
EAPOLEAPTIMEOUT...........: 2500000 usec
REPLAYCOUNT...............: 64376
ANONCE....................: 37c8574ab1480f497f0b39b09947b6e3c3b2d128c9f2eb0b24149764fdfa9624
SNONCE....................: 5744a50ccbef7cb2e7290287b84a6b6e2c1bfc83ef58bb6b955dc5b0abb5de5a

08:23:59   1 ffffffffffff ac84c6b3e988 Torrecillas VII [BEACON]
08:23:59   1 ffffffffffff 14cc20b54db3 Renata 2.4GHz [BEACON]
08:23:59   1 34f39a07a7ec 14cc20b54db3 Renata 2.4GHz [PROBERESPONSE]
08:23:59   1 ffffffffffff 6872513abca8 CARILO_HOUSE [BEACON]
08:23:59   1 ffffffffffff 6014b3d7d3d0 Fibertel WiFi935 2.4GHz [BEACON]
08:24:00   6 ffffffffffff 28be9b9e170f Carilo A9 [BEACON]

driver is busy: failed to transmit proberesponse
```

gonzabrusco, 2020-12-21:
I'm now thinking in buying a Tp Link Archer T2uh because this 722N is having too many driver problems. Can you recommend that adapter?

ZerBea, 2020-12-21:
It is not easy to recommend a special device. Running into issues or not is driven by many different factors:
- the operation system - I prefer a rolling release running with latest kernel 
- the maintainer of the driver - https://patchwork.kernel.org/project/linux-wireless/list/
- the antenna (external, internal)
- the chipset of the device (sensitive or not - tx power doesn't matter because we are packet oriented)
- the chipset of the computer system (https://bugzilla.kernel.org/show_bug.cgi?id=202541)

My preferred devices are running a mt76 chipset and/or a RT2870/RT3070 chipset.

Also you should know that issues could happen - and we have had lots of them:
https://bugzilla.kernel.org/show_bug.cgi?id=202243
https://bugzilla.kernel.org/show_bug.cgi?id=201875
https://github.com/openwrt/mt76/issues/216#issuecomment-500999516
... and more.

gonzabrusco, 2020-12-22:
Thanks @ZerBea . I think I will try my luck with that adapter. 

I tried again my  TP-LINK TL-WN722N v1 with a live Kali from USB (without virtual machine) and it fails too. It just hangs. No warning from hcxdumptool. Dmesg also hangs. Something very wrongs is happening with the driver. I guess I will just sell it. For airodump works perfectly but I want to use your tool.

ZerBea, 2020-12-22:
My m76 devices are running fine. That include TP-LINK Archer T2UH as well as EDIMAX EW-7711UAN, ALLNET ALL-WA0150N and SEMPRE WU150-1.
What will happen if you run aireplay-ng and or mdk3/4 in parallel with airodump-ng on your WN722N. Both tools are active like hcxdumptool and produce high tx workload:
$ aireplay-ng -0 10000 ...... 
Does the driver freeze?

What happen if you run hcxdumptool passive (--silent)? Running this mode hcxdumptool will not transmit and we have no tx workload. Does the driver freeze?

Cheers
Mike

ZerBea, 2020-12-22:
Please notice that hcxdumptool (in attack mode) produce a hundred times more workload than the whole aircrack-ng suite. The tool is working as 512 ACCESS POINTs , 1024 CLIENTs, a dumper and a deauther, simultanious. That include also all EAP authentications. Additional it will request missing packets on packet loss to calculate a valid EAPOL message pair.

ZerBea, 2020-12-22:
Also you should know that the TP-LINK TL-WN722N v1 is a very old device: FCC approval date: 18 November 2009
It was replaced by v2 in 2017. Later on,  v2 was replaced by v3.
https://deviwiki.com/wiki/TP-LINK_TL-WN722N_v1.x

gonzabrusco, 2020-12-22:
> mdk3/4

Hello. Thank you for all your explanations. 

I can confirm that running airodump + aireplay + mdk3 **does not** freeze the adapter. Everything keeps working. I tried that for at least 30 minutes without problem. Instead hcxdumptool (without silent flag) freezed after 15 minutos approximately. 

I will try now the **silent** mode and leave it working for several hours. I will report back.

ZerBea, 2020-12-22:
Thanks for that information. Maybe we can find a solution.
BTW:
My TL-WN722N v1  also freezes when connected to my router under high traffic.

gonzabrusco, 2020-12-22:
After 5 hours running in silent mode, hcxdumptool did not freeze.

ZerBea, 2020-12-22:
Great, thanks for testing it. Now we can assume that the tx queue overflow on heavy workload and the device doesn't accept packets to transmit any longer.

ZerBea, 2020-12-22:
The issue passible can be caused by hardware (overheat) as mentioned here:
https://bugzilla.kernel.org/show_bug.cgi?id=61111
Until we don't know what happens exactly, I can't recommend this device.

gonzabrusco, 2020-12-22:
Yes, I think we will not resolve this mistery. At this point it's easier to change the device for another.