ZerBea/hcxdumptool #81: Problem to install

ggrswiss, 2019-12-12:
Hello,

i have this message when i wanted to install:
root@xxxxx:~# cd hcxdumptool
root@xxxxx:~/hcxdumptool# make
cc -O3 -Wall -Wextra -std=gnu99  -o hcxpioff hcxpioff.c 
cc -O3 -Wall -Wextra -std=gnu99  -o hcxdumptool hcxdumptool.c  -lcrypto
hcxdumptool.c:39:10: fatal error: openssl/evp.h: Aucun fichier ou dossier de ce type
 #include <openssl/evp.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:21: build] Error 1

I read many other comments related to this problem but it doesn't work for me.
what should i do ?

Thank you,

ZerBea, 2019-12-12:
You are missing the development files, which are required to compile hcxdumptool. They are not installed as default on KALI (I assume, you're running KALI).
To install this requirements use the following 'apt-get install libcurl4-openssl-dev libssl-dev zlib1g-dev'.
.Ça devrait fonctionner...

ZerBea, 2019-12-12:
read more here:
https://github.com/ZerBea/hcxdumptool/issues/73

ZerBea, 2019-12-13:
I changed the head line, because this issue is related to a missing dependency on KALI.

ggrswiss, 2019-12-13:
Yes, i'm using Kali Linux, the last version updated etc....
i tried the command line as you wrote it but it crushed;

root@xxxxx:~# apt-get install libcurl4-openssl-dev libssl-dev zlib1g-dev
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
E: Impossible de trouver le paquet libcurl4-openssl-dev
E: Impossible de trouver le paquet libssl-dev
E: Impossible de trouver le paquet zlib1g-dev

ZerBea, 2019-12-13:
Looks like you package management system is not up to date or broken and and your installation is not able to fetch the required packages "Impossible de trouver le paquet "
Try to update the packages index list:
https://linuxconfig.org/how-to-update-kali-linux
$ apt-get update

than try to install the packages
https://installlion.com/kali/kali/main/l/libcurl4-openssl-dev/install/index.html
$ sudo apt-get install libcurl4-openssl-dev libssl-dev zlib1g-dev

ggrswiss, 2019-12-13:
I have done all these steps but it still doesn't work.
Kali Linux is running on a SSD plugged in USB on my laptop. It's not a live Linux ! I have installed Kali on this hard drive as a classic OS as i have no place to put one more hard drive in the laptop.
Could it make some troubles ? Should i reinstall Kali ? which version do you suggest ? 
https://www.kali.org/downloads/

ZerBea, 2019-12-13:
I think there is something wrong with your installation, because you're not able to update and/or install packages.
To answer your questions:
Could it make some troubles ?
yes

Should i reinstall Kali ?
no, it doesn't make sense, because you're running into the same issues.

Which version do you suggest ? 
If you are looking for a Linux distribution to learn the basics of Linux and need a good starting point, Kali Linux is not the ideal distribution for you. You may want to begin with Ubuntu, Mint, or Debian instead. If you're interested in getting hands-on with the internals of Linux,  I suggest Arch Linux.
https://www.slant.co/versus/2690/2718/~arch-linux_vs_kali-linux

https://itsfoss.com/kali-linux-review/
https://www.kali.org/docs/introduction/should-i-use-kali-linux/

ZerBea, 2019-12-13:
BTW:
Arch has an excellent documentation (en français):
https://archlinux.fr/
https://wiki.archlinux.fr/
https://wiki.archlinux.fr/Installation

I'm running it on desktops, notebooks and Raspberry Pi's.

ggrswiss, 2019-12-14:
Actually, i have reinstalled Kali with an other version (the biggest one, which is i think KDE...)anyway .... now it works ! but i will look after this Arch Linux by simple curiosity !
Now, i hope your App will work, i have not tried it by now ... anyway, i will see !
Thank you !

ZerBea, 2019-12-15:
If hcxdumptool works or not depend on your system configuration (NetworkManager and wpa-supplicant stopped, crda/wireless regdb), how do you create the monitor interface (do not use airmon-ng) and the hardware (WiFi adapter.) Not every driver support mandatory monitor mode and full packet injection. Also, we have some kernel issues regarding some hardware combinations (eg. adpater connected to USB3 port).
It is mandatory to install hcxtools, too, because we're using pcapng. It is an extended capture format with many, many advantages. It is full compatible to Wireshark/tshark family (used and recommended default capture format, too), but not to aircrack-ng.

ZerBea, 2020-01-03:
Closed, becauee a missing dependency isn't a hcxdumptool issue.

closed 2020-01-03 by ZerBea