ZerBea/hcxdumptool #85: Filterlist is not working

mato200, 2019-12-30:
Idk what changed but for the past 1 whole hour i have been looking at tutorials and trying to figure out why is hcxdumptool atacking every single AP in range when i specifically told it to only attack the ones that have specific MAC (written in txt file),
Example:

hcxdumptool -o hash -i wlan0mon --filterlist_ap=mac.txt --filtermode=2 --enable_status=3

It even gave me pmkid of a AP that i wasnt wanting, but not of the one that i filtered for
I tried changing file names, even reinstalled the script. Tried with another AP's mac in the filterlist.

Another Question : What does   MP:M1M2 RC:1 EAPOLTIME:12584 (myAPssidHere)   mean ?

for the record, i am attacking my own AP at home 3 meters from the laptop, and the wild script is targeting the whole neighbourhood even though I have set the filterlist with AP MAC in it

ZerBea, 2020-01-02:
MP:M1M2 RC:1 EAPOLTIME:12584
You received(!) a handshake M1 from AP and M2 from Client and the time between this messages is 12564 usec
Are you sure hcxdumptool is attacking the whole neighbourhood? All filters are in the transmission branch and not(!) in the receiving branch. hcxdumptool is also a passive dumper (receiving branch). That means it will show you everything what it received independently of  the filter list entries, but it doesn't run active attacks (deauthentications, disassociations, requesting PMKIDs).

BTW:
Does the startup display show your filterlist entries:
FILTERLIST ACCESS POINT.: 5 entries
FILTERLIST CLIENT.......: 0 entries
FILTERMODE..............: 2
 
BTW 2:
Running hcxdumptool on virtual interfaces (wlan0mon) isn't a good idea. You should have received a warning.

ZerBea, 2020-01-02:
The behavior  is described in --help:
 
--filtermode= : mode for filter list
1: use filter list as protection list (default) in transmission branch
receive everything, interact with all APs and CLIENTs in range,
except(!) the ones from the filter lists
2: use filter list as target list in transmission branch
receive everything, only interact with APs and CLIENTs in range,
from the filter lists

You choose 2:
receive everything (!), interact with (attack) APs and CLIENTs in range,
from the filter lists

ZerBea, 2020-01-23:
Closed, because filter list is working as in help described.

closed 2020-01-23 by ZerBea