ZerBea/hcxdumptool #86: Is hcxdumptool PMKID attack meant to deauth devices?

CyrisXD, 2020-01-01:
Excuse my ignorance as I'm currently learning this field. I was under the impression that the PMKID attack was "silent", but when I use hcxdumptool it deauth's and disconnects my devices from my AP. 

My command I'm using is: 

`hcxdumptool -i wlan0mon -o test.pcapng --enable_status=3`

Any help is appreciated, 
Cheers

ZerBea, 2020-01-02:
None of the attacks is silent, unless you use this options:
--disable_client_attacks           : do not attack clients
                                     affected: ap-less (EAPOL 2/4 - M2) attack
--disable_ap_attacks               : do not attack access points
                                     affected: client-less (PMKID) attack
--silent                           : do not transmit!
                                     hcxdumptool is acting like a passive dumper

or a filter list:
-filterlist_ap=<file>             : access point mac filter list
                                     format: 112233445566 + comment
                                     maximum entries 256
                                     run first --do_rcascan to retrieve information about the target
--filterlist_client=<file>         : client mac filter list
                                     format: 112233445566 # comment
                                     maximum entries 256
                                     run first --do_rcascan to retrieve information about the target
--filtermode=<digit>               : mode for filter list
                                     1: use filter list as protection list (default) in transmission branch
                                        receive everything, interact with all APs and CLIENTs in range,
                                        except(!) the ones from the filter lists
                                     2: use filter list as target list in transmission branch
                                        receive everything, only interact with APs and CLIENTs in range,
                                        from the filter lists

BTW:
Running hcxdumptool on a virtual interface (wlan0mon) isn't a goog idea..

CyrisXD, 2020-01-02:
Thanks for the clarification @ZerBea. Much appreciated.

closed 2020-01-02 by CyrisXD

ZerBea, 2020-01-03:
No problem, you're welcome.
BTW:
filter list options doesn't affect real time display options.
--enable_status=3 means, that you requested to see  all received EAPOL and PROBE REQUEST/PROBE RESPONSE frames on the channel.

MarcusXavierr, 2021-02-25:
> None of the attacks is silent, unless you use this options:
> --disable_client_attacks : do not attack clients
> affected: ap-less (EAPOL 2/4 - M2) attack
> --disable_ap_attacks : do not attack access points
> affected: client-less (PMKID) attack
> --silent : do not transmit!
> hcxdumptool is acting like a passive dumper
> 
> or a filter list:
> -filterlist_ap= : access point mac filter list
> format: 112233445566 + comment
> maximum entries 256
> run first --do_rcascan to retrieve information about the target
> --filterlist_client= : client mac filter list
> format: 112233445566 # comment
> maximum entries 256
> run first --do_rcascan to retrieve information about the target
> --filtermode= : mode for filter list
> 1: use filter list as protection list (default) in transmission branch
> receive everything, interact with all APs and CLIENTs in range,
> except(!) the ones from the filter lists
> 2: use filter list as target list in transmission branch
> receive everything, only interact with APs and CLIENTs in range,
> from the filter lists
> 
> BTW:
> Running hcxdumptool on a virtual interface (wlan0mon) isn't a goog idea..

@ZerBea 
If running hcxdumptool on a virtual interface is not a good idea, what types of interfaces would be good options? What are the problems of running hcxdumptool on virtual interfaces? Can I fail to succeed in the invasion thanks to this? Is there any risk of damaging my hardware?

ZerBea, 2021-02-25:
There is no risk to damage your hardware.
hcxdumptool doesn't need a tool that set monitor mode. Instead it will set monitor mode because it need full access to the hardware of the interface. It doesn't like to share the interfaces with other tools which are able to change the channel.
If you set monitor mode by a third party tool (e.g.: iw or airmon-ng, that internal use iw) this is not guaranteed and hcxdumptool may not work as expected.
Also notice:
airmon-ng is designed to be used by aircrack-ng tools and not(!) by hcxdumptool.

Recommended workflow:
stop all(!) network services that take access to the interface
run hcxdumptool
when terminated, restart the network services

atonn, 2021-06-04:
sorry to post in a closed issue, but i have an old script lying around which in addition to --silent and the others mentioned here also uses --disable_active_scan, but this option no longer exists in the current version and I am failing to find out why. Cheers!

ZerBea, 2021-06-04:
Instead of the old option, now we use several other options to control the behavior:
silent = do not transmit (= disable_ap_attacks + disable_client_attacks)
disable_ap_attacks = do not attack APs (do not transmit frames to APs)
disable_client_attacks = do not attack CLIENTs  (do not transmit frames to APs)
and
stop_ap_attacks  = do not attack APs any longer if we received N BEACONs
stop_client_m2_attacks = do not attack a CLIENT any longer, if we received N M2 frames
additional there are some options to resume attacks again.

disable_active_scan was only useful in rca scan mode when scanning for new networks doing a passive scan while the new options working in every mode.
You simply don't need it any longer (silent is doing the same).