ZerBea/hcxdumptool #88: New version, new output - where is the "FOUND PMKID" status?

mtnsec, 2020-01-03:
Is there a combination for --enable_status that will bring back the human readable messages that the old version output?  We've become reliant on being able to count the messages that say:

"FOUND PMKID"
"FOUND HANDSHAKE AP-LESS"
"FOUND AUTHORIZED HANDSHAKE"

And the stats bar that updates when you press enter:

"INFO: cha=9, rx=13802, rx(dropped)=1073, tx=319, powned=2, err=0"

Or if these messages are gone, how do we interpret the new output to differentiate between handshakes and PMKID while hcxdumptool is still running, not after when we run hcxpcap tool against it.

Thanks!

ZerBea, 2020-01-03:
FOUND PMKID found is no longer needed, instead the complete PMKID is displayed, inclusive ESSID, MAC_AP and MAC CLIENT.
example:
18:52:29   6 acde48520227 <-> 001579c3e78f PMKID:364e20a580cb242529973ae9f63ab2d8 (testESSID)
You can use this information to feed hashcat directly while hcxdumptool is running.
Read more about the new hashline here:
https://github.com/hashcat/hashcat/issues/1816

Do you really want the stupid old format back?

FOUND AUTHORIZED HANDSHAKE is no longer needed, because complete information about the handshake are displayed inclusive ESSID, MAC_AP and MAC CLIENT and detailed information about the handshake.
example:
18:52:29   6 acde48520227 <-> 001579c3e78f MP:M1M2 RC:1 EAPOLTIME:2954 (testESSID)
handshake message pair M1M2 (unauthorized) inclusive replaycount and time between the messages
to determine if we are outside of the APs EAPOLTIMER values.

M1M2 = unauthorized
M1M4 = unauthorized
M2M3 = authorized
M3M4 = authorized

If replaycount is the same as REPLYCOUNT in head, it is an AP-LESS attack.

$ hcxdumptool -i wlp3s0f0u2 --enable_status=63 --reactive_beacon
initialization...

start capturing (stop with ctrl+c)
NMEA 0183 SENTENCE......: N/A
INTERFACE NAME..........: wlp3s0f0u2
INTERFACE HARDWARE MAC..: c83a35cb08e3
DRIVER..................: rt2800usb
DRIVER VERSION..........: 5.4.7-arch1-1
DRIVER FIRMWARE VERSION.: 0.36
ERRORMAX................: 100 errors
FILTERLIST ACCESS POINT.: 0 entries
FILTERLIST CLIENT.......: 0 entries
FILTERMODE..............: 0
WEAK CANDIDATE..........: 12345678
PREDEFINED ACCESS POINT.: 0 entries
MAC ACCESS POINT........: 00054fdbf654 (incremented on every new client)
MAC CLIENT..............: a4a6a9d6a9bd
REPLAYCOUNT.............: 62711
ANONCE..................: 909f0a4f63b0428589bfcafaa06e1867623f21dca0866f5c05d6e0c9c907490a
SNONCE..................: 51969af01281e278f95d4579770611a807eb0f050de28b799bcfd10138e1ed2b

Do you really want the stupid old format back?

The internal status isn't gone. It can be activated via option (bitmask 64):
--enable_status=<digit>            : enable real-time display (waterfall)
                                     some messages ​​are shown only once at the first occurrence
                                       0: no status (default)
                                       1: EAPOL
                                       2: PROBE REQUEST/PROBE RESPONSE
                                       4: AUTHENTICATON
                                       8: ASSOCIATION/REASSOCIATION
                                      16: BEACON
                                      32: GPS (once a minute)
                                      64: internal status
                                     128: run as server
                                     256: run as client

19:02:00   6 INFO ERROR:2 INCOMMING:2849 OUTGOING:696 GPS:0 RINGBUFFER:41

hcxpcaptool is deprecated (successor hcxpcapngtool), hash modes 250x and 1680x are deprecated, too. hccapx will R.I.P, soon.

Read more about the new hash line, here:
https://github.com/hashcat/hashcat/issues/1816

ZerBea, 2020-01-03:
BTW:
The stats bar never(!) updated by pressing enter. Instead it was and it is updated by internal timer.

mtnsec, 2020-01-03:
Thanks for the explanation.  We've been using the latest stable hashcat version 5.1.0 so we were unaware that there is a newer hash format and newer hcxpcapngtool to use.  After double checking, we confirmed that the stable hashcat version is still 5.1.0 it doesn't have this new format available, neither do the official hashcat docs or wiki. We don't use dev versions, so until hashcat updates their stable to this new version that supports the new hcxdumptool output we're still stuck in the past needing the old tools to work.

ZerBea, 2020-01-03:
For sure, just wait for hashcat 6.0.0 (will come soon).

ZerBea, 2020-01-03:
By latest git head, you will get an aditional information about received PMKIDs and EAPOL message pairs:
19:32:00  11 INFO ERROR:0 INCOMMING:929 OUTGOING:333 PMKID:32 MP:0 GPS:0 RINGBUFFER:31
We count all PMKIDs and all EAPOL message pairs, here.

mtnsec, 2020-01-03:
Awesome, thanks!

closed 2020-01-23 by ZerBea