ZerBea/hcxdumptool #90: Version 6.01, missing save_rcascan

BulletByte, 2020-01-19:
Hi all, i'm missing the option. It's there till 5.2.x but with the segfault when quitting hcxdumptool. I couldnt find a hint if this is by purpose, hcxdumptool.c did not include it.

ZerBea, 2020-01-19:
Really ugly issue and epic fail. I was a little bit busy to implement hashct's hashline to hcxtools and  forgot to include the whole part to save rca_scan traffic to pcapng file. I will start to fix this issue, now.
Thanks for reminding me.

ZerBea, 2020-01-19:
Ok, by this commit:
https://github.com/ZerBea/hcxdumptool/commit/476a0eb244e0933c9cc7b33a31cd38982951d462
issue should be fixed.  We just missed a simple line of code
$ hcxdumptool -i interface --do_rcascan -o test.pcapng
should do the job, now.
Again, thanks for reminding me.
Please test and close issue if it works for you.

BulletByte, 2020-01-20:
Confirmed, thanks.
Schönes Tool übrigens

closed 2020-01-20 by BulletByte

ZerBea, 2020-01-20:
Das freut mich. Danke.