ZerBea/hcxdumptool #91: Allow to set AP MAC address

DaniloNC, 2020-01-21:
Hi!

Since the rewrite, the option to set the AP Mac address and the Client Mac Address has been removed. Right now both are randomly generated and cannot be configured.

Please, add an option to define the initial AP Mac Address (for clientless attacks).

ZerBea, 2020-01-21:
Done by latest commit:
https://github.com/ZerBea/hcxdumptool/commit/436fc4185c3faaed14fbb2abbfbfc7040f2b3990
--mac_ap                           : use this MAC as ACCESS POINT MAC instead of a randomized one
                                     format: 112233445566
--mac_client                       : use this MAC as CLIENT MAC instead of a randomized one
                                     format: 112233445566

Please keep in mind:
mac_ap is only the initial mac. To prevent counter measures, it became more and more randomized or directed proberequest dependent.
Please test and close issue, if it works.

DaniloNC, 2020-01-21:
Thanks!
I will test and report here.

DaniloNC, 2020-01-27:
Is it possible to fix the vendor part (first 3 bytes) of the Mac Address?
It seems the MAC will randomize the full mac address after some time, including the vendor part.

My use case to use mac_ap is to blacklist those MACs on Wigle/Kismet so I can know those beacons came from me.

ZerBea, 2020-01-27:
No, that is not possible, because of full randomization.
On init, we use a start MAC_AP and increment it as long as we get new ESSIs from undirected proberequests. After short period we start to use real(!) MACs, received from directed proberequests and proberesponses. I can't change that, because this behavior prevent countermeasures against hcxdumptool and has a stealth function. After a while, hcxdumptool will act as a "normal AP", running received MAC/ESSID combinations.
Shall we remove the MAC_AP option - because it is nearly useless.

DaniloNC, 2020-01-27:
Thanks for the update and explanation!
Yes, I agree that mac_ap and mac_client could be removed.

Off-Topic:
have you considered creating automated hcxdumptool RPi images? (e.g. https://github.com/evilsocket/pwnagotchi/tree/master/builder )

If you think this is a good idea, I could help with it.

ZerBea, 2020-01-27:
Ok, I'm going to remove the options.
Running Arch Linux on a Raspberry pi is very easy.
`
sudo fdisk /dev/sdx
 o p n p 1 <ret> +100M t c n p 2 <ret> <ret> w 

sudo mkfs.fat -F32 /dev/sdx1
sudo mkfs.ext4 /dev/sdx2

mkdir boot
mount /dev/sdx1 boot
mkdir root 
mount /dev/sdx2 root

wget http://archlinuxarm.org/os/ArchLinuxARM-rpi-latest.tar.gz 
su
tar -xvf ArchLinuxARM-rpi-latest.tar.gz -C root
sync
mv root/boot/* boot
sync
umount boot root

SD card into the Raspberry Pi
root
root

passwd
echo LANG=de_DE.UTF-8 > /etc/locale.conf
echo LC_COLLATE=C >> /etc/locale.conf
echo KEYMAP=de-latin1 > /etc/vconsole.conf
ln -sf /usr/share/zoneinfo/Europe/Berlin /etc/localtime

nano /etc/locale.gen
#de_DE.UTF-8 UTF-8
#de_DE ISO-8859-1
#de_DE@euro ISO-8859-15
locale-gen

nano /etc/hostname
alarmpi <- wlanpi

pacman -Syu
pacman -S make gcc watchdog openssl rsync fake-hwclock zlib crda linux-firmware linux-raspberrypi-headers systemd-resolvconf 
pacman -Rs netctl wpa_supplicant

pacman -S iw
systemctl enable fake-hwclock fake-hwclock-save.timer
systemctl start fake-hwclock

watchdog:
nano /etc/watchdog.conf
watchdog-device	= /dev/watchdog
watchdog-timeout = 15
max-load-1 = 24

systemctl enable watchdog.service

nano /etc/sshd/sshd_config
#PermitRootLogin without-password
PermitRootLogin yes

ssh-copy-id -i your_key.pub root@192.168.0.10
ssh -i your_key root@192.168.0.10
nano /etc/ssh/sshd_config
PasswordAuthentication no

systemctl enable sshd.service

cp /usr/lib/systemd/system/getty@.service /etc/systemd/system/autologin@.service
nano /etc/systemd/system/autologin@.service
ExecStart=-/sbin/agetty  -a root --noclear %I $TERM
Type=simple

Hide kernel messages from terminal
/etc/sysctl.d/20-quiet-printk.conf
kernel.printk = 3 3 3 3

Disable the old getty service and enable the new autologin service.
systemctl disable getty@tty1
systemctl enable autologin@tty1

statische ip
nano /etc/dhcpcd.conf
interface eth0
static ip_address=192.168.0.10/24
static routers=192.168.0.1
static domain_name_servers=192.168.0.1

nur wenn eth0 autostart:
systemctl enable dhcpcd@eth0.service

kurzfristig ein:
systemctl start dhcpcd@eth0.service

nano /etc/systemd/network/eth0.network
DHCP=no

nano /etc/conf.d/wireless-regdom
uncomment VE

nano /boot/cmdline.txt
ipv6.disable=1.

Alternatively
ipv6.disable_ipv6=1
instead will keep the IPv6 stack functional but will not assign IPv6 addresses to any of your network devices.

rm /etc/modules-load.d/raspberrypi.conf
`

ZerBea, 2020-01-27:
Once the Arch is installed, the only commands you need are:
update system:
$ pacman -Syu
Clean cache:
$ pacman -Scc

ZerBea, 2020-01-27:
Once Arch is installed and configured, you can use the supplied scripts to backup your system to tgz files (boot and root).
create backup:
https://github.com/ZerBea/hcxdumptool/blob/master/usefulscripts/pireadcard
restore backup
https://github.com/ZerBea/hcxdumptool/blob/master/usefulscripts/piwritecard

Installing Arch Linux by hand is a good idea. You will learn much about Linux.
There no real need to create an automated initial install script.

ZerBea, 2020-01-29:
And the integration in Arch Linux is perfect!
Today I released v6.0.1. Only a few minutes later it was flagged out:
https://www.archlinux.org/packages/?q=hcxdumptool
and we can expect an update via pacman, soon.

The same applies to:
https://archlinuxarm.org/packages

Arch Linux is really fast and 100% configurable. Wiki and support is fantastic. 
That is the major reason for me to use and support this distribution!

ZerBea, 2020-01-29:
Scripts to aid in installing Arch Linux:
https://archlinuxarm.org/packages/arm/arch-install-scripts
https://git.archlinux.org/arch-install-scripts.git

A guide is here:
https://wiki.archlinux.org/index.php/Installation_guide

closed 2020-06-18 by DaniloNC