ZerBea/hcxdumptool #94: might be a bug, might not be

ukscone, 2020-02-13:
When running hcxdumptool for a long time (several hours) I sometimes get some "weird" values in the probe request that although I guess could be correct look a little weird. I've also had my screen go all weird (sorry no screenshot of that) where the text on the screen (ssh session via tmux) goes all weird with line drawing chartacters & graphic chars & unreadable and I have to disconnect and reconnect to get back to normal alphabetic characters

[image]
[image]
[image]

ZerBea, 2020-02-13:
When running hcxdumptool for a long time (several hours) I sometimes get some "weird" values in the probe request that although I guess could be correct look a little weird.
-> Correct, caused by bit errors in the PLCP header
In monitor mode the adapter does not check to see if the cyclic redundancy check (CRC
values are correct for packets captured. The device is able to detect the Physical Layer
Convergence Procedure (PLCP) preamble and is able to synchronize to it, but if there is
a bit error in the payload. This can lead to unexpected results.

I've also had my screen go all weird
-> That is an issue. Thanks for reporting it. I'm going to fix that. Please stay tuned...

ZerBea, 2020-02-14:
Should be fixed by this commit:
https://github.com/ZerBea/hcxdumptool/commit/1731c1fccca53740c0a98abcff3899e5f6307f41

Now we replace characters < 0x20 and characters > 0x7e with ".". Additional we handle ESCAPE sequences within ESSIDs. Terminal should no longer be destroyed.
Please test and close issue, if it works for you.

BTW:
Unfortunately we can't handle bit errors. To detect them, run hcxpcapngtool your_dump_file.pcapng.
hcxpcapngtool will give you an overview about this errors.

What WiFi adapter do you use? Regarding the the bit errors, I assume an ATHEROS chipset.

Cheers
Mike

ukscone, 2020-02-14:
Using a Tenda  W311U+ -- 148f:3070

i'll start a long run and keep an eye on it after coffee

ukscone, 2020-02-14:
4 hours and counting and so far all is looking good. i'll leave it running until tea time and see if anything happens

ZerBea, 2020-02-14:
I have several Tenda W311U+. We can expect a few PLCP errors, running this devices. But that is ok.
BTW:
Please use latest git head. I fixed several other issues.

ukscone, 2020-02-15:
i had it running from 10:33 to 19:00 and it all seemed good. no corrupted screen so i'll close this for now as it's looking ok (so far)

ZerBea, 2020-02-15:
Great, Thanks for your feedback.

ukscone, 2020-02-17:
ah it went weird again. latest commit. 
[image]
[image]

ukscone, 2020-02-17:
btw the blank lines are because I was in the wrong window and pressed return key a few times. but it's not thise that cause it

ZerBea, 2020-02-17:
Maybe tmux isn't configured correct:
https://hackernoon.com/customizing-tmux-b3d2a5050207
There are also several issues related to tmux:
https://github.com/tmux/tmux/issues/322
https://askubuntu.com/questions/410048/utf-8-character-not-showing-properly-in-tmux
https://superuser.com/questions/417027/why-are-random-characters-inserted-into-my-tmux-session
https://gitlab.com/gnachman/iterm2/issues/1737
https://stackoverflow.com/questions/19479810/zsh-tmux-oh-my-zsh-autocomplete-produces-remnant-characters

By the latest fix, we replace all non printable characters with a dot:
for(c = 0; c < essidlen; c++)
	{
	if((essid[c] < 0x20) && (essid[c] > 0x7e)) essidstring[p++] = '.';
	else if(essid[c] == 0x5c)
		{
		essidstring[p++] = 0x5c;
		essidstring[p++] = 0x5c;
		}
	else essidstring[p++] = essid[c];
	}

Also we ignore escape sequences.

I suggest not to use tmux. Instead use hcxdumtpools build in server/client function.

ZerBea, 2020-02-17:
Please take a look at this here:
https://superuser.com/questions/531537/tmux-visual-artifacts-strange-character-map
Looking similar to your output. So we can assume a tmux misconfiguration or a tmux issue.

Just do a google search: "tmux artifacts" and you'll get several hits.

ukscone, 2020-02-17:
yes that looks like it. sorry for the false issue report

closed 2020-02-17 by ukscone

ZerBea, 2020-02-18:
No problem. Now we know that is was a good decision to add server/client function to hcxdumptool.