ZerBea/hcxdumptool #95: observation not an issue

ukscone, 2020-02-16:
with exactly the same hardware & options i've noticed that since you fixed issue #94 I seem to be getting a lot less pmkid's/MP's over the same timeframe. I was getting something like 1000+ PMKIDs & 200 or so MPs (works out at about 20 or 30 of crackable both) according to --enable_status=127 (internal stats) since the fix the internal stats over the same time from are in the 10s. no changes to hardware, software or positioning of the pi (running kali 2019.3 rolling)

ZerBea, 2020-02-16:
Good observation. The counted PMKIDs and MPs in status line are raw values.
Now we should get much less PMKIDs and MPs here, because we're making much less noise.
INFO ERROR:0 INCOMING:958 OUTGOING:522 PMKID:17 MP:0 GPS:0 RINGBUFFER:3

To identify if it is really an issue or not, please run
old version for at least 15 minutes
than convert with hcxpcapngtool and check how many PMKIDs and MPs are written to output file.

now run
new version for at least 15 minutes
than convert with hcxpcapngtool and check how many PMKIDs and MPs are written to output file.

If we have the same value +/- 5, everything is ok. If not, I've to hunt for that issue.

ukscone, 2020-02-16:
ran both for about an hour each and usable pmkids and mp's is within 2 so think all is good

closed 2020-02-16 by ukscone

ZerBea, 2020-02-16:
Great. Goal is to reduce the noise on the current channel. The less noise the more we can receive. Additional I added a new attack vector.