ZerBea/hcxdumptool #100: Understanding hcxdumptool output , PMKIDs (zeroed and useless) ?

FakuVe, 2020-02-21:
Hi there,
I am launching this attack to some of my routers succesfully before,
Now I've changed and I'm doing it with some other newones , and for some reason I am getting just 2 PMKIDS when I use to have like more than 6
This is the output I get after  putting the following input.
-------------------------
root@kali:~# hcxdumptool -i wlan1mon -o sberXXXXX.pcapng --enable_status=1
initialization...
warning: wlan1mon is probably a monitor interface
interface is already in monitor mode
                                                                                                                  
start capturing (stop with ctrl+c)                                                                                
NMEA 0183 SENTENCE........: N/A                                                                                   
INTERFACE NAME............: wlan1mon                                                                              
INTERFACE HARDWARE MAC....: 00c0ca96b1d9                                                                          
DRIVER....................: rt2800usb                                                                             
DRIVER VERSION............: 5.3.0-kali2-amd64                                                                     
DRIVER FIRMWARE VERSION...: 0.36                                                                                  
ERRORMAX..................: 100 errors                                                                            
FILTERLIST ACCESS POINT...: 0 entries                                                                             
FILTERLIST CLIENT.........: 0 entries                                                                             
FILTERMODE................: 0                                                                                     
WEAK CANDIDATE............: 12345678                                                                              
PREDEFINED ACCESS POINT...: 0 entries                                                                             
MAC ACCESS POINT BROADCAST: 4876048f206b                                                                          
MAC ACCESS POINT..........: 4876048f206c (incremented on every new client)                                        
MAC CLIENT................: c8aacc11eb69                                                                          
REPLAYCOUNT...............: 63833                                                                                 
ANONCE....................: d3d821027bac5443166495d06b3f928ed027c1f341ed93a32a70e8c049ae7cc3
SNONCE....................: 5440d9011830720d7946618188dafb6768f1a93da09631a987bfc0d301dcbd4d

21:30:02   6 020fb511eb69 <-> b8ee0edc4588 MP:M1M2 RC:0 EAPOLTIME:4816 (BTHub6-XXXX)
21:30:02   6 020fb511eb69 <-> b8ee0edc4588 MP:M2M3 RC:1 EAPOLTIME:4591 (BTHub6-XXX)
21:30:22  10 c8aacc11eb69 <-> e4a7c58fe46e PMKID:03a7d262d18dadfac106555cb02b3e5a (XXXX)
21:34:29   4 50dce793923c <-> d80d17b01042 MP:M1M2 RC:63833 EAPOLTIME:18569 (XXXXX)
21:34:29   4 50dce793923c <-> d80d17b01042 MP:M2M3 RC:2 EAPOLTIME:3335 (XXXXXX)
21:34:51   6 a8e5440e91dd <-> 342eb6534f22 MP:M1M2 RC:3 EAPOLTIME:28274 (XXXXXX)
21:37:03  11 6cc7ec9ae7f3 <-> bca5111434c8 MP:M1M2 RC:1 EAPOLTIME:6143 (XXXXXXX)
21:38:08   3 14205ece9987 <-> b8c1a2642040 MP:M1M2 RC:0 EAPOLTIME:130530 (XXXXXXX)
21:38:47   4 cc50e3527ef9 <-> 4876048f207b MP:M1M2 RC:63833 EAPOLTIME:9369 (XXXXX)
21:42:07  10 c0eefbd7f0e8 <-> 0c8063a465a8 MP:M2M3 RC:2 EAPOLTIME:3371 (XXXXXX)
21:47:47   3 5c7776913ddd <-> 4876048f206f MP:M1M2 RC:63833 EAPOLTIME:78069 (blackberry)
21:49:43   1 c048e6f2b52c <-> 4c1b86b5e39e MP:M1M2 RC:63833 EAPOLTIME:54867 (BT-XXXX)
21:52:59   6 bc98df062aa2 <-> d05794b19bb8 MP:M1M2 RC:63833 EAPOLTIME:111927 (XXXXX)
22:00:11   8 7c2a314055ce <-> 4876048f2089 MP:M1M2 RC:63833 EAPOLTIME:141684 (XXXXX)
22:04:45  11 020fb51990c9 <-> b8ee0eddb4c2 MP:M1M2 RC:63833 EAPOLTIME:11049 (BTHub6-XXXXX)
22:05:47   6 c49ded1b9b29 <-> 6ecb4d65d6de MP:M1M2 RC:63833 EAPOLTIME:9889 (iPhone)
22:09:00  11 a45046ebde97 <-> 40313c90220f MP:M2M3 RC:2 EAPOLTIME:3018 (BTHub6-XXXXXX)
22:14:27   3 7c2a314055ce <-> ac3b7702a69e MP:M1M2 RC:63833 EAPOLTIME:35285 (XXXXX)
22:15:26  11 c0e8621990c9 <-> 18c5014b03d8 MP:M1M2 RC:2 EAPOLTIME:7142 (EXT2-BTHub6-XXXX)
22:15:26  11 c0e8621990c9 <-> 18c5014b03d8 MP:M2M3 RC:3 EAPOLTIME:22938 (EXT2-BTHub6-XXXX)
22:16:20   1 020fb53b2089 <-> 5890432a2470 MP:M1M2 RC:63833 EAPOLTIME:14804 (BTHub6-XXX)
22:21:14   6 c49ded1b9b29 <-> ce4b1b8e9326 MP:M1M2 RC:63833 EAPOLTIME:6875 (iPhone)
22:25:06   3 047295239e9b <-> 4876048f2099 MP:M1M2 RC:63833 EAPOLTIME:16940 (BTHub6-XXX)
22:28:08   6 c49ded1b9b29 <-> 2ac02d3c5a16 MP:M1M2 RC:63833 EAPOLTIME:18735 ()
22:28:14  11 c49ded1b9b29 <-> 1e3483e9e2f9 MP:M1M2 RC:63833 EAPOLTIME:19336 (iPhone)
22:32:27   6 a477331d5c56 <-> ac3b7702806a MP:M1M2 RC:63833 EAPOLTIME:49890 (BTHub6-XXXX)
22:38:22  11 18810e5e2da9 <-> c83a35f3a2b0 MP:M1M2 RC:0 EAPOLTIME:23963 (EXT-XXXXX)
22:48:05   4 a20460eaa829 <-> b8ee0edc17f2 MP:M1M2 RC:63833 EAPOLTIME:12500 (BTHub6XXX)
22:59:35   1 94fb293a0293 <-> 4876048f209e MP:M1M2 RC:63833 EAPOLTIME:138163 (XXXXX)
23:00:10   6 806c1bc3e46c <-> 08d59da2110c MP:M1M2 RC:0 EAPOLTIME:3568 (BTHub6-XXX)
23:06:06  11 d89c67ce490d <-> 589043c3a69c MP:M1M2 RC:63833 EAPOLTIME:35047 (BTHub6-XXX)
23:29:43   4 60f4455a0bf3 <-> b8ee0edcd642 MP:M1M2 RC:63833 EAPOLTIME:5009 (BTHub6-XXXX)
23:32:59   8 c49ded1b9b29 <-> a00460eaa829 MP:M1M2 RC:63833 EAPOLTIME:83953 (BTHub6-TXXXT)
23:42:50   6 2816a85a4674 <-> 50d4f7aadc93 MP:M1M2 RC:63833 EAPOLTIME:7735 (BTHub6-XXXX)
-------------------------

Of those when I do the conversion to hashcat readable files it  only gets me 2 PMKIDS

---------------------------

root@kali:~# hcxpcaptool -E essudkust -I identitylist -U usernamelist -z sbeXXXXXX.16800 sberXXXXXX.pcapng 

reading from sbXXXXXX.pcapng
                                                
summary capture file:                           
---------------------
file name........................: sXXXXXX.pcapng
file type........................: pcapng 1.0
file hardware information........: x86_64
capture device vendor information: 00c0ca
file os information..............: Linux 5.3.0-kali2-amd64
file application information.....: hcxdumptool 6.0.1 (custom options)
network type.....................: DLT_IEEE802_11_RADIO (127)
endianness.......................: little endian
read errors......................: flawless
minimum time stamp...............: 20.02.2020 21:29:27 (GMT)
maximum time stamp...............: 20.02.2020 23:44:29 (GMT)
packets inside...................: 7017
skipped damaged packets..........: 0
packets with GPS NMEA data.......: 0
packets with GPS data (JSON old).: 0
packets with FCS.................: 0
beacons (total)..................: 347
beacons (WPS info inside)........: 94
beacons (device info inside).....: 1
probe requests...................: 138
probe responses..................: 251
association requests.............: 34
association responses............: 90
reassociation requests...........: 12
reassociation responses..........: 24
authentications (OPEN SYSTEM)....: 137
authentications (BROADCOM).......: 62
authentications (APPLE)..........: 8
EAPOL packets (total)............: 5984
EAPOL packets (WPA2).............: 5984
PMKIDs (zeroed and useless)......: 194
PMKIDs (not zeroed - total)......: 2
PMKIDs (WPA2)....................: 203
PMKIDs from access points........: 2
best handshakes (total)..........: 34 (ap-less: 23)
best PMKIDs (total)..............: 2

summary output file(s):
-----------------------
2 PMKID(s) written to sbXXXX.16800

------------------------------------------------------------------------------------------------------------

Does anyone knows how to understand what are the above zeroed and useless,  and why I can only use 2 at the end?
What am I posibly doing wrong?
Has the attack already been patched by the router manufacturers or some kind of like that?

Thanks for your assistance comunity!!

ZerBea, 2020-02-21:
Zeroed PMK mean:
The initial EAPOL sequence is calculated using a zeroed PMK. It is not the real PMK of the network. That can happen if a CLIENT run an empty wpa_supplicant.conf or an AP run an empty hostapd.conf. hcxdumptool receive and accept that because it is designed as penetration testing and analysis tool (as well as hcxtools).

Zeroed PMKID mean:
On first connect of a CLIENT, the AP use an initial zeroed PMKID. You can't recover a PSK from it.

Additional you received 2 warnings and you got to take this seriously:
warning: wlan1mon is probably a monitor interface
interface is already in monitor mode

Do not use airmon-ng to set monitor mode! It is designed to run in combination with aircrack-ng suite and creates a virtual NETLINK interface!
hcxdumptool need full access to the hardware and run it's own monitor mode. It doesn't use NETLINK.
NETLINK is good to stream multimedia, but not to perform high speed attacks.
https://www.quora.com/What-are-the-differences-between-netlink-sockets-and-ioctl-calls

Keep also in mind:
Not every router transmit a PMKID!!

Closed this, because it isn't an issue, but you can still ask your questions, here.

closed 2020-02-21 by ZerBea

ZerBea, 2020-02-21:
PMKIDs (WPA2)....................: 203
PMKIDs from access points........: 2
best PMKIDs (total)..............: 2

You received 203 PMKIDs total, but 201 are duplicates. So only 2 unique PMKIDs left!

FakuVe, 2020-02-21:
Thanks your help really.

You shouldnt be dealing with n00bs like me haha. Thanks again for the
community for such a strong tool.

ZerBea, 2020-02-21:
No problem at all, you're welcome.
802.11 stuff is really hard core. hcxtools are designed as analysis tools. So every tool will show you a complete information (all useful frames and how often they are detected).

FakuVe, 2020-02-21:
Thanks. I will have a look.

Btw just as a personal question. Im quite interested in Cybersecurity.
Do you have yourself a private career about it? Did you thought about? Do
you think this will have potential in the futurre? Any relevant university
degree or course?
Thanks for your time.

B. R.
Rafael

ZerBea, 2020-02-21:
Unfortunately I can't answer this. I have never been in this profession.

FakuVe, 2020-02-21:
No probs just wondering. Thanks for the opinion anyways.

B. R.