ZerBea/hcxdumptool #99: Unable to run in the OSX

springfred, 2020-02-21:
OS not supported

ZerBea, 2020-02-21:
From README.md:
Operating system: Arch Linux (strict), Kernel >= 4.19 (strict). It may work on other Linux systems (notebooks, desktops) and distributions, too (no support for other distributions, no support for other operating systems)

I am tired to wrestle against proprietary (closed source), commercial operating systems.

closed 2020-02-21 by ZerBea

ZerBea, 2020-02-21:
The driver must support monitor mode an full packet injection:
https://github.com/ZerBea/hcxdumptool/issues/98#issuecomment-589551355
That isn't the case on proprietary operating systems.

If you need that feature, you should request it here:
https://support.apple.com/en-ca